magnusja/libaums #114: It's slow to read and write!

FanHongchuang, 2017-10-13:
### It's slow to read and write for USB Mass Storage devices on Android. 
### How to speed it up?

magnusja, 2017-10-13:
Do you have some numbers? And do you have something to compare these numbers to? Is it faster with other apps?

FanHongchuang, 2017-10-13:
### I copy the same files by  " ES File Explorer " and "libaums" respectively, " ES File Explorer " is a lot faster than "libaums".

### I'm so sorry, my English is terrible.

magnusja, 2017-10-13:
No worries, I can understand you.
Interesting, so I never optimized for speed, so that can certainly be. Can you show the code you are using for copying?

FanHongchuang, 2017-11-06:
/* time format */
            SimpleDateFormat df = new SimpleDateFormat("yyyyMMddHHmmss");
            /* create a new directory */
            try {
                UsbFile dir = currentFs.getRootDirectory().createDirectory(df.format(new Date()));

                Iterator<Integer> iterator = positionSet.iterator();
                while(iterator.hasNext()) {
                    int i = iterator.next();

                    /* Create the file */
                    UsbFile file = dir.createFile(mDataList.get(i).file.getName());

                    /* Get the size of the file */
                    size = mDataList.get(i).file.length();

                    /* Get input stream from the file */
                    InputStream is = new BufferedInputStream(new FileInputStream(mDataList.get(i).file.getAbsolutePath()));

                    /* Get the output stream for new file */
                    OutputStream os = UsbFileStreamFactory.createBufferedOutputStream(file, currentFs);

                    byte[] bytes = new byte[8192];
                    int count;
                    long total = 0;
                    while ((count = is.read(bytes)) != -1){
                        /* write into new file */
                        os.write(bytes, 0, count);

                        total += count;
                        int progress = (int)total;
                        publishProgress(progress);
                    }

                    os.close();
                    is.close();

                    /* checkbox set false */
                    mDataList.get(i).isSelect = false;

                    /* remove */
                    iterator.remove();
                }

                runOnUiThread(new Runnable() {
                    @Override
                    public void run() {
                        mPathAdapter.notifyDataSetChanged();
                    }
                });
            } catch (IOException e) {
                e.printStackTrace();
            } 

### This is my code.
### Thank very much!

magnusja, 2017-11-06:
Can you try hint number 2 for me: https://github.com/magnusja/libaums#hints

FanHongchuang, 2017-11-07:
You are right!
I use UsbFile.setLength(long) before copy this file, the speed increase many.
Thank you very much.

magnusja, 2017-11-07:
Perfect! Glad I could help

closed 2017-11-07 by magnusja

jackyzhougithub, 2020-09-19:
I copy ordinary U disk files to the phone at a speed of 30M/s, but the speed of copying a SanDisk memory card is only 500KB/s. What could be the reason? thank you very much

Depau, 2020-09-19:
Maybe the SanDisk card is simply slower?

jackyzhougithub, 2020-09-19:
> Maybe the SanDisk card is simply slower?

This is probably the case. I tried multiple cards and one is slower, but the strange thing is that this card is very fast with ES File Explorer