magnusja/libaums #117: Error:Project :httpserver declares a dependency from configuration 'compile' to configuration 'default' which is not declared in the descriptor for project :libaums.

dcmcshan, 2017-10-25:
Apologies, if this is obvious. I'm not overly android savvy...

I get this error from within Android Studio.  Any advice on how to repair?  Thanks!

Error:Project :httpserver declares a dependency from configuration 'compile' to configuration 'default' which is not declared in the descriptor for project :libaums.

magnusja, 2017-10-25:
I have never seen this  are you trying the example app or are you including the libraries through Gradle in your project?

dcmcshan, 2017-10-25:
I literally just opened the app/build.gradle in Android Studio and it says “Gradle project sync failed. Basic functionality (e.g. editing, debugging) will not work properly”

magnusja, 2017-10-25:
Try to open the top level `build.gradle` which is for the whole project (all modules).

closed 2017-10-30 by magnusja