magnusja/libaums #118: OutOfMemoryError

XIONG-WEN, 2017-10-26:
My U disk has a size of 0 file.
When I deleted it, there was an error.

XIONG-WEN, 2017-10-26:
W/System.err: java.lang.OutOfMemoryError: Failed to allocate a 154589920 byte allocation with 4194304 free bytes and 119MB until OOM
W/System.err:     at java.util.ArrayList.add(ArrayList.java:118)
W/System.err:     at com.github.mjdev.libaums.fs.fat32.FAT.getChain(FAT.java:127)
W/System.err:     at com.github.mjdev.libaums.fs.fat32.ClusterChain.<init>(ClusterChain.java:64)
W/System.err:     at com.github.mjdev.libaums.fs.fat32.FatFile.initChain(FatFile.java:90)
W/System.err:     at com.github.mjdev.libaums.fs.fat32.FatFile.delete(FatFile.java:207)

XIONG-WEN, 2017-10-26:
FatFile--->delete()--->initChain()--->(entry.getFileSize() = 0 && entry.getStartCluster() = 39483)-->Fat.getChain(long startCluster)-->startCluster != 0

XIONG-WEN, 2017-10-26:
[image]
[image]

XIONG-WEN, 2017-10-26:
I change code FatFile->initChain():
	private void initChain() throws IOException {
		if (chain == null) {
			long startCluster = entry.getStartCluster();
			if (0 == entry.getFileSize() && 0 != startCluster) {
				startCluster = 0;
			}
			chain = new ClusterChain(startCluster, blockDevice, fat, bootSector);
		}
	}
it works，but I'm not sure if this is reasonable

magnusja, 2017-10-26:
Hello there, very interesting! Thanks for the feedback and the fix. Can you maybe create a Pull Request for this? Would highly appreciate this :)

magnusja, 2017-10-26:
Seems related to this: https://github.com/magnusja/libaums/issues/112#issuecomment-330234623

cumbie, 2017-11-09:
Just noticed this post.  So I tried this update, however it didn't resolve my issue.  My drive still appears corrupt on Windows after I do a delete().

closed 2019-09-09 by magnusja