magnusja/libaums #123: Usbfileman onCreate() is frozen

fgpzen, 2017-12-01:
Hey, I got some problems when using Usbfileman (example applicatoin):

1. At first, I connected Android phone with MicroSD card reader by using OTG cable (Figure 1-1), then started Usbfileman and authorized. In this case, Usbfileman started normally. Then I connected MP3, found that **[usbManager]** could be read, but a Log: **[no device found!]** would be printed, So I went to see the source code of **[MassStorageDevices = UsbMassStorageDevice.getMassStorageDevices (this)]**, and I found that the value of **[usbInterface.getInterfaceSubclass ()]** was 5 instead of 6, I didn’t know what did this value mean, so I tried to modify the IF statement, as shown in [Figure 1-2]. I restarted Usbfileman, then problem 2 came out.

Figure 1-1:
[image]
Figure 1-2:
[image]

Code position:
https://github.com/magnusja/libaums/blob/develop/libaums/src/main/java/com/github/mjdev/libaums/UsbMassStorageDevice.java#L138

2. I connected Android phone with MP3 by using OTG, then started Usbfileman and authorized. Usbfileman will be frozen for about 3 minutes before reading the MP3 file system (Figure 2-1,2-2,2-3), so there came problem 3. I found there was a MicroSD card slot on the MP3 (Figure 2-4), so inserting the MicroSD card (Figure 2-5), connecting the phone, authorizing. And Usbfileman started normally, but MicroSD card information couldn’t be read, there came problem 4.

Figure 2-1:
[image]
Figure 2-2,2-3 Android File Explorer:
[image]    [image]
Figure 2-4:
[image]
Figure 2-5:
[image]

Frozen code:
boolean claim = deviceConnection.claimInterface(usbInterface, true);

Frozen code position:
https://github.com/magnusja/libaums/blob/develop/libaums/src/main/java/com/github/mjdev/libaums/UsbMassStorageDevice.java#L222

3. I tried to use a card reader without MicroSD card to connect the phone (Figure 3-1), starting Usbfileman, Authorizing, then Usbfileman would be frozen.

Figure 3-1:
[image]

4. I inserted MicroSD to the card slot of MP3, but Usbfileman couldn’t read the MicroSD card information (Figure 4-1,4-2), while the system's file explorer could read it (Figure 4-3), printing out that Log **[massStorageDevices.length]** had a value of 1.

Figure 4-1:
[image]
Figure 4-2:
[image]
Figure 4-3:
[image]

Code:
massStorageDevices = UsbMassStorageDevice.getMassStorageDevices(this);

Code position:
https://github.com/magnusja/libaums/blob/develop/app/src/main/java/com/github/mjdev/libaums/usbfileman/MainActivity.java#L726

magnusja, 2017-12-29:
Hey there, 

sorry for the late response.

So in general changing, some constants is probably not a good idea, especially when you do not know what they mean.

According to this document (http://www.usblyzer.com/usb-mass-storage-class-msc-decoder.htm) a 6 means SFF-8070i. My library only supports SCSI transparent command set. 

That said, it is weird that it seems to work (at least partly). Unfortunately, I do not have time at the moment to look into this, so I would suggest that you do some research about SFF-8070i.

reardenlife, 2019-02-19:
Alright.  Did anyone figured out how to work with 8070i?