magnusja/libaums #124: Wrong CSW status

MaartenStottelaar, 2017-12-17:
Some USB devices seem to return an invalid CSW status, even when the operation apears to be correct. If I return TRUE in cases where csw.getbCswStatus== 1 in "transferCommand" (line 134) with these USB devices, the data  transfer seems to be OK. For now I ignore csw.getbCswStatus== 1, but like to solve this problem.

Maarten

magnusja, 2017-12-29:
Hey there,

and sorry for the late response.

Thanks for this information, I never had this problem, to be honest. Can you maybe elaborate a little which phone and pen drive you use? Is there a pattern?

MaartenStottelaar, 2017-12-31:
Dear Magnusja,

I am running the code on a Enybox 'Android box' with Android 6 (I tried several pieces). It seems that about 50% of the USB sticks do not work, but I have at least one official Sandisk 32G that works fine. I also noticed that (smaller) FAT formatted devices do not work, but that can of course eailly be solved.

Maarten

evernightking, 2018-02-02:
Thanks for solving my problem.

dandan8936, 2018-06-07:
Hi, I have the same problem now,  ScsiInquiryResponse [peripheralQualifier=0, peripheralDeviceType=0, removableMedia=false, spcVersion=2, responseDataFormat=2] java.io.IOException: Unsuccessful Csw status: 1

alVal6, 2018-07-20:
I get this error when i setupDevice() as in your **MainActivity** code with 2 usb stick, both fat 32 and mbr.
06:11:09.200 2229-2229/com.usb E/2: error setting up device java.io.IOException: **Unsuccessful Csw** status: 1 at com.github.mjdev.libaums.driver.scsi.ScsiBlockDevice.transferCommand(ScsiBlockDevice.java:185) at com.github.mjdev.libaums.driver.scsi.ScsiBlockDevice.init(ScsiBlockDevice.java:98) at com.github.mjdev.libaums.UsbMassStorageDevice.setupDevice(UsbMassStorageDevice.java:232) at com.github.mjdev.libaums.UsbMassStorageDevice.init(UsbMassStorageDevice.java:192) i always get same error with some usb sticks 

Please help

alVal6, 2018-07-20:
Hi!, i solved the problem, just need a delay before calling discover function and all usb sticks work.

NARUTONBM, 2018-08-03:
Hi there! I sleep the thread 1 second before i call `device.init();` and it works!

alVal6, 2018-08-03:
Nice!!! its great to hear it, you can try timer Countdown in order to avoid sleeping the thread

bealebe, 2018-10-25:
I'm still getting this. The 1 second sleep doesn't help.

alVal6, 2018-11-06:
Try with 3 seconds delay

[https://ci3.googleusercontent.com/proxy/GYeW-WfXrv_SHe0fBJj--7wGxKXhXu22PUhEGDHeEPu98HxeGrUyMeXjFyLdSYJTwwajU6av-6Y9Fw=s0-d-e1-ft#http://bizintek.es/logo-bizintek.png]

C/ Nemesio Mogrobejo 9A
48015 - Bilbao (Vizcaya)
Tel: +34 94 437 26 96
Fax: +34 94 434 47 05
www.bizintek.es<http://www.bizintek.es/>
Álvaro Valtierra De Pablo
Firmware Engineer

a.valtierra@bizintekinnova.com<mailto:m.viqueira@bizintekinnova.com>

[https://ci4.googleusercontent.com/proxy/iN-a4agKdd_Pwr0dNwJWrzAozdp2wO52fOLxMKG7ANaQPfRy5EglRL8zAbD_oS4=s0-d-e1-ft#http://bizintek.es/bg.png]