magnusja/libaums #125: Corrupted file writes / incorrect start cluster

alashow, 2018-01-06:
Hi,
I'm using libaums to copy a folder from devices internal storage to a flash drive. At first, I thought everything was okay after junk files bug (#37, #78) fix.
I can see files when I insert the drive to my MacBook, read them.
So I published the to play store for my client. Basically, the app allows the user to purchase and download design/pattern files to internal storage.
Afterwards, all design files can be copied to a USB flash drive. Then the user will insert that flash drive to a sewing machine (for ex. Janome 350e) to print those designs.

The problem is, the sewing machine can't read copied files by my app.
Files are readable on a MacBook, and if I open a design file with text editor and just save it, the machine will be able to read only that file.
Or, if I zip/compress a parent folder containing design files, and then just extract it, all design files will be readable by the machine.
I tried copying design files via [ES File Explorer](https://play.google.com/store/apps/details?id=com.estrongs.android.pop&hl=en), and it works okay too (files are readable by the machine).

Clearly, the problem was with libaums or the way I use it.

adb log: https://gist.github.com/anonymous/4bb981d4588e2eaca723b86caee130ee
My code to copy a folder recursively: https://gist.github.com/anonymous/a58bcb4614e8a2a21a8084f70dd0b15b

I used osx's `diskutil` to verify files after writing them, and here are the results:
```bash
> diskutil verifyVolume NO\ NAME
Started file system verification on disk2s1 NO NAME
Verifying file system
Volume was successfully unmounted
Performing fsck_msdos -n /dev/rdisk2s1
** /dev/rdisk2s1
** Phase 1 - Preparing FAT
** Phase 2 - Checking Directories
`..' entry in /embf5/m-317 has incorrect start cluster
Correct? no
`..' entry in /embf5/m-297 has incorrect start cluster
Correct? no
`..' entry in /emb/m-297 has incorrect start cluster
Correct? no
`..' entry in /emb/m-317 has incorrect start cluster
Correct? no
** Phase 3 - Checking for Orphan Clusters
117 files, 31318240 KiB free (978695 clusters)
File system check exit code is 8
Restoring the original state found as mounted
Error: -69845: File system verify or repair failed
Underlying error: 8: Exec format error
alashov at Alashovs-MacBook-Pro in /Volumes                                                                    × 3:43
> diskutil repairvolume NO\ NAME
Started file system repair on disk2s1 NO NAME
Repairing file system
Volume was successfully unmounted
Performing fsck_msdos -y /dev/rdisk2s1
** /dev/rdisk2s1
** Phase 1 - Preparing FAT
** Phase 2 - Checking Directories
`..' entry in /embf5/m-317 has incorrect start cluster
Correct? yes
`..' entry in /embf5/m-297 has incorrect start cluster
Correct? yes
`..' entry in /emb/m-297 has incorrect start cluster
Correct? yes
`..' entry in /emb/m-317 has incorrect start cluster
Correct? yes
** Phase 3 - Checking for Orphan Clusters
137 files, 31317568 KiB free (978674 clusters)

***** FILE SYSTEM WAS MODIFIED *****
File system check exit code is 0
Restoring the original state found as mounted
Finished file system repair on disk2s1 NO NAME
alashov at Alashovs-MacBook-Pro in /Volumes                                                                      3:43
> diskutil verifyVolume NO\ NAME
Started file system verification on disk2s1 NO NAME
Verifying file system
Volume was successfully unmounted
Performing fsck_msdos -n /dev/rdisk2s1
** /dev/rdisk2s1
** Phase 1 - Preparing FAT
** Phase 2 - Checking Directories
** Phase 3 - Checking for Orphan Clusters
130 files, 31317728 KiB free (978679 clusters)
File system check exit code is 0
Restoring the original state found as mounted
Finished file system verification on disk2s1 NO NAME
```

Note: Disk verification doesn't show errors when I copy files via ES File Explorer.

I thought that this was related to #112, so I tried not using `delete()` formatting/cleaning a flash drive and by disabling overwriting function from my code, didn't make any difference (verification fails).

What do you think the problem is?

Thanks.

magnusja, 2018-01-09:
Hey there,

so it looks like that the '..' entry in /embf5/m-317 has wrong start cluster. The '..' entry stands for the parent directory. I don't really use that in the library and probably OS X does not use it as well. But for a embedded device with limited memory I could imagine that this could come in quite handy to traverse to parent.

A quick look at the code reveals that I am setting that when creating a new directory: https://github.com/magnusja/libaums/blob/develop/libaums/src/main/java/com/github/mjdev/libaums/fs/fat32/FatDirectory.java#L403

I do not have a lot of time at the moment, so maybe you want to debug this a little. Did you try to just create a directory and then run this check?

alashow, 2018-01-09:
Yay! Good news.

I debugged little by little (don't know why I didn't try this before), by creating a simple directory, then writing string a to usbFile, then reading from internal storage and writing to usbFile.
Every step worked without verification errors.

Then it clicked me, I was writing files only to one level directories (i.e `/foo/bar.txt` ), but my folder structure has minimum 2 levels (`/foo/bar/file.txt`).
So I tried writing to `/sub/foo/bar.txt`, and the error was there.
```
** /dev/rdisk2s1
** Phase 1 - Preparing FAT
** Phase 2 - Checking Directories
`..' entry in /root/foo has incorrect start cluster
Correct? no
/sub/foo/bar.txt has too many clusters allocated (logical=0, physical=32768)
Drop superfluous clusters? no
** Phase 3 - Checking for Orphan Clusters
100 files, 31322464 KiB free (978827 clusters)
File system check exit code is 8
Restoring the original state found as mounted
Error: -69845: File system verify or repair failed
Underlying error: 8: Exec format error
```

alashow, 2018-01-09:
To make sure, I tried this:

<details>
  <summary>Java code</summary>
```java
FileSystem fs = selectedDevice.getPartitions().get(0).getFileSystem();
UsbFile root = fs.getRootDirectory();

{
    deleteIfExists(root, "foo");
    UsbFile foo = root.createDirectory("foo");
    deleteIfExists(foo, "bar.txt");
    UsbFile file = foo.createFile("bar.txt");
    // write to a file
    OutputStream os = new UsbFileOutputStream(file);
    os.write("hello".getBytes());
    os.close();
}
{
    deleteIfExists(root, "sub");
    UsbFile sub = root.createDirectory("sub");
    deleteIfExists(sub, "foo");
    UsbFile foo = sub.createDirectory("foo");
    deleteIfExists(foo, "bar.txt");
    UsbFile file = foo.createFile("bar.txt");
    // write to a file
    OutputStream os = new UsbFileOutputStream(file);
    os.write("hello".getBytes());
    os.close();
}
...

private void deleteIfExists(UsbFile parent, String child) throws IOException {
    UsbFile file = parent.search(child);
    if (file != null) {
        file.delete();
    }
}
```
</details>

<details>
```
<summary>Verification errors</summary>
Started file system verification on disk2s1 NO NAME
Verifying file system
Volume was successfully unmounted
Performing fsck_msdos -n /dev/rdisk2s1
** /dev/rdisk2s1
** Phase 1 - Preparing FAT
** Phase 2 - Checking Directories
`..' entry in /sub/foo has incorrect start cluster
Correct? no
/sub/foo/bar.txt has too many clusters allocated (logical=0, physical=32768)
Drop superfluous clusters? no
** Phase 3 - Checking for Orphan Clusters
88 files, 31322816 KiB free (978838 clusters)
File system check exit code is 8
Restoring the original state found as mounted
Error: -69845: File system verify or repair failed
Underlying error: 8: Exec format error
```
</details>

alashow, 2018-01-09:
Seems like the problems is with [`getStartCluster()`](https://github.com/magnusja/libaums/blob/develop/libaums/src/main/java/com/github/mjdev/libaums/fs/fat32/FatDirectoryEntry.java#L472-L496)

magnusja, 2018-01-09:
Hmm not sure about that. Because then I would assume that it also cannot read directories.

Did you check what the difference in the startCluster is when debugging? So '..' entry vs parent directory?

This error is new? `/sub/foo/bar.txt has too many clusters allocated (logical=0, physical=32768)`

alashow, 2018-01-09:
Do you mean comparing startCluster of `/foo`,`/sub` and `/sub/foo`? If yes, how do I get `FatDirectoryEntry` from UsbFile? or should I use breakpoint debugging>

Yes, I didn't that error before.

magnusja, 2018-01-09:
Yeah I meant breakpoint debugging.

You can also add some log statements if you have the libaums  code in your project. Otherwise reflection to get the private values but then breakpoints might be easier.

alashow, 2018-01-09:
I put a breakpoint [on this line](https://github.com/magnusja/libaums/blob/8ccaae4b4a2038fb1c034ecdc03f8fe1fee26c3b/libaums/src/main/java/com/github/mjdev/libaums/fs/fat32/FatDirectory.java#L404)

Looks like it's incrementing on each write.
```
foo: 5
sub: 7
foo: 8

foo: 10
sub: 12
foo: 13

foo: 15
sub: 17
foo: 18

foo: 20
sub: 22
foo: 23

foo: 25
sub: 27
foo: 28

foo: 30
sub: 32
foo: 33

foo: 35
sub: 37
foo: 38
```
(missing numbers must be `bar.txt`'s)

magnusja, 2018-01-09:
What if you do not delete and just overwrite?

Looks like the delete is rearranging the clusters a little.

alashow, 2018-01-09:
But the bug is still present in the first write?

alashow, 2018-01-09:
Weird stuff is happening now,
I pre created folders and files (`/foo/bar.txt`, `/bar/foo/bar.txt`). Then I run this code:
```java
 {
    UsbFile file = root.search("foo/bar.txt");
    // write to a file
    OutputStream os = new UsbFileOutputStream(file);
    os.write("hello".getBytes());
    os.close();
}
{
    UsbFile file = root.search("bar/foo/bar.txt");
    // write to a file
    OutputStream os = new UsbFileOutputStream(file);
    os.write("hello".getBytes());
    os.close();
}
```

First, there was an exception that I couldn't see (I think it was nullPointer, because on my laptop I created `sub` in root, but in code I wrote `bar`).
After setting a breakpoint to print an exception, I started seeing IOB exceptions from the first block. Lost that trace too (somehow it got cleared). But I remember this: Index: 0, Size: 0, something about cluster chain.

Then I formatted the drive and created those folders on ES Explorer (now `bar` so it matches with the code). No exceptions were thrown. I tried reading the file in ES FIle Explorer, it couldn't read so I inserted it on MacBook to check, and found this:
```
Started file system verification on disk2s1 NO NAME
Verifying file system
Volume was successfully unmounted
Performing fsck_msdos -n /dev/rdisk2s1
** /dev/rdisk2s1
** Phase 1 - Preparing FAT
** Phase 2 - Checking Directories
/bar/foo/bar.txt has no clusters
Truncate? no
size of /bar/foo/bar.txt is 5, should at most be 0
Truncate? no
** Phase 3 - Checking for Orphan Clusters
Found orphan cluster(s)
Fix? no
Found 1 orphaned clusters
89 files, 31322752 KiB free (978836 clusters)
File system check exit code is 8
Restoring the original state found as mounted
Error: -69845: File system verify or repair failed
Underlying error: 8: Exec format error
```

alashow, 2018-01-09:
Somehow that IOB exception got reproduced:
```ruby
java.lang.ArrayIndexOutOfBoundsException: length=0; index=0
	at com.github.mjdev.libaums.fs.fat32.ClusterChain.write(ClusterChain.java:160)
	at com.github.mjdev.libaums.fs.fat32.FatFile.write(FatFile.java:170)
	t com.github.mjdev.libaums.fs.UsbFileOutputStream.write(UsbFileOutputStream.java:65)
	at tm.app.FileActionsFragment.copyToSelectedDevice(FileActionsFragment.java:241)
	at tm.app.FileActionsFragment.access$000(FileActionsFragment.java:57)
	at tm.app.FileActionsFragment$1.onReceive(FileActionsFragment.java:76)
	at android.app.LoadedApk$ReceiverDispatcher$Args.lambda$-android_app_LoadedApk$ReceiverDispatcher$Args_52497(LoadedApk.java:1313)
	at android.app.-$Lambda$aS31cHIhRx41653CMnd4gZqshIQ.$m$7(Unknown Source:4)
	at android.app.-$Lambda$aS31cHIhRx41653CMnd4gZqshIQ.run(Unknown Source:39)
	at android.os.Handler.handleCallback(Handler.java:790)
	at android.os.Handler.dispatchMessage(Handler.java:99)
	at android.os.Looper.loop(Looper.java:164)
	at android.app.ActivityThread.main(ActivityThread.java:6494)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:438)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:807)
```

magnusja, 2018-01-09:
Ouch, looks like this could be some debugging.
I, unfortunately, do not have any time soon to have an in-depth look at this.

Is this IOB always reproducible with the steps above?

alashow, 2018-01-09:
No.. Now I can't even reproduce that.

For now, I will probably restructure folders to replace folders as file prefixes.

Hope you will get to the bottom of this, when you are free :)

Thanks for your help.

alashow, 2018-01-09:
Turns out, my `prefix instead of subfolder` won't work (those machines require 2lvl folders..).

Can you give me more clues for debugging, where to continue?

magnusja, 2018-01-09:
Okay so one problem I see is that the length of the file is not set properly when it is shrunk: https://github.com/magnusja/libaums/blob/develop/libaums/src/main/java/com/github/mjdev/libaums/fs/fat32/FatFile.java#L164

One workaround which should make your last example work is to set the length to 0.

```
{
    UsbFile file = root.search("foo/bar.txt");
    file.setLength(0);
    // write to a file
    OutputStream os = new UsbFileOutputStream(file);
    os.write("hello".getBytes());
    os.close();
}
{
    UsbFile file = root.search("bar/foo/bar.txt");
    file.setLength(0);
    // write to a file
    OutputStream os = new UsbFileOutputStream(file);
    os.write("hello".getBytes());
    os.close();
}
```

This should for sure fix the cluster allocation error in the verification tool. Maybe it also fixes the IOB?

alashow, 2018-01-09:
But in my original [folder copying code](https://gist.github.com/anonymous/a58bcb4614e8a2a21a8084f70dd0b15b#file-fragment-java-L94), I set the length from source file?

magnusja, 2018-01-09:
Sure, I think you just found a second bug :D
So I just wanted to make your code work, that we can investigate on the original bug.

alashow, 2018-01-09:
I see :D

So I run that code, weird stuff continues to happen:
```
Started file system verification on disk2s1 UsbStorage
Verifying file system
Volume was successfully unmounted
Performing fsck_msdos -n /dev/rdisk2s1
** /dev/rdisk2s1
** Phase 1 - Preparing FAT
** Phase 2 - Checking Directories
/sub/foo/bar.txt starts with free cluster
Truncate? no
/sub/foo/bar.txt: Cluster chain starting at 9 ends with cluster marked free
Truncate? no
/sub/foo/bar.txt has too many clusters allocated (logical=0, physical=32768)
Drop superfluous clusters? no
/foo/bar.txt starts with free cluster
Truncate? no
/foo/bar.txt: Cluster chain starting at 6 ends with cluster marked free
Truncate? no
** Phase 3 - Checking for Orphan Clusters
Found orphan cluster(s)
Fix? no
Found 2 orphaned clusters
Free space in FSInfo block (978838) not correct (978836)
Fix? no
88 files, 31322752 KiB free (978836 clusters)
File system check exit code is 8
Restoring the original state found as mounted
Error: -69845: File system verify or repair failed
Underlying error: 8: Exec format error
```

magnusja, 2018-01-09:
Hmmm :D

Can you try `setLength(1)` instead?

alashow, 2018-01-09:
I did, no errors on verification.
But what does it mean? `hello` string has more than 1 length (5 I guess), right?

magnusja, 2018-01-09:
Okay, that is good. Now you can go on with the original example you wanted to try to investigate on the original bug.

If you look into the write function (https://github.com/magnusja/libaums/blob/develop/libaums/src/main/java/com/github/mjdev/libaums/fs/fat32/FatFile.java#L164) you can see that the length will be automatically increased if needed. It is just not shrunk when needed. If you set the length to zero, the cluster chain is removed and the file deleted, which is not in sync with the directory entry. That is why you get the errors during verification.

alashow, 2018-01-09:
Thanks for the explanation.
So, what's next now? I still don't get why wrong clusters set on 2nd level folders.

magnusja, 2018-01-09:
Yes exactly me neither.

That is why you wanted to try this: https://github.com/magnusja/libaums/issues/125#issuecomment-356172746
But then we got interrupted by the other issue. So I would try this and have a look at the start clusters.

alashow, 2018-01-09:
So, I'm continuing with this code:
```java
{
    UsbFile folder = root.createDirectory("foo");
    UsbFile file = folder.createFile("bar.txt");
    // write to a file
    OutputStream os = new UsbFileOutputStream(file);
    os.write("hello".getBytes());
    os.close();
}
{
    UsbFile subFolder = root.createDirectory("sub");
    UsbFile folder = subFolder.createDirectory("foo");
    UsbFile file = folder.createFile("bar.txt");
    // write to a file
    OutputStream os = new UsbFileOutputStream(file);
    os.write("hello".getBytes());
    os.close();
}
```
As expected, verification errors shown:
```
Started file system verification on disk2s1 NO NAME
Verifying file system
Volume was successfully unmounted
Performing fsck_msdos -n /dev/rdisk2s1
** /dev/rdisk2s1
** Phase 1 - Preparing FAT
** Phase 2 - Checking Directories
`..' entry in /sub/foo has incorrect start cluster
Correct? no
/sub/foo/bar.txt has too many clusters allocated (logical=0, physical=32768)
Drop superfluous clusters? no
** Phase 3 - Checking for Orphan Clusters
86 files, 31322880 KiB free (978840 clusters)
File system check exit code is 8
Restoring the original state found as mounted
Error: -69845: File system verify or repair failed
Underlying error: 8: Exec format error
```

But, I didn't know that `/sub/foo/bar.txt` is empty (0 bytes), and `/foo/bar.txt` has `hello`.
I put breakpoints on FatDirectory ([line 404](https://github.com/magnusja/libaums/blob/develop/libaums/src/main/java/com/github/mjdev/libaums/fs/fat32/FatDirectory.java#L404), and [line 355](https://github.com/magnusja/libaums/blob/develop/libaums/src/main/java/com/github/mjdev/libaums/fs/fat32/FatDirectory.java#L355) @`createFile`) : 
```
foo: 5
bar.txt file: 6
sub: 7
foo: 8
bar.txt file: 9
```

Don't know why it starts from 5, for `/foo` but incrementing seemed okay, until I remembered that contents of files have also cluster numbers (after watching [this](https://youtu.be/HjVktRd35G8?t=332)). i.e if `/foo/bar.txt` has 5 bytes data (`hello`), what is its cluster number? if `bar.txt`'s data had cluster number, `bar`'s start cluster would be 8 (learned from). What do you think?

magnusja, 2018-01-10:
Can you change this line: https://github.com/magnusja/libaums/blob/develop/libaums/src/main/java/com/github/mjdev/libaums/fs/fat32/FatDirectory.java#L403

`entry.getStartCluster()` to `this.entry.getStartCluster()`

alashow, 2018-01-10:
Yay! Seems like it worked! Lemme check again with my original code..

alashow, 2018-01-10:
Yas! It's working, no verification errors now!
[image]

alashow, 2018-01-10:
Glad to help, thanks!

I guess that was because `/sub/foo` had a wrong cluster or something.

magnusja, 2018-01-10:
Lets hope it was, otherwise we have to keep looking :D

closed 2018-01-10 by magnusja

alashow, 2018-01-11:
Hey,

So my actual problem didn't go away, the sewing machines still can't read those files.
My app copies files to a drive, disk check shows no errors.
Say there are folders like this: `/foo/bar/file.jef`. Machines can see `foo` and `bar`, but when he (the client) opens `bar`, it's empty. But when he inserts the drive to a Windows machine, creates folder `/foo/qux `, and copies files from `/foo/bar/` to it, including `files.jef`, machines can see it.

Another weirder example, when he copies a folder from internal storage `/sdcard/app/foo` to a flash drive to root folder via ES File Explorer, machines can't see it. But after copying, if he does the same thing as in computer (creating `/foo/qux `, copying files from `/foo/bar`), machines can see it..

I even tried to simulate it (for each folder in `/foo` (`folder`) move files to create and move files `/foo/temp`, delete `folder`, rename `/foo/temp` to deleted `folder`'s name), and for the sake of my own sanity, it didn't work.

Do you have any ideas about this?

magnusja, 2018-01-12:
Very hard to say. 
If he copies a new file to `/foo/bar` can the machine see it then?

One thing my library does not do I think, is to update the backup FAT. Usually there are two FATs and one is the main/master one, I am only updating that one currently. But this is completely out of the blue. It is interesting that ES has that problem as well. Did you try Total Commander and the USB plugin? There is also another app called Nexus Media Importer which you could try.

alashow, 2018-01-12:
Yes, if he copies files from `/foo/bar` to a computer, and delete all files `/foo/bar`, then copy back copied files, machines can see it. Same as zipping and unzipping (as I described above).

Why do you think ES doesn't work when just copying, but works if you do new folder and copy thing?
I will ask him to try total commander and nexus media importer.

Is not updating backup FAT normal? and will it be hard to implement it?

magnusja, 2018-01-12:
So only when he deletes all the files? He cannot see newly copied ones if he leaves the files copied by libaums?

Well, that is very hard to say, I have no idea. As you say it is very strange. Maybe it has something to do with the directory names? Does it work when he is doing the `/foo/bar` using libaums, then the `/foo/qux` from computer and then a new folder within libaums?

No, it is not normal. Usually, you update both. But it is also not normal to rely on the wrong FAT. Have a look into FAT.java, it should be fairly easy to change that. You just need to put a for loop around the write and then write to the appropriate FAT offset.

What you can try to do is using a different FAT implementation. I have a module called java-fs based on the file system code from the JNode operating system (https://github.com/jnode/jnode).

Comment out this line: https://github.com/magnusja/libaums/blob/develop/libaums/src/main/java/com/github/mjdev/libaums/fs/FileSystemFactory.java#L47

And inlcude this line somewhere in the code BEFORE using libaums: https://github.com/magnusja/libaums/blob/develop/app/src/main/java/com/github/mjdev/libaums/usbfileman/MainActivity.java#L102

This will switch from my FAT32 implementation to the JNode on. But please note that one is highly experimental and probably worse performance wise.

alashow, 2018-01-15:
Sorry for late reply,

I did what you said, but this exception is thrown on any io operation:
```ruby
java.lang.RuntimeException: An error occurred while executing doInBackground()
   at android.os.AsyncTask$3.done(AsyncTask.java:353)
   at java.util.concurrent.FutureTask.finishCompletion(FutureTask.java:383)
   at java.util.concurrent.FutureTask.setException(FutureTask.java:252)
   at java.util.concurrent.FutureTask.run(FutureTask.java:271)
   at android.os.AsyncTask$SerialExecutor$1.run(AsyncTask.java:245)
   at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1162)
   at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:636)
   at java.lang.Thread.run(Thread.java:764)
Caused by: java.lang.NullPointerException: Attempt to invoke virtual method 'int org.jnode.fs.jfat.FatShortDirEntry.getIndex()' on a null object reference
   at org.jnode.fs.jfat.FatEntry.getId(FatEntry.java:109)
   at com.github.mjdev.libaums.javafs.wrapper.fs.UsbFileWrapper.isRoot(UsbFileWrapper.java:226)
   at com.github.mjdev.libaums.fs.AbstractUsbFile.search(AbstractUsbFile.java:18)
   at tm.app.ui.fragment.FileActionsFragment$CopyToUsbTask.copy(FileActionsFragment.java:402)
   at tm.app.ui.fragment.FileActionsFragment$CopyToUsbTask.doInBackground(FileActionsFragment.java:379)
   at tm.app.ui.fragment.FileActionsFragment$CopyToUsbTask.doInBackground(FileActionsFragment.java:325)
   at android.os.AsyncTask$2.call(AsyncTask.java:333)
   at java.util.concurrent.FutureTask.run(FutureTask.java:266)
   at android.os.AsyncTask$SerialExecutor$1.run(AsyncTask.java:245) 
   at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1162) 
   at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:636) 
   at java.lang.Thread.run(Thread.java:764) 
java.lang.RuntimeException: An error occurred while executing doInBackground()
      at android.os.AsyncTask$3.done(AsyncTask.java:353)
      at java.util.concurrent.FutureTask.finishCompletion(FutureTask.java:383)
      at java.util.concurrent.FutureTask.setException(FutureTask.java:252)
      at java.util.concurrent.FutureTask.run(FutureTask.java:271)
      at android.os.AsyncTask$SerialExecutor$1.run(AsyncTask.java:245)
      at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1162)
      at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:636)
      at java.lang.Thread.run(Thread.java:764)
   Caused by: java.lang.NullPointerException: Attempt to invoke virtual method 'int org.jnode.fs.jfat.FatShortDirEntry.getIndex()' on a null object reference
      at org.jnode.fs.jfat.FatEntry.getId(FatEntry.java:109)
      at com.github.mjdev.libaums.javafs.wrapper.fs.UsbFileWrapper.isRoot(UsbFileWrapper.java:226)
      at com.github.mjdev.libaums.fs.AbstractUsbFile.search(AbstractUsbFile.java:18)
      at tm.app.ui.fragment.FileActionsFragment$CopyToUsbTask.copy(FileActionsFragment.java:402)
      at tm.app.ui.fragment.FileActionsFragment$CopyToUsbTask.doInBackground(FileActionsFragment.java:379)
      at tm.app.ui.fragment.FileActionsFragment$CopyToUsbTask.doInBackground(FileActionsFragment.java:325)
      at android.os.AsyncTask$2.call(AsyncTask.java:333)
      at java.util.concurrent.FutureTask.run(FutureTask.java:266)
      at android.os.AsyncTask$SerialExecutor$1.run(AsyncTask.java:245) 
      at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1162) 
      at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:636) 
      at java.lang.Thread.run(Thread.java:764) 
```

magnusja, 2018-01-15:
Hmm can you comment otu this line: https://github.com/magnusja/libaums/blob/develop/javafs/src/main/java/com/github/magnusja/libaums/javafs/JavaFsFileSystemCreator.java#L50

alashow, 2018-01-15:
I tried, it threw IOB because of `device.getPartitions` would return an empty array.
Then I tried commenting out next line instead, and this exception was thrown:
```ruby
 Error copying folder: Attempt to invoke virtual method 'int org.jnode.fs.jfat.FatShortDirEntry.getIndex()' on a null object reference : java.lang.NullPointerException: Attempt to invoke virtual method 'int org.jnode.fs.jfat.FatShortDirEntry.getIndex()' on a null object reference
```

magnusja, 2018-01-15:
Hmm okay then something broke recently with java-fs, I need to look into that...

magnusja, 2018-01-16:
Hmm so for me this works, are you by any chance calling search on a file object instead of a directory?

alashow, 2018-01-16:
Don't think so, because my recursive code is [only called ](https://gist.github.com/anonymous/a58bcb4614e8a2a21a8084f70dd0b15b#file-fragment-java-L91) when source file [is a directory](https://gist.github.com/anonymous/a58bcb4614e8a2a21a8084f70dd0b15b#file-fragment-java-L80).

magnusja, 2018-01-16:
Then I would suggest that you debug
https://github.com/magnusja/libaums/blob/develop/javafs/src/main/java/com/github/magnusja/libaums/javafs/wrapper/fs/UsbFileWrapper.java#L35
Is it really null on that line?
Or here
https://github.com/magnusja/libaums/blob/develop/javafs/src/main/java/com/github/magnusja/libaums/javafs/wrapper/fs/UsbFileWrapper.java#L228

You also might want to try to use a freshly formatted device.

alashow, 2018-01-16:
I did try on freshly formatted device,
it wasn't null on both lines, but it was null [in here](https://github.com/jnode/jnode/blob/master/fs/src/fs/org/jnode/fs/jfat/FatEntry.java#L109).

magnusja, 2018-01-16:
Hmm that is bad.
So what is the value of the record here: https://github.com/jnode/jnode/blob/master/fs/src/fs/org/jnode/fs/jfat/FatEntry.java#L57 
Does it have a long name?

alashow, 2018-01-16:
no, `this.name` is an empty string (`""`).

magnusja, 2018-01-16:
So for me the entry gets set here: https://github.com/jnode/jnode/blob/master/fs/src/fs/org/jnode/fs/jfat/FatEntriesFactory.java#L74
and the name "" is correct.

It is a little weird, you probably have to debug a little bit why it is not set in your case.

magnusja, 2018-01-16:
Oh did you set a name for the device? Apparently it only gets set if there is a volume label, which is probably a bug :D

alashow, 2018-01-16:
well, it mounts as "NO NAME" :D lemme try

magnusja, 2018-01-16:
Yeah NO NAME is usually not volume label ;)

alashow, 2018-01-16:
I tried with a name set, the same exception.
Also, I tried setting a breakpoint [in here](https://github.com/jnode/jnode/blob/master/fs/src/fs/org/jnode/fs/jfat/FatEntriesFactory.java#L74). But it never gets there. And when I suspend in debugging, the stacktrace leads to `getRootDirectory` method, as if `search` function has nothing to do with it.

magnusja, 2018-01-16:
Okay, then try changing this line https://github.com/magnusja/libaums/blob/develop/javafs/src/main/java/com/github/magnusja/libaums/javafs/wrapper/fs/UsbFileWrapper.java#L228 to:
```
entry.getName().equals("")
```

alashow, 2018-01-16:
[It's true](https://youtu.be/pAekiKe1IWM?t=108)

alashow, 2018-01-16:
Oh, wait, did you mean returning that value? As an indication for `isRoot`.

alashow, 2018-01-16:
Looks like you did (I was just debugging before), so I changed that line to it, now this exception is thrown:
```ruby
Error copying folder: Could not read from device, result == -1 errno 1337 errno-lib could not be loaded! : java.io.IOException: Could not read from device, result == -1 errno 1337 errno-lib could not be loaded!
at com.github.mjdev.libaums.usb.JellyBeanMr2Communication.bulkInTransfer(JellyBeanMr2Communication.java:58)
at com.github.mjdev.libaums.driver.scsi.ScsiBlockDevice.transferCommand(ScsiBlockDevice.java:176)
at com.github.mjdev.libaums.driver.scsi.ScsiBlockDevice.write(ScsiBlockDevice.java:230)
at com.github.mjdev.libaums.driver.ByteBlockDevice.write(ByteBlockDevice.java:117)
at com.github.mjdev.libaums.javafs.wrapper.device.FSBlockDeviceWrapper.write(FSBlockDeviceWrapper.java:64)
at org.jnode.fs.jfat.FatCache$CacheElement.write(FatCache.java:406)
at org.jnode.fs.jfat.FatCache$CacheElement.flush(FatCache.java:422)
at org.jnode.fs.jfat.FatCache$CacheElement.access$1300(FatCache.java:349)
at org.jnode.fs.jfat.FatCache.flush(FatCache.java:187)
at org.jnode.fs.jfat.Fat.flush(Fat.java:247)
at org.jnode.fs.jfat.FatChain.allocateAndClear(FatChain.java:269)
at org.jnode.fs.jfat.FatDirectory.initialize(FatDirectory.java:66)
at org.jnode.fs.jfat.FatDirectory.addDirectory(FatDirectory.java:347)
at com.github.mjdev.libaums.javafs.wrapper.fs.UsbFileWrapper.createDirectory(UsbFileWrapper.java:200)
at tm.app.ui.fragment.FileActionsFragment$CopyToUsbTask.copy(FileActionsFragment.java:406)
at tm.app.ui.fragment.FileActionsFragment$CopyToUsbTask.doInBackground(FileActionsFragment.java:379)
at tm.app.ui.fragment.FileActionsFragment$CopyToUsbTask.doInBackground(FileActionsFragment.java:325)
at android.os.AsyncTask$2.call(AsyncTask.java:333)
at java.util.concurrent.FutureTask.run(FutureTask.java:266)
at android.os.AsyncTask$SerialExecutor$1.run(AsyncTask.java:245)
at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1162)
at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:636)
at java.lang.Thread.run(Thread.java:764)
```

magnusja, 2018-01-17:
Yeah, I actually get the same exception. That's bad. I guess Android does not like how this library writes to the USB drive. I did not implement any caching layer yet, so it can be that there are thousands of writes which could be merged into one.

There is another FAT32 library: https://github.com/waldheinz/fat32-lib

But I did not implement any support for that. You can try if you want. It should be quite easy, the API is quite similar and you can look how I implemented the wrappers (UsbFIleWrapper,BlockDeviceWrapper,FileSystemWrapper) in the javafs module. But I cannot guarantee that this library will do better, it originated from the JNode OS code as far as I know.

alashow, 2018-01-17:
I was doing okay until I got to `DeviceWrapper` and `JavaFsFileSystemCreator`. Can you help me with those?

magnusja, 2018-01-17:
For the `FileSystemCreator` you can probably use something like this: https://github.com/magnusja/libaums/blob/develop/libaums/src/main/java/com/github/mjdev/libaums/fs/fat32/Fat32FileSystemCreator.java
Just with your wrapped file system for the fat32-lib from waldheinz. 

For the block device you basically have to subclass this interface: https://github.com/waldheinz/fat32-lib/blob/master/src/main/java/de/waldheinz/fs/BlockDevice.java
This is what the fat32-lib uses to get raw data from a block device (the USB drive). What you do in this class is just delegate everything to an instance of my BlockDevice class (https://github.com/magnusja/libaums/blob/develop/javafs/src/main/java/com/github/magnusja/libaums/javafs/wrapper/device/FSBlockDeviceWrapper.java).

Then in the FIleSystemcreator you get the block device and can create an instance of your block device wrapper and that you give to the file system implementation in the fat32-lib. Then you return a wrapper around that file system so that it is compatible with my file system API.

For methods like read only or isClosed just return false.

alashow, 2018-01-17:
Mostly I was confused with implementing FileSystemCreator, then finally this line did it:
```java
new FileSystemWrapper(FatFileSystem.read(new FSBlockDeviceWrapper(blockDevice, entry), true));
```
It compiled, but it throws this exception:
[`java.io.IOException: suspicious sectors per cluster count 0`](https://github.com/waldheinz/fat32-lib/blob/3d8f1a986339573576e02eb0b6ea49bfdc9cce26/src/main/java/de/waldheinz/fs/fat/BootSector.java#L85)

magnusja, 2018-01-17:
Hmm so usually you need to check if that is actually a file system you are reading. I am doing that in the `read` method of my FAT32FileSystem. As a quick fix you can check if `PartitionTableEntry` is null then then just return null. This assumes that your drive is formatted with MBR, which is usually the case.

alashow, 2018-01-22:
Hello, sorry for the delayed response,

Unfortunately, it didn't work. It's throwing NPE-s on UsbFileWrapper's `getEntry`, or `Error copying folder: invalid cluster value 0 : java.lang.IllegalArgumentException: invalid cluster value 0`.

magnusja, 2018-01-23:
Hmm hard to say whats wrong. Do you have the code somewhere?