magnusja/libaums #129: connect camera UsbFile is null

pangxinlong, 2018-01-18:
I tried to connect the camera to get the photo，but UsbFile is null 

           device.init();//初始化
            //设备分区
            Partition partition = device.getPartitions().get(0);
            //文件系统
            FileSystem currentFs = partition.getFileSystem();
            currentFs.getVolumeLabel();//可以获取到设备的标识
            //通过FileSystem可以获取当前U盘的一些存储信息，包括剩余空间大小，容量等等
            Log.e("Capacity: ", currentFs.getCapacity() + "");
            Log.e("Occupied Space: ", currentFs.getOccupiedSpace() + "");
            Log.e("Free Space: ", currentFs.getFreeSpace() + "");
            Log.e("Chunk size: ", currentFs.getChunkSize() + "");
            cFolder = currentFs.getRootDirectory();//设置当前文件对象为根目录

magnusja, 2018-01-18:
Does the camera support the USB mass storage class?

pangxinlong, 2018-01-19:
I'm not sure, maybe ptp or mtp. Does this project only support usb mass strage?

magnusja, 2018-01-19:
Exactly it does! PRs are welcome though!

pangxinlong, 2018-01-19:
OK，Thank you very much！

closed 2018-01-19 by pangxinlong