magnusja/libaums #132: Missing permission to access usb device

fiasko131, 2018-02-05:
Hello,
I'm trying to initialize the device like this:
`UsbMassStorageDevice[] devices = UsbMassStorageDevice.getMassStorageDevices(getActivity());
		for(UsbMassStorageDevice device: devices) {
			// before interacting with a device you need to call init()!
			try {
				device.init();
				
			} catch (final Exception e) {
				e.printStackTrace();
			
			}
		}`

The logcat return:
java.lang.IllegalStateException: Missing permission to access usb device: UsbDevice[mName=/dev/bus/usb/001/002....

What kind of permission does it take to initialize the device?

Thanks

magnusja, 2018-02-06:
Closing this because not an actual issue.
Permissions are described in the readme, and there is a link to the official Android documentation.

closed 2018-02-06 by magnusja

fiasko131, 2018-02-06:
Thank you for your reply,

but in your readme you specify that you need permission to interact with the device but not to initialize it?

magnusja, 2018-02-07:
The library follows common convention and refers to `/` as root. Not sure why you would need to "get" the root path. The root directory is the main entry point for interacting with a file system.

fiasko131, 2018-02-14:
Sorry,
I wanted to talk about the mount point.
For example / storage / ....

magnusja, 2018-02-14:
This is not possible. Think about it. This is a third party library not a part of Androids system. So it is not possible to include the functionality directly in the system. You can do this rooting your device.

This library supports spanning a HTTP server or the storage framework to share data with other apps. Unfortunately you need to get a little creative here, and it might be not possible at all using this library depending on your use case. 

Lets move the discussion to https://gitter.im/libaums/Lobby, I do not want to pollute all the issues with other discussions.