magnusja/libaums #134: Canon Camera Data

luoshihai, 2018-02-10:
Hello, I need to read the Canon camera data, can give me a suggestion? I see the Android system photo album can read camera photos, but I do not know how to read the system album

magnusja, 2018-02-10:
This library is only supporting the USB mass storage protocol. If you would like to add other protocols which support cameras I am happy to take a PR :)

Not sure what you mean with "but I do not know how to read the system album". If the vanilla Android application shows the camera photos then you probably can find out in the code. Remember that Android is open source!

closed 2018-02-10 by magnusja

luoshihai, 2018-02-11:
I am in China, many foreign websites I can not visit, so I would like to find the source code of Android 7.0 is not an easy thing, what software do you chat in general, can I add you a friend? I would love to know the life abroad

magnusja, 2018-02-13:
Please do not hesitate to write me an email which can be found on my profile page/website.