magnusja/libaums #135: Error:Execution failed for task ':app:transformDexArchiveWithExternalLibsDexMergerForDebug'. > java.lang.RuntimeException: java.lang.RuntimeException: com.android.builder.dexing.DexArchiveMergerException: Unable to merge dex

umarfrq92, 2018-02-15:
PLEASE NOTE:
Before creating an issue, please think about wheter you are really reporting an issue or just have a question/problem. The latter are better suited for Stackoverflow or gitter (https://gitter.im/libaums). The best approach for a question or problem regarding libaums is to ask on Stackoverflow and hint me to the question via gitter!

If you are sure that you would like to open an issue, please delete all of this message and continue. 
Thank you!

magnusja, 2018-02-15:
Yep, I have that problem, too. Downgrading to Android Studio 2.x should work. I have not figured out yet what is wrong. So if you do please let me know.

irsen001, 2018-03-29:
in module storageprovider, file build.gradle, Line:38, delete org.apache.commons:commons-io:1.3.2, and IOUtiols in ParcelFileDescriptorUtil.java, it will build ok in Android Studio 3x

umarfrq92, 2018-03-30:
A failure occurred while executing com.android.build.gradle.internal.transforms.JacocoTransform$JacocoWorkerAction
in andriod studio 3.1

Metske, 2018-04-23:
Add `        classpath 'org.codehaus.groovy:groovy-all:2.4.12'
        classpath('com.dicedmelon.gradle:jacoco-android:0.1.2'){
            exclude group: 'org.codehaus.groovy', module: 'groovy-all'
        }`
to build.gradle (Project: libaums-develop)

closed 2018-08-14 by magnusja