magnusja/libaums #136: java.io.IOException: Could not write to device, result == -1 errno 2 No such file or directory

fiasko131, 2018-02-22:
Hello,

I am able to list the files at the root of my usb key with:
``
FileSystem currentFs = storageItem.usbMassStorageDevice.getPartitions().get(0).getFileSystem();
UsbFile root = currentFs.getRootDirectory();
UsbFile[] files = root.listFiles();
``

files ----> dirA , dirB, dirC, fileD, fileE etc...

But when I want to get the list of a directory that is at the root, 

UsbFile[] files = dirA.listFiles();

I get this error : 

> 02-22 01:56:04.981 18923-18923/com.techniwork.fastsharing W/System.err: java.io.IOException: Could not write to device, result == -1 errno 2 No such file or directory
>      at com.github.mjdev.libaums.usb.JellyBeanMr2Communication.bulkOutTransfer(JellyBeanMr2Communication.java:45)
>      at com.github.mjdev.libaums.driver.scsi.ScsiBlockDevice.transferCommand(ScsiBlockDevice.java:143)
>      at com.github.mjdev.libaums.driver.scsi.ScsiBlockDevice.read(ScsiBlockDevice.java:209)
> 	 at com.github.mjdev.libaums.partition.Partition.read(Partition.java:142)
>      at com.github.mjdev.libaums.fs.fat32.FAT.getChain(FAT.java:134)
>      at com.github.mjdev.libaums.fs.fat32.ClusterChain.<init>(ClusterChain.java:64)
>      at com.github.mjdev.libaums.fs.fat32.FatDirectory.init(FatDirectory.java:156)
>      at com.github.mjdev.libaums.fs.fat32.FatDirectory.listFiles(FatDirectory.java:485)
>      at com.techniwork.explorer.explorateur.ExplorateurActivity$4$1$1.onAnimationEnd(ExplorateurActivity.java:866)
>      at android.view.animation.Animation$3.run(Animation.java:376)
>      at android.os.Handler.handleCallback(Handler.java:739)
>      at android.os.Handler.dispatchMessage(Handler.java:95)
>      at android.os.Looper.loop(Looper.java:148)
>      at android.app.ActivityThread.main(ActivityThread.java:5417)
>      at java.lang.reflect.Method.invoke(Native Method)
>      at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:726)
>      at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:616)
> 
or at the second try:

> java.io.IOException: Could not write to device, result == -1 errno 16 Device or resource busy

etc..
tried on: galaxy s4 api 21, nexus 5 api 23, galaxy s7 api 24.

Any ideas?

Thanks

magnusja, 2018-02-22:
Hey there,

See: https://github.com/magnusja/libaums/issues/95

This is actually a known issue. Especially on Samsung devices. But so far no one came close to a conclusion to what this might cause. I occasionally have this but after replugging everything is fine. I do not own a Samsung device but apparently, it is more likely to happen on these devices. Maybe it is also newer Android versions? Not sure.

Unfortunately, I only own a Sony device and I am not really able to dig into this at the moment.

fiasko131, 2018-02-22:
Hi,

What created the problem for me, was when I invoked several times for the same device:
``FileSystem currentFs = storageItem.usbMassStorageDevice.getPartitions().get(0).getFileSystem(); ``
``UsbFile root = currentFs.getRootDirectory();``

So ,for a given device I register in a variable the UsbFile corresponding to the root.
The problem is solved for me for and  my three phones.
I hope it can help ...

**Out of issue**: I do not understand how to access an Intent.ACTION_VIEW from a UsbFile
I've added the library as shown, but I'm a little stuck to understand the procedure to follow.
A clarification would have been welcome :)

Thank you for your support and your work.

magnusja, 2018-02-22:
Hmm, interesting, these calls should not do any combination with the device because all information is cached.

Well, now you should be able to start an Intent and it shows the USB drive as an additional source like for instance Google Drive and you can open a file. This is most useful to allow third-party applications to access a file without any interaction from your app.