magnusja/libaums #137: How can i get Root Directory name for connected USB device ?

Jay268, 2018-03-19:
Hii @magnusja 

I am using this library for USB OTG device connection and i am displaying all the data inside app.

I need your help to get Root Directory name

say for E.G 

If i have images in my USB drive and when i click upon one of them i need to display it in bigger view but i am not be able to get path instead i am only getting name.

So what i am trying is to get path something like below which is used by ES File Explorer app while viewing the image from connected USB Drive.

usb://1002/UsbStorage/image.jpeg

I can able to get 1002 as it seems looking device id using below code.

`Toast.makeText(mContext, "getDeviceId : " + massStorageDevices[currentDevice].getUsbDevice().getDeviceId(), Toast.LENGTH_SHORT).show();`

 But i am not be able to get UsbStorage i mean root directory name dynamically.

Any help will be greatly appreciated.

I tried to get using `massStorageDevices[currentDevice].getPartitions().get(0).getFileSystem().getRootDirectory().getName()` but is is only showing me / in toast.

magnusja, 2018-03-19:
I do not know what you mean by root directory name, the root directory does not have a name attached to it. 
If you can get the device id you should be able to construct that name manually, right? If you need the volume label of the USB device, you can get that using `getVolumeLabel()`.

closed 2018-03-19 by magnusja

Jay268, 2018-03-20:
@magnusja I mean to say `usb://1002/UsbStorage/image.jpeg` what is UsbStorage here inside path used by ES File Explorer while viewing any image  , videos etc... from USB drive itself. Is that common for all ?

magnusja, 2018-03-20:
I do not know what ES File Explorer exactly does, unfortunately.  I guess this does not changed, so maybe it is just always prepended?

Jay268, 2018-03-20:
@magnusja No it isn't common for all. While i am changing the USB pen drive it is getting different name their. So i think it is dynamically concated in the path above. I am working around to get that.

magnusja, 2018-03-20:
And why would you need that?
Is it maybe the same you see when mounting the drive on your computer?

Jay268, 2018-03-20:
@magnusja I need it to display any image , videos etc... from USB drive itself. Currently what happens with your library is that when you click on any of the file it directly gets download to internal storage first and then it pops up the view to open file with default apps. 

But what i need is if i tap on image or videos etc.. we can able to view that file without download it to internal storage first. So for that i taken reference of an ES File Explorer they are using this kind of path to view the images or videos from USB drive.

Can you get my point what i am trying ?

magnusja, 2018-03-20:
This is not possible with libaums directly because it works in user space and is not able to create a mount point. You need to use the http module for that.

If ES File Explorer does that then it is very likely that your device supports mass storage devices by default and you do not need libaums to access them.

Jay268, 2018-03-20:
@magnusja Nope in my app i have used libaums because at start up i dont be able to get any USB drive detection by default be searching more i got libaums and i setup all requirement of my app based on that. I also not sure about what ES File Explorer is using.