magnusja/libaums #138: A failure occurred while executing com.android.build.gradle.internal.transforms.JacocoTransform$JacocoWorkerAction in andriod studio 3.1

umarfrq92, 2018-04-03:
PLEASE NOTE:
Before creating an issue, please think about wheter you are really reporting an issue or just have a question/problem. The latter are better suited for Stackoverflow or gitter (https://gitter.im/libaums). The best approach for a question or problem regarding libaums is to ask on Stackoverflow and hint me to the question via gitter!

If you are sure that you would like to open an issue, please delete all of this message and continue. 
Thank you!

Metske, 2018-04-24:
Fixed by adding  `classpath 'org.codehaus.groovy:groovy-all:2.4.12'` to build.gradle (Project: libaums-develop)

closed 2018-08-14 by magnusja

Ahmedbinyameen, 2020-02-05:
Hi, its not working.

debjeetdebnath, 2020-10-17:
I got this problem when I directly downloaded code files from GitHub but it was showing this error, but my colleague told me to use "Git bash here" and use the command to Gitclone it. After doing so it works fine.

PR #155 (2018-08-14): Fix Gradle build with latest build tools
Fixes a lot of issues with Gradle.

Closes #138