magnusja/libaums #143: How to get file absolute path ?

XtheWiz, 2018-05-15:
Hi, thank you for your library, it help me a lot !!
But one more thing, that my project would like to load mp3 file with MediaMetaDataRetriever that need Uri to file. 

Are there any method that I can get the path of particular file ?

Jay268, 2018-05-23:
@magnusja 

How could we go with the storageprovider or httpmodule for that ? Since i have applied required part from readme file.

My concern is how usbfile directly be open using Intent.ActionView ？ How could we get Mount path of connected Usb Pendrive and how can we open any of the image , video , audio directly from their instead if copying it to the internal storage first. I don't want to copy file to internal storage when USB OTG pendrive gets connected what are the way we can open image , video , audio directly.

Please help me for that Sir.

magnusja, 2018-05-23:
There is no way of getting the mount path because it is not mounted under Android. 

Please use the http module to stream videos, audio, and images to another app.

Jay268, 2018-05-24:
@magnusja 

So any workaround to open UsbFile directly through Intent.ActionView ? I tried with it but it every time showing media not found on click of UsbFile adapter item. I tried to view an image. 

So i need your help in that regards. I also added httpmodule but i doesn't get idea how can i start with View image , video , audio etc.. 

So any help will be greatly appreciated in this.

If you need to look at  code snippet that i am trying i will provide that also sir.

magnusja, 2018-05-24:
Please have a look into the example application, it is all implemented there.

Jay268, 2018-05-24:
@magnusja 

I tried with  whatever implemented in example application. But when inside example application itself going to open an image it is asking to open in any browser and then something like localhost:8000/Image.png loading and it's get stuck showing nothing in example application itself.

What need to do in that case ?

magnusja, 2018-05-24:
Yeah for images the standard external application is usually the browser.

In the example application, I do not show anything I am only opening third-party apps. If you want to show that in your application you need to pass the URL from the http module to an ImageView or video player widget.