magnusja/libaums #144: Delete failing with javafs (fat16)

Metske, 2018-05-16:
Hi, first I would like to say how great the library is! It has helped me so much!

I have reading from a fat16 device working really well, but I am stuck on deleting files and then writing.

I understand that the javafs integration is experimental at the moment but I cannot figure out why the delete function is not working. I am getting this info from the log:

W/System.err: java.io.IOException: free entry in chain at: 31254
        at org.jnode.fs.jfat.FatChain$ChainIterator.next(FatChain.java:711)
        at org.jnode.fs.jfat.FatChain.freeAllClusters(FatChain.java:323)
        at org.jnode.fs.jfat.FatEntry.freeAllClusters(FatEntry.java:96)
        at org.jnode.fs.jfat.FatDirectory.remove(FatDirectory.java:363)
W/System.err:     at com.github.magnusja.libaums.javafs.wrapper.fs.UsbFileWrapper.delete(UsbFileWrapper.java:222)

Any help would be very much appreciated!!

Metske, 2018-05-17:
I tried commenting out "fsTypes.add(new FatFileSystemType());" and .listFiles(); seemed to stop working beyond the root directory.

"Caused by: java.lang.NullPointerException: Attempt to get length of null array"

I commented out "fsTypes.add(new org.jnode.fs.fat.FatFileSystemType());" instead and got:

W/System.err: java.io.IOException: circular chain at: 0
W/System.err:     at org.jnode.fs.jfat.FatChain$ChainIterator.next(FatChain.java:708)
        at org.jnode.fs.jfat.FatChain.freeAllClusters(FatChain.java:323)
        at org.jnode.fs.jfat.FatEntry.freeAllClusters(FatEntry.java:96)
        at org.jnode.fs.jfat.FatDirectory.remove(FatDirectory.java:363)
        at com.github.magnusja.libaums.javafs.wrapper.fs.UsbFileWrapper.delete(UsbFileWrapper.java:222)

Thank you so much for your help and time!

magnusja, 2018-05-23:
Well, that's bad. I really don't know whats wrong here.

You could try to update https://github.com/magnusja/java-fs with recent changes from https://github.com/tmyroadctfig/jnode and see if it works then.

Metske, 2018-05-29:
Hi, Yeah I forked it and committed the changes not sure exactly how to link up with these changes though. The javafs module in the Libuams development library has "com.github.magnusja:java-fs:0.1.3" as a dependency. ( "https://github.com/magnusja/java-fs" is version 0.1.2 ?)

I tried importing my fork in as a module and refactoring but I got into a whole load of compiling issues.

Sorry still fairly new to github and not 100% sure what to do.

Metske, 2018-05-29:
Used jitpack and added my fork as the dependency, still getting a lot of "W/System.err: java.io.IOException: circular chain at: 0" errors!

magnusja, 2018-06-03:
What happens if you remove the comment you added, and use the other FAT32 implementation? Maybe there are some recent fixes.

Metske, 2018-06-20:
Finally had some time to sit down and look at this again, I still haven't managed to get anywhere.
One time I tried and it started working a bit and it was writing fine up until the last file then it crashed with "Free entry in chain at: 1" I haven't been able to reproduce this again. I tried with the other implementation but it won't initialize due to UnixFSAccessRights issues.

One thing I don't think I've mentioned is I get this when initializing:

I/ScsiBlockDevice: Block size: 512
I/ScsiBlockDevice: Last block address: 786431
I/FileSystemPartitionTable: Found a device without partition table, yay!
W/FileSystemPartitionTable: fs capacity is not multiple of block size

magnusja, 2023-03-11:
Should be fixed in v0.9.4

closed 2023-03-11 by magnusja