magnusja/libaums #148: When you write to the USB flash drive, the file attribute shows N

akiyev, 2018-07-10:
Hello. When you write to the USB flash drive, the file attribute shows N as in the [photo](https://drive.google.com/open?id=10B8vQ8Skq00mIHQ-AFMiRgwuJfSFvkdc). How can I change it to attribute A or remove it?

magnusja, 2018-07-11:
Are you sure that this is a feature of the file system and not Windows itself?

https://superuser.com/questions/516709/what-is-the-n-folder-attribute-on-windows

akiyev, 2018-07-11:
When I write files using your library in files there is an attribute N. When I use the usual file explorer in android then it has the attribute A.
Overall I have an application that writes pictures to the USB storage (.jef format) for the cars of JANOME. When I write pictures using your library the machine does not see the files. If I copy this file in android by the usual file explorer, then I see it. I think the difference in the attributes of N and A.

magnusja, 2018-07-25:
Hmm ok. I don't think the file attribute is the problem here. More of a problem is that files cannot be seen. Does that happen with every smartphone/pen drive you tried?

akiyev, 2018-09-04:
Grieve the problem in this. Files with the H attribute

akiyev, 2018-09-04:
H->N

tommydawu, 2018-10-24:
we have same problem.When I copy a file to usb using your library the machine does not see the files. If I copy this file in android by the usual file explorer, then I see it. I think the difference in the attributes of N and A.

magnusja, 2018-10-24:
Have a look here: https://github.com/magnusja/libaums/issues/158#issuecomment-432606062

closed 2019-12-09 by magnusja