magnusja/libaums #149: Copying to Sd card performance

liaoxsong, 2018-07-18:
Is there a way to copying file faster to Sd card? I used BufferedInputStream with the length being the chunk size of the flash drive, it takes forever to write a big file ( >=400mb).

liaoxsong, 2018-07-18:
private int copyFile(UsbFile fromFile, File toFile, int chunksize) {
        if (toFile == null) return 0;//dont make a copy for a folder itself

        BufferedInputStream fosfrom = null;//using bufferedinputstream for performance
        BufferedOutputStream fosto = null;
        try {
            
            fosfrom = new BufferedInputStream(new UsbFileInputStream(fromFile), chunksize);
            fosto = new BufferedOutputStream(new FileOutputStream(toFile), chunksize);
            //UsbFileStreamFactory.createBufferedOutputStream()

            byte buffer[] = new byte[chunksize];
            int c;
            int total = 0;

            while ((c = fosfrom.read(buffer)) > 0) {
                fosto.write(buffer, 0, c);
                total+=c;

                String tcopied = String.format("%.1f", total/1024f/1024f);
                String percent = String.format("%.2f", 100*total/(double) fromFile.getLength());
                publishProgress("Copied:" + tcopied + "mb(" + percent + "%)");
            }
            fosfrom.close();
            fosto.close();
            return 1;
        } catch (Exception ex) {
            LogUtil.e(TAG, "copying error:" + ex.getMessage());
            return 0;
        }
    }

magnusja, 2018-07-18:
Yes, the library is not really performance optimized.

Did you try different devices (phone and USB pen drive)? Do you get any warnings? Did you test the performance using Total Commander or some other application which supports USB?

liaoxsong, 2018-07-18:
I tried using an Android 8.0 device and a Android 5.1 device, the 8.0 device is 10 times faster..

liaoxsong, 2018-07-18:
I just tried Total Commander on my Android 8.0 it takes roughly 3 to 4 times faster than this library.
I think i'm using the default FAT32.

magnusja, 2018-07-25:
Ok, I see. Unfortunately, I do not have any time to look into this, but you probably want to profile which function takes most of the time.

Do you get any warnings?

ljliu1985, 2019-04-15:
> Ok, I see. Unfortunately, I do not have any time to look into this, but you probably want to profile which function takes most of the time.
> 
> Do you get any warnings?

If the directory has many files, 4000 or more， It takes too much time to close the input stream

liaoxsong, 2020-01-18:
actually... I didn't have a lot of files back then, it was just a huge file about 500mb, later I found out it was problem with this old device. I tried other usb apps they were slow too on this old device. The performance on newer android devices are actually good :)

closed 2020-01-18 by liaoxsong