magnusja/libaums #152: ByteBlockDevice.getBlockSize() should return 1

Depau, 2018-08-13:
From `BlockDeviceDriver.java`:
```java
	/**
	 * Returns the block size of the block device. Every block device can only
	 * read and store bytes in a specific block with a certain size.
	 * <p>
	 * That means that it is only possible to read or write hole blocks!
	 * 
	 * @return The block size in bytes, mostly 512 bytes.
	 */
	public int getBlockSize();
```

From `ByteBlockDevice.java`:
```
Simple class which wraps around an existing {@link BlockDeviceDriver} to enable byte addressing of content
```

According to this information, in my opinion, `ByteBlockDevice.getBlockSize()` should return 1.

magnusja, 2018-08-13:
I guess for logical reasons you are right. But this change might decrease performance because some parts of the code use this block size to allocate buffers to read and write. If there is only one byte written at once performance decreases.

Also, this is only a wrapper around a real block device, so it makes sense to delegate this call to the real block device.

Let me know what you think.

Depau, 2018-08-13:
That makes sense too.
I'll check the code that uses this class later.

(BTW I'm working on a pr to fix the build issues, Gradle is being a bit of a pain)

closed 2019-09-09 by magnusja