magnusja/libaums #158: Make it possible to add file attributes

akiyev, 2018-09-04:
Make it possible to add file attributes
R  Read-only (1)
A  Archive (32)
S  System (4)
H  Hidden (2)

magnusja, 2018-10-24:
https://github.com/magnusja/libaums/blob/develop/libaums/src/main/java/com/github/mjdev/libaums/fs/fat32/FatDirectoryEntry.java#L199-L216

magnusja, 2018-10-24:
Maybe try to set the flag here for new files: https://github.com/magnusja/libaums/blob/develop/libaums/src/main/java/com/github/mjdev/libaums/fs/fat32/FatDirectory.java#L352

Then you can check if this flag is really the problem

akiyev, 2018-10-24:
> Maybe try to set the flag here for new files: https://github.com/magnusja/libaums/blob/develop/libaums/src/main/java/com/github/mjdev/libaums/fs/fat32/FatDirectory.java#L352
> 
> Then you can check if this flag is really the problem

OK

magnusja, 2018-10-24:
https://github.com/magnusja/libaums/blob/develop/libaums/src/main/java/com/github/mjdev/libaums/fs/fat32/FatDirectoryEntry.java#L194-L197

tommydawu, 2018-10-24:
Thank you, I think this should be what I want.  for some reason，My project can't include this libamus source code. I know this is a bit too much.....But ....Can you update a new version with a default attribute of A ?

magnusja, 2018-10-25:
You can just download the code and include it in your project directly.

tommydawu, 2018-10-25:
FatDirectoryEntry can setFlag .....but  FatLfnDirectoryEntry can not !