magnusja/libaums #162: Creating multiple files will get slower and slower

angelcql, 2018-11-08:
UsbFile rootFolder = currentFs.getRootDirectory();
for (int i = 0 ; i < 1000 ; i ++) {
       UsbFile file = rootFolder.createFile(i + ".txt");
}

As more files are created, new files are created more slowly

magnusja, 2018-11-09:
Hmm not sure what could cause this problem.

Maybe you wanna profile the following method: https://github.com/magnusja/libaums/blob/develop/libaums/src/main/java/com/github/mjdev/libaums/fs/fat32/FAT.java#L159

Could be that the last allocated cluster hint is wrong for some reason and searching for a free cluster needs far more iterations the more files you create

angelcql, 2018-11-10:
we tested plenty ways, but if our “being written documents” are over 300, the duration of creating a document would prolong immediately (every document need 3 to 7 seconds to create.). We can not understand the program completely, may you help us to optimize the problem? Thanks a lot.

angelcql, 2018-11-10:
we tested plenty ways, but if our “being create files ”are over 300, the duration of creating a file would prolong immediately (every file need 3 to 7 seconds to create.). We can not understand the code and fat32 completely, may you help us to optimize the problem? Thanks a lot.

angelcql, 2018-11-10:
Through our debugging process, we found if we use “readEntries” when we createFile in “Init()”, the program would read all the files under the entry. It causes createFile process gets slower and slower. Could you tell us how to optimize the program? We tried that if creating an entry for every single file, the performance of the program would not be effected.

magnusja, 2018-11-12:
That is weird, are you sure?
init() should only be called if it has not been called already: https://github.com/magnusja/libaums/blob/develop/libaums/src/main/java/com/github/mjdev/libaums/fs/fat32/FatDirectory.java#L168-L170

But it could be that writing the entries takes longer and longer because all the entries are written every time. Changing this into a smarter write procedure would need some redesign though, there is probably no quick fix

gxshao, 2019-07-16:
i met same issue, i think here is too many files in same folder will effect `createFile` speed. because before create file, it has to check item exists...

magnusja, 2023-03-11:
Closing in favor of #162

closed 2023-03-11 by magnusja