magnusja/libaums #167: Use third party to play video through uri?

wbjaa, 2019-02-21:
Hi all,
 I want to play Big video file(3GB) through third party app(ex MXPlayer),
if I use http server method, it will open browser to download file, and it costs too much time.
So I wonder if there is a way, I can convert usbfile to uri directly, much thx.

wbjaa, 2019-02-21:
Or can I have usbfileman apk? I can't build on my android studio, thx.

magnusja, 2019-02-21:
Yes this is exactly what the http server is for. Usually you can choose which application shall open if there are multiple which can handle the URI. Maybe look into the settings if you accidentally chose to always open the browser.
Maybe show some code, and explain why you cannot build the example app?
In addition, browser should be able to stream as well? Maybe the file name does not end with an extension related to a video file? Yeah all in all hard to say, you probably want to provide some further details otherwise I can just guess.

wbjaa, 2019-02-22:
Thank you for reply.
It just return "SIMPLE: Error configuring"
Gradle(app)
android {
    compileSdkVersion 28
    buildToolsVersion '27.0.3'

    defaultConfig {
        applicationId "com.github.mjdev.usbfileman"
        minSdkVersion 16
        targetSdkVersion 28
        versionCode 1
        versionName "1.0"
    }
    buildTypes {
        release {
            minifyEnabled false
            proguardFiles getDefaultProguardFile('proguard-android.txt'), 'proguard-rules.pro'
        }
    }

    lintOptions {
        abortOnError false
    }
}

Gradle(Android)
buildscript {
    repositories {
        jcenter()
        maven {
            url 'https://dl.bintray.com/magnusja/maven'
        }
        maven {
            url 'https://maven.google.com/'
            name 'Google'
        }
    }
    dependencies {
        classpath 'com.android.tools.build:gradle:3.1.4'

        classpath 'com.jfrog.bintray.gradle:gradle-bintray-plugin:1.7.3'
        classpath 'com.github.dcendents:android-maven-gradle-plugin:1.5'
        classpath 'com.dicedmelon.gradle:jacoco-android:0.1.1'

        classpath 'org.codehaus.groovy:groovy-all:2.4.12'

        // NOTE: Do not place your application dependencies here; they belong
        // in the individual module build.gradle files
    }
}

So I don't know why I cannot build in my platform.

When I use HttpServer method(os 8.1), it just crash at ActivityThread.java ,
and log is below...
Fatal Exception: android.app.RemoteServiceException
Bad notification for startForeground: java.lang.RuntimeException: invalid channel for service notification: Notification(channel=null pri=0 contentView=null vibrate=null sound=null defaults=0x0 flags=0x40 color=0x00000000 vis=PRIVATE)

android.app.ActivityThread$H.handleMessage (ActivityThread.java:1780)
android.os.Looper.loop (Looper.java:172)
android.app.ActivityThread.main (ActivityThread.java:6590)
java.lang.reflect.Method.invoke (Method.java)
com.android.internal.os.ZygoteInit.main (ZygoteInit.java:807)

thank you for your help, thx.

magnusja, 2019-02-22:
Are you sing Android >= 8.1?

https://stackoverflow.com/questions/47531742/startforeground-fail-after-upgrade-to-android-8-1

magnusja, 2019-02-22:
Please try https://github.com/magnusja/libaums/tree/feature/fix-service-android-O

I, unfortunately, do not have any device with 8.1 to test this.

closed 2019-02-25 by magnusja

wbjaa, 2019-02-27:
Thank you for all the work u did.