magnusja/libaums #175: java.io.IOException: Could not write to device, result == -1 errno 88 Socket operation on non-socket

MrLiGitHub, 2019-03-07:
I meet this question that the USB drive is out when Android Device is copying.So it's a good tips to hold the connection.

DoubleZhangbo, 2019-03-30:
Hi author, i also meet a similar problem. when my app to send "read/write"code (0x28/0x2a), the function(deviceConnection.bulkTransfer) return error code -1 and it shows "IO exception". and then it always fail. but when i send other code (eg. Inquiry0x12), it is normal. look forward you reply.

shenqinwei, 2019-10-16:
每次断开连接重新插上都要new UsbFile，这样才能解决这个问题。

magnusja, 2020-01-08:
Please try latest version

magnusja, 2020-01-13:
closing in favor of #237

Depau, 2020-01-13:
> errno 88 Socket operation on non-socket

This usually happens when you disconnect the USB drive while it's working. Maybe get a new USB-OTG adapter.

I'd close this.

magnusja, 2020-01-14:
Oh okay, I did not know that, will close :) thanks!

closed 2020-01-14 by magnusja