magnusja/libaums #176: unable to read USB filesystem if there is no partition table available

ajayane, 2019-03-16:
### Problem: unable to read USB filesystem if there is no partition table available 

### Expected behaviour: when there is no partition table available, the whole disk has to be considered as single disk partition.

### Actual behaviour: UsbMassStorageDevice.getPartitions() return empty list.

#### Stacktrace of Excpetion

```
(if exists)
```

### Code where the problem occurs :

magnusja, 2019-03-16:
This is implemented https://github.com/magnusja/libaums/pull/96

ajayane, 2019-03-18:
The problem was in MasterBootRecordCreator.read where you allocate a 512 buffer size and our disk block size was 4096. If you can set the buffer size to blockDevice.getBlockSize(), this will work like a charm!!. but dont know will it harm other devices... Anyways thanks for the fast reply.

magnusja, 2019-03-20:
You mean like https://github.com/magnusja/libaums/pull/177?

ajayane, 2019-03-20:
Yes!.Exactly. But I couldn't find the code fix in the pull..

magnusja, 2019-03-20:
What do you mean in th pull? This is an open PR so it is not merged to develop yet.

magnusja, 2019-03-20:
What kind of USB drive are you using? I have never seen a block size of 4096 and I really wonder what the problem is because the `ByteBlockDevice` reads 4096 bytes and discards the rest, so the MBR buffer should be OK.

There is now exception or anything?

ajayane, 2019-03-20:
Okay. Alright. Got it. It is an accessory with a block size of 4096 and FAT32 FS.

ajayane, 2019-03-20:
There was an exception when I changed the code and tested it worked well, If you can merge #177 to develop, it will be a great help.

magnusja, 2019-03-20:
Okay, can you maybe post the exception and stack trace here. I am curious why this happens.

I will merge when I verified it does not have any unexpected side effects.

Just checkout the branch fix/mbr-buffer-size for now

closed 2019-12-09 by magnusja