magnusja/libaums #182: When there are too many catalog files, it takes more time to shut down the IO stream.

ljliu1985, 2019-04-15:
### Problem
There are thousands of pictures under the U disk directory. When copy the files to SD Card, close the IO stream time too much.

### Expected behavior

### Actual behavior

#### Stacktrace of Excpetion

```
(if exists)
```

### Code where problem occurs
   for (int i = 0; i < files.length; i++) {
                        long t = SystemClock.elapsedRealtime();
                        if (!files[i].isDirectory()) {
                            InputStream is = null;
                            OutputStream uos = null;
                            try {
                                File targetFile = new File(Constant.PHOTOPATH, files[i].getName());
                                if (targetFile.exists()) {
                                    targetFile.delete();
                                }
                                is = new UsbFileInputStream(files[i]);
                                uos = new BufferedOutputStream(new FileOutputStream(targetFile));
                                int bytesRead;
                                byte[] buffer = new byte[1024];
                                Log.i("tag1", "copy index:" + i + ",file name:" + files[i].getName() + ",file size:" + Formatter.formatFileSize(mContext, files[i].getLength()));
                                while ((bytesRead = is.read(buffer)) != -1) {
                                    uos.write(buffer, 0, bytesRead);
                                }
                                uos.flush();
                                Log.i("tag2", "dtime" + (SystemClock.elapsedRealtime() - t));
                            } catch (Exception e) {
                                failCount++;
                                e.printStackTrace();
                            } finally {
                                Log.i("tag3", "dtime" + (SystemClock.elapsedRealtime() - t));
                                try {
                                    if (null != is) {
                                        is.close();
                                    }
                                } catch (IOException e) {
                                } finally {
                                    is = null;
                                }
                                Log.i("tag4", "dtime" + (SystemClock.elapsedRealtime() - t));
                                try {
                                    if (null != uos) {
                                        uos.close();
                                    }
                                } catch (IOException e) {
                                } finally {
                                    uos = null;
                                }
                                Log.i("tag5", "dtime" + (SystemClock.elapsedRealtime() - t));
                            }
                        }
                    }

-----------------------log
04-15 11:34:01.272 7598-15261/com.lingkang.sport I/tag: files length:2931
04-15 11:34:01.277 7598-15261/com.lingkang.sport I/tag1: copy index:0,file name:2014200012.jpg,file size:85.46 KB
04-15 11:34:01.354 7598-15261/com.lingkang.sport I/tag2: dtime81
04-15 11:34:01.355 7598-15261/com.lingkang.sport I/tag3: dtime83
04-15 11:34:01.531 7598-15261/com.lingkang.sport I/tag4: dtime258
04-15 11:34:01.532 7598-15261/com.lingkang.sport I/tag5: dtime260
04-15 11:34:01.535 7598-15261/com.lingkang.sport I/tag1: copy index:1,file name:2014200104.jpg,file size:30.96 KB
04-15 11:34:01.563 7598-15261/com.lingkang.sport I/tag2: dtime31
04-15 11:34:01.563 7598-15261/com.lingkang.sport I/tag3: dtime31
04-15 11:34:01.690 7598-15261/com.lingkang.sport I/tag4: dtime158
04-15 11:34:01.690 7598-15261/com.lingkang.sport I/tag5: dtime158
04-15 11:34:01.692 7598-15261/com.lingkang.sport I/tag1: copy index:2,file name:2014200127.jpg,file size:59.96 KB
04-15 11:34:01.753 7598-15261/com.lingkang.sport I/tag2: dtime62
04-15 11:34:01.753 7598-15261/com.lingkang.sport I/tag3: dtime62
04-15 11:34:01.888 7598-15261/com.lingkang.sport I/tag4: dtime198
04-15 11:34:01.888 7598-15261/com.lingkang.sport I/tag5: dtime198
04-15 11:34:01.893 7598-15261/com.lingkang.sport I/tag1: copy index:3,file name:2014200177.jpg,file size:30.61 KB
04-15 11:34:01.925 7598-15261/com.lingkang.sport I/tag2: dtime37
04-15 11:34:01.925 7598-15261/com.lingkang.sport I/tag3: dtime37
04-15 11:34:02.068 7598-15261/com.lingkang.sport I/tag4: dtime179
04-15 11:34:02.068 7598-15261/com.lingkang.sport I/tag5: dtime180
04-15 11:34:02.073 7598-15261/com.lingkang.sport I/tag1: copy index:4,file name:2014200182.jpg,file size:28.69 KB
04-15 11:34:02.112 7598-15261/com.lingkang.sport I/tag2: dtime44
04-15 11:34:02.112 7598-15261/com.lingkang.sport I/tag3: dtime44
04-15 11:34:02.257 7598-15261/com.lingkang.sport I/tag4: dtime189
04-15 11:34:02.257 7598-15261/com.lingkang.sport I/tag5: dtime189
04-15 11:34:02.261 7598-15261/com.lingkang.sport I/tag1: copy index:5,file name:2014200196.jpg,file size:34.64 KB
04-15 11:34:02.313 7598-15261/com.lingkang.sport I/tag2: dtime56
04-15 11:34:02.314 7598-15261/com.lingkang.sport I/tag3: dtime56
04-15 11:34:02.442 7598-15261/com.lingkang.sport I/tag4: dtime185
04-15 11:34:02.442 7598-15261/com.lingkang.sport I/tag5: dtime185
04-15 11:34:02.447 7598-15261/com.lingkang.sport I/tag1: copy index:6,file name:2014200224.jpg,file size:29.39 KB
04-15 11:34:02.489 7598-15261/com.lingkang.sport I/tag2: dtime47
04-15 11:34:02.489 7598-15261/com.lingkang.sport I/tag3: dtime47
04-15 11:34:02.640 7598-15261/com.lingkang.sport I/tag4: dtime197
04-15 11:34:02.640 7598-15261/com.lingkang.sport I/tag5: dtime198
04-15 11:34:02.646 7598-15261/com.lingkang.sport I/tag1: copy index:7,file name:2014200254.jpg,file size:97.21 KB
04-15 11:34:02.758 7598-15261/com.lingkang.sport I/tag2: dtime118
04-15 11:34:02.759 7598-15261/com.lingkang.sport I/tag3: dtime118
04-15 11:34:02.908 7598-15261/com.lingkang.sport I/tag4: dtime268
04-15 11:34:02.908 7598-15261/com.lingkang.sport I/tag5: dtime268
04-15 11:34:02.912 7598-15261/com.lingkang.sport I/tag1: copy index:8,file name:2014200262.jpg,file size:29.30 KB
04-15 11:34:02.941 7598-15261/com.lingkang.sport I/tag2: dtime32
04-15 11:34:02.941 7598-15261/com.lingkang.sport I/tag3: dtime32
04-15 11:34:03.085 7598-15261/com.lingkang.sport I/tag4: dtime177
04-15 11:34:03.085 7598-15261/com.lingkang.sport I/tag5: dtime177
04-15 11:34:03.103 7598-15261/com.lingkang.sport I/tag1: copy index:9,file name:2014200314.jpg,file size:109 KB
04-15 11:34:03.231 7598-15261/com.lingkang.sport I/tag2: dtime146
04-15 11:34:03.232 7598-15261/com.lingkang.sport I/tag3: dtime146
04-15 11:34:04.510 7598-15261/com.lingkang.sport I/tag4: dtime1425
04-15 11:34:04.511 7598-15261/com.lingkang.sport I/tag5: dtime1425
04-15 11:34:04.517 7598-15261/com.lingkang.sport I/tag1: copy index:10,file name:2014200317.jpg,file size:55.94 KB
04-15 11:34:04.562 7598-15261/com.lingkang.sport I/tag2: dtime51
04-15 11:34:04.562 7598-15261/com.lingkang.sport I/tag3: dtime51
04-15 11:34:04.696 7598-15261/com.lingkang.sport I/tag4: dtime186
04-15 11:34:04.696 7598-15261/com.lingkang.sport I/tag5: dtime186

magnusja, 2019-04-15:
Hey there,

yes this is currently a limitation. I didnt think about that when I wrote the initial code. 
As a workaround I suggest try to only call `close()` when you wrote the very last file

ljliu1985, 2019-04-15:
OK, Thanks

magnusja, 2019-04-15:
Lets keep this open.