magnusja/libaums #184: device.init(); ------error

liyanmei, 2019-05-07:
[image]

` private synchronized void readDevice(UsbMassStorageDevice device) {

        try {
            device.init();//初始化
//          Only uses the first partition on the device
            if (device.getPartitions() == null || device.getPartitions().size() == 0) {
                return;
            }
            // TODO: 2019/5/7 test

            Partition partition = device.getPartitions().get(0);
            currentFs = partition.getFileSystem();
//fileSystem.getVolumeLabel()可以获取到设备的标识
//通过FileSystem可以获取当前U盘的一些存储信息，包括剩余空间大小，容量等等
//            Log.d(TAG, "Capacity: " + currentFs.getCapacity());
//            Log.d(TAG, "Occupied Space: " + currentFs.getOccupiedSpace());
//            Log.d(TAG, "Free Space: " + currentFs.getFreeSpace());
//            Log.d(TAG, "Chunk size: " + currentFs.getChunkSize());
            rootDir = currentFs.getRootDirectory();//获取根目录

            String deviceName = currentFs.getVolumeLabel();//获取设备标签
            Log.d(TAG, "readDevice: 正在读取U盘  " + deviceName);
            cFolder = rootDir;//设置当前文件对象
            createVideoAndPicDir();// 创建adas需要的目录
            moveSdFile2UsbFileHandler();// 移动sd卡中的adas文件到usb里面
        } catch (Exception e) {
            e.printStackTrace();
        }

    }`

当插入usb的时候，接收到插入的广播， device.init();//初始化---可以正常运行；但是当关闭程序，包括后台程序之后，再次打开应用，调用 device.init(); ----程序将会报错，请问这种情景下，有解决方法吗？

Hank-Long, 2019-07-31:
同样在init出异常，alert出来的异常原因为空。
小伙伴可找到可能原因？

kingiis, 2019-10-24:
com.github.mjdev.libaums.partition.PartitionTableFactory$UnsupportedPartitionTableException
same error,only got some device info,but cant read the disk

closed 2019-12-09 by magnusja

beiger, 2022-03-25:
> [image]
>
> ` private synchronized void readDevice(UsbMassStorageDevice device) {
>
> ```
>     try {
>         device.init();//初始化
> ```
>
> // Only uses the first partition on the device if [...]

same error.
请问这个问题解决了吗？