magnusja/libaums #187: Program stops when copying a file

josegerardorocha, 2019-06-13:
### Problem
Hi,
First of all thanks for this helpful library. I am trying to create a method for copying (duplicating) a file after its creation and the program stops with an exception. In order to reproduce the problem, I changed a little bit the createBigFile() method and added a copyfile() method in the MainActivity.java of the example that comes with the library. The code is shown below.

Thanks in advance
J.Rocha

#### Stacktrace of Excpetion
06-13 09:34:58.395 12827-12827/com.github.mjdev.usbfileman E/InputEventReceiver: Exception dispatching input event.
06-13 09:34:58.397 12827-12827/com.github.mjdev.usbfileman E/MessageQueue-JNI: Exception in MessageQueue callback: handleReceiveCallback
06-13 09:34:58.400 12827-12827/com.github.mjdev.usbfileman E/MessageQueue-JNI: java.lang.ArrayIndexOutOfBoundsException: length=1; index=1
        at com.github.mjdev.libaums.fs.fat32.ClusterChain.read(ClusterChain.java:112)
        at com.github.mjdev.libaums.fs.fat32.FatFile.read(FatFile.java:155)
        at com.github.mjdev.libaums.fs.UsbFileInputStream.read(UsbFileInputStream.java:104)
        at java.io.BufferedInputStream.read(BufferedInputStream.java:290)
        at java.io.InputStream.read(InputStream.java:162)
        at com.github.mjdev.libaums.usbfileman.MainActivity.copyfile(MainActivity.java:1095)
        at com.github.mjdev.libaums.usbfileman.MainActivity.createBigFile(MainActivity.java:1141)
        at com.github.mjdev.libaums.usbfileman.MainActivity.onOptionsItemSelected(MainActivity.java:816)
        at android.app.Activity.onMenuItemSelected(Activity.java:2948)

### Code where problem occurs
	private void copyfile(UsbFile dir,String fsrc,String fdest){
	    try {
            UsbFile srcPtr = dir.search(fsrc);
            long srcLen=srcPtr.getLength();
            UsbFile dstPtr = dir.search(fdest);
            if(dstPtr!=null){
                dstPtr.delete();
            }
            dstPtr=dir.createFile(fdest);

            dstPtr.setLength(srcLen);

            BufferedInputStream inputStream = UsbFileStreamFactory.createBufferedInputStream(srcPtr, currentFs);
            BufferedOutputStream outStream  = UsbFileStreamFactory.createBufferedOutputStream(dstPtr, currentFs);

            byte[] bytes = new byte[currentFs.getChunkSize()];

            int count;
            while((count=inputStream.read(bytes))>0) {
                outStream.write(bytes,0,count);
                Log.i(TAG,"Count=" + count);
            }
            inputStream.close();inputStream=null;
            outStream.close();outStream=null;

            return;
        }catch (IOException e){
	        Log.e(TAG,"Error copying: file " + e);
        }
    }
	/**
	 * This methods creates a very big file for testing purposes. It writes only
	 * a small chunk of bytes in every loop iteration, so the offset where the
	 * write starts will not always be a multiple of the cluster or block size
	 * of the file system or block device. As a plus the file has to be grown
	 * after every loop iteration which tests for example on FAT32 the dynamic
	 * growth of a cluster chain.
	 */
    private void createBigFile() {
		UsbFile dir = adapter.getCurrentDir();
		UsbFile file=null,file2=null,file3=null;
		try {
			file = dir.search("big_file_test.txt");
			if(file!=null){
				file.delete();
			}
			file = dir.createFile("big_file_test.txt");
            OutputStream outputStream = UsbFileStreamFactory.createBufferedOutputStream(file, currentFs);
			outputStream.write("START\n".getBytes());
			int i;

			for (i = 6; i < 40000; i += 5) {
                outputStream.write("TEST\n".getBytes());
			}

            outputStream.write("END\n".getBytes());

            outputStream.close();

			adapter.refresh();
		} catch (IOException e) {
			Log.e(TAG, "error creating big file!", e);
		}
		copyfile(dir,"big_file_test.txt","big_file_test2.txt");
	}

josegerardorocha, 2019-06-14:
A way to bypass the problem (not to solve it) is to put comments in lines 113 to 116 of fs/fat32/FAT.java:
`//Long[] cachedChain = cache.get(startCluster);

  //if(cachedChain != null) {

  //return cachedChain;

//}`
so, i guess the problem is related with a cache of the fat chain, that becomes invalid after a write operation, but it is not deleted.

magnusja, 2019-06-14:
Hey there,
very interesting hint. I will have a look over the weekend.
Thanks for the support.

magnusja, 2019-06-14:
Have a look at branch fix/issue-187 and play around, thanks

commit 6ab22a15 (2019-06-14): fix #187
closed 2019-06-18 by magnusja