magnusja/libaums #189: crashes on 64 and 128 GB usb drives

letisoft, 2019-06-17:
Hello,

crashed on larger USB drives 64 or 182 GB, is this expected?

magnusja, 2019-06-17:
Nope should work fine. Tested it on devices >500GB

letisoft, 2019-07-08:
Hello,

Sorry for coming back that late but I was busy with something else.

I found the problem please find bellow full description:
1) I formatted the usb stick on Android device (nothing special just form the storage settings format option)
2) On windows I can see it is fat32 formated
3) when the lib tries to read it - it is falling this check:
Fat32FileSystem
public static Fat32FileSystem read(BlockDeviceDriver blockDevice) throws IOException

		if ((char) buffer.get(82) != 'F' ||
			(char) buffer.get(83) != 'A' ||
			(char) buffer.get(84) != 'T' ||
			(char) buffer.get(85) != '3' ||
			(char) buffer.get(86) != '2' ||
			(char) buffer.get(87) != ' ' ||
			(char) buffer.get(88) != ' ' ||
			(char) buffer.get(89) != ' ') {
			return null;
		}
I can see the buffer is almost only zeros, but the drive is fat32 and works ok on linux and windows.

Any ideas or anything I can check to get this sorted?

Thank you very much!

magnusja, 2019-07-08:
Maybe try commenting that out? 

It is a little bit unfortunate that each OS has its own ways of formatting and recognizing file systems :/

letisoft, 2019-07-08:
Hello,

I tried this already :)

but as almost the whole buffer is made of zeros 
private Fat32FileSystem(BlockDeviceDriver blockDevice, ByteBuffer first512Bytes) throws IOException {

fails with FsInfoStructure 

		if (buffer.getInt(LEAD_SIGNATURE_OFF) != LEAD_SIGNATURE
				|| buffer.getInt(STRUCT_SIGNATURE_OFF) != STRUCT_SIGNATURE
				|| buffer.getInt(TRAIL_SIGNATURE_OFF) != TRAIL_SIGNATURE) {
			throw new IOException("invalid fs info structure!");
		}
Any idea?

letisoft, 2019-07-08:
The worse is it works with dongles I format on Windows but for dongles I format on android (it is fat32 windows recgonize it as fat32) it is not.

I have such device and can send you logs?

Thanks again

magnusja, 2019-07-08:
I see. then there is probably something wrong with the MBR itself. 

Can you make sure it is MBR formatted and not GPT? You should be able to see that in Windows.

Maybe you can make an image from the first couple of MB of the Android formatted drive and then I can have a look.

letisoft, 2019-07-08:
Hi,

First thank you for coming back!

UniqueId           : USBSTOR\DISK&VEN_GENERIC&PROD_MASSSTORAGECLASS&REV_CY01\00000000CY01&0:DESKTOP-IT8J8FF
Number             : 2
Path               : \\?\usbstor#disk&ven_generic&prod_massstorageclass&rev_cy01#00000000cy01&0#{53f56307-b6bf-11d0-94f
                     2-00a0c91efb8b}
Manufacturer       : Generic
Model              : MassStorageClass
SerialNumber       : 00000000CY01
Size               : 29.16 GB
AllocatedSize      : 31312576512
LogicalSectorSize  : 512
PhysicalSectorSize : 512
NumberOfPartitions : 1
PartitionStyle     : MBR
IsReadOnly         : False
IsSystem           : False
IsBoot             : False

Yes it seems MBR, please let me know what you need me to record and I will send

magnusja, 2019-07-09:
Let me have a look if I can format USB drives on my Android tablet.

If not you can provide me with an image of the USB Stick. I am actually not sure how to do that on Windows but on Linux you usually use the command line tool `dd`. Maybe that is also available for Windows

letisoft, 2019-07-09:
Hello, 

thanks allot - let me find a smaller stick and I will upload the image in about 1/2 hour

letisoft, 2019-07-09:
Hello,

Sorry but the smallest usb I have is 16GB and even after compressing the image it became 10GB so to big to upload, but I tried formatting and usb on 2 devices and got the same problem so I hope you can just format a drive?

letisoft, 2019-07-10:
Hello,

Please let me know if you need me to upload the huge image

magnusja, 2019-07-10:
I only need the first 100MB lets say

letisoft, 2019-07-10:
Hi,

How can I read them? Sorry for the pain but I am not expert :)

magnusja, 2019-07-10:
Can you use ´dd´ in Windows? And then after a couple of seconds just hit Ctrl+c to exit the process. Then you should have a file with only a couple of MB.

letisoft, 2019-07-10:
I can do it on Linux. Just dd?

And stop it after few seconds?

magnusja, 2019-07-10:
Yes if you have access to a Linux system somewhere this should work 

```
```

magnusja, 2019-07-10:
See also here https://www.cyberciti.biz/faq/linux-copy-clone-usb-stick-including-partitions/

letisoft, 2019-07-11:
Hello,

Here we go!

https://drive.google.com/file/d/1zxi4KlUwsNPxdZkzteSVVSeUM5l_RP3S/view?usp=sharing

Thank you very much for your patience.

letisoft, 2019-07-15:
Hello,

Is the image good enough for you to inspect?

Please let me know if I can help in anyway

Thanks!

magnusja, 2019-07-22:
I am very sorry I am quite busy atm. I will hopefully have time to have a look by the end of the week!

letisoft, 2019-07-22:
Thanks for the update.

I had tones of issues with SAF so your lib is much better approach (if it works)

I can explain in more details if you have time?

magnusja, 2019-08-03:
Hey there,

so I finally had time to try the image you provided. Flashed it and for me it works! Have a folder `
LOST.DIR` and a folder called `System Volume Information` on there. Can you maybe try with latest develop branch?

magnusja, 2019-08-03:
`System Volume Information` suggests that you had plugged it into a Windows machine?

Also formatted an SD card with my Sony XPeria 2 tablet and then used libaums and a SD card reader. This works as well and shows the directories `.android_secure` and `LOST.DIR`.

Not sure how to debug this :/

letisoft, 2019-08-04:
Hello,

So in this method:

	public static Fat32FileSystem read(BlockDeviceDriver blockDevice) throws IOException {

		ByteBuffer buffer = ByteBuffer.allocate(512);
		blockDevice.read(0, buffer);
		buffer.flip();

		if ((char) buffer.get(82) != 'F' ||
			(char) buffer.get(83) != 'A' ||
			(char) buffer.get(84) != 'T' ||
			(char) buffer.get(85) != '3' ||
			(char) buffer.get(86) != '2' ||
			(char) buffer.get(87) != ' ' ||
			(char) buffer.get(88) != ' ' ||
			(char) buffer.get(89) != ' ') {
			return null;
		}

		return new Fat32FileSystem(blockDevice, buffer);
	}

You are not getting all zeros in the buffer?

If you have address I can send you real USB drive with the problem?
Thanks!

magnusja, 2019-08-04:
Maybe you can tell me first what libaums version and what Android version you are using.

Maybe I can find an Android device with the same version and try again

closed 2019-12-09 by magnusja