magnusja/libaums #190: UsbMassStorageDevice.Init() exceptio timeout

khongta0932488598, 2019-06-24:
I am using this lib in my Xamarin.Android project

devices = UsbMassStorageDevice.GetMassStorageDevices(this);

            foreach (UsbMassStorageDevice device in devices)
            {
                Java.Lang.Thread.Sleep(7000);
                // before interacting with a device you need to call init()!
                device.Init();
                // Only uses the first partition on the device
                var currentFs = device.Partitions.FirstOrDefault().FileSystem;
                Log.Debug(TAG, "Capacity: " + currentFs.Capacity);
                Log.Debug(TAG, "Occupied Space: " + currentFs.OccupiedSpace);
                Log.Debug(TAG, "Free Space: " + currentFs.FreeSpace);
                Log.Debug(TAG, "Chunk size: " + currentFs.ChunkSize);
            }
So When i call device.Init() , it throw exception 
Java.IO.IOException: <Timeout exceeded getting exception details>

Could anyone help me for that?

magnusja, 2019-06-24:
IOExcpetion is very generic, can you paste the full stacktrace please?

khongta0932488598, 2019-06-24:
This is the output after I call device.Init()

06-24 15:37:02.436 D/SemPersonaManager( 5513): Knox:: claimInterface : request for user -0 and interface reuqest -8
06-24 15:37:02.440 D/SemPersonaManager( 5513): Knox:: claimInterface : calling isPackageAllowedToAccessExternalSdcard for user- 0 and callingUid-10318
06-24 15:37:02.448 D/SemPersonaManager( 5513): Knox:: claimInterface : calling isPackageAllowedToAccessExternalSdcard allowed-true
06-24 15:37:02.471 I/UsbMassStorageDevice( 5513): MAX LUN 0
06-24 15:37:02.480 D/ScsiBlockDevice( 5513): inquiry response: ScsiInquiryResponse [peripheralQualifier=0, peripheralDeviceType=0, removableMedia=false, spcVersion=4, responseDataFormat=2]
06-24 15:37:02.485 I/ScsiBlockDevice( 5513): Block size: 4096
06-24 15:37:02.485 I/ScsiBlockDevice( 5513): Last block address: 257984
06-24 15:37:02.526 D/ScsiBlockDevice( 5513): read time: 36
06-24 15:37:02.560 D/ScsiBlockDevice( 5513): read time: 33
06-24 15:37:02.562 D/MasterBootRecord( 5513): Unknown partition type 114
06-24 15:37:02.562 D/MasterBootRecord( 5513): Unknown partition type 101
06-24 15:37:02.562 D/MasterBootRecord( 5513): Unknown partition type 121
06-24 15:37:02.563 D/MasterBootRecord( 5513): Unknown partition type 13
Unhandled Exception:

Java.IO.IOException: <Timeout exceeded getting exception details>

magnusja, 2019-06-24:
What kind of USB device are you trying to read?
Can you just try a newly formatted FAT32 USB drive?

khongta0932488598, 2019-06-24:
I am trying to work with files in disk drive of scientific device. It use FAT32 formatting.

magnusja, 2019-06-24:
Okay, can you make sure it uses MBR partition table and not GPT?

khongta0932488598, 2019-06-24:
I just checked the disk drive and It has three inner disk drives which is one is MBR,GPT and Not applicable.

magnusja, 2019-06-25:
That is an interesting constellation. Unfortunately, libaums only supports MBR at this point (no GPT).
Is it possible for you to reformat the drive to MBR only?

khongta0932488598, 2019-06-29:
Hi @magnusja,thanks for your time. That is silly mistake. I just tried other approach.

closed 2019-07-01 by magnusja