magnusja/libaums #191: not getting usb files

azhar5678, 2019-07-03:
UsbMassStorageDevice[] devices = UsbMassStorageDevice.getMassStorageDevices(this /* Context or Activity */);

for(UsbMassStorageDevice device: devices) {
    
    // before interacting with a device you need to call init()!
    device.init();
    
    // Only uses the first partition on the device
    FileSystem currentFs = device.getPartitions().get(0).getFileSystem();
    Log.d(TAG, "Capacity: " + currentFs.getCapacity());
    Log.d(TAG, "Occupied Space: " + currentFs.getOccupiedSpace());
    Log.d(TAG, "Free Space: " + currentFs.getFreeSpace());
    Log.d(TAG, "Chunk size: " + currentFs.getChunkSize());
}

app crashes with this code am using it in on create method of activity

and how do i get permisson to read and what is broadcast reciever doing on start

please give me simple code to list files with all the permission

because ur example activity contains lots of other code which i dont need

azhar5678, 2019-07-03:
here it is what i tried but it say unfortunately app stop when i connect pendrive 
when i dont connect pendriveit execute and show buttons.  please help

package com.github.mjdev.libaums.usbfileman;

import android.app.Activity;
import android.app.PendingIntent;
import android.content.BroadcastReceiver;
import android.content.Context;
import android.content.Intent;
import android.content.res.Configuration;
import android.hardware.usb.UsbDevice;
import android.hardware.usb.UsbManager;
import android.os.Bundle;
import android.text.method.ScrollingMovementMethod;
import android.util.Log;
import android.view.Gravity;
import android.view.View;
import android.view.ViewGroup;
import android.widget.Button;
import android.widget.EditText;
import android.widget.HorizontalScrollView;
import android.widget.LinearLayout;
import android.widget.MediaController;
import android.widget.ScrollView;
import android.widget.TextView;
import android.widget.Toast;
import android.widget.VideoView;

import com.example.playenc.CryptoException;
import com.example.playenc.playvedio;
import com.github.magnusja.libaums.javafs.JavaFsFileSystemCreator;
import com.github.mjdev.libaums.UsbMassStorageDevice;
import com.github.mjdev.libaums.fs.FileSystem;
import com.github.mjdev.libaums.fs.FileSystemFactory;
import com.github.mjdev.libaums.fs.UsbFile;
import com.github.mjdev.libaums.fs.UsbFileInputStream;
import com.github.mjdev.libaums.fs.UsbFileOutputStream;

import java.io.BufferedReader;
import java.io.IOException;
import java.io.InputStream;
import java.io.InputStreamReader;
import java.io.OutputStream;

import static com.github.mjdev.libaums.usbfileman.R.layout.activity_mainplay;

public class newjava  extends Activity {
    private VideoView jj;

    private static final String ACTION_USB_PERMISSION = "com.github.mjdev.libaums.USB_PERMISSION";
    private static final String TAG = MainActivity.class.getSimpleName();

    private static final int COPY_STORAGE_PROVIDER_RESULT = 0;
    private static final int OPEN_STORAGE_PROVIDER_RESULT = 1;
    private static final int OPEN_DOCUMENT_TREE_RESULT = 2;
    private int currentDevice = -1;

    private static final int REQUEST_EXT_STORAGE_WRITE_PERM = 0;
    UsbMassStorageDevice[] massStorageDevices;

    static {
        FileSystemFactory.registerFileSystem(new JavaFsFileSystemCreator());
    }

    /**
     * Action string to request the permission to communicate with an UsbDevice.
     */

    private final BroadcastReceiver usbReceiver = new BroadcastReceiver() {
[... 357 lines cut ...]
                                }
                            }

                        }

                    getfile(ff, mediaController, jj, ein);
                } catch (IOException e) {
                    e.printStackTrace();
                }
            }
        }

    }

    private int countOccurrences(String haystack, char needle) {
        int count = 0;
        for (int i = 0; i < haystack.length(); i++) {
            if (haystack.charAt(i) == needle) {
                count++;
            }
        }
        return count;
    }

}

magnusja, 2019-07-04:
Does the exmaple application work for you?

What is the stacktrace of the exception?

azhar5678, 2019-07-04:
Example application works fine.
but that contains inner classes and
It's hard to predict what's going on .
when all code is in single page.

may permission needed all the time of accessing USB.
But I am unable to understand what's going on.

In example I don't need copy task.
Run tests.

magnusja, 2019-07-04:
Well then just compare to the example app, that what it is for.

closed 2019-07-04 by magnusja

azhar5678, 2019-07-04:
I will talk to u later.
thanks. For the reply.

azhar5678, 2019-07-05:
error on using toast.maketext
for currentFs.getChunkSize()

errorr recieving bradcast
intent act=com.github...........................usb_permission

magnusja, 2019-07-05:
Can you post the full thing?

Eldho-Abraham-coder, 2021-10-05:
i am working on android 6 , what version of libaums do we need to use in it. do we need to create stub IUsbManager.java as mentioned in https://stackoverflow.com/questions/13726583/bypass-android-usb-host-permission-confirmation-dialog