magnusja/libaums #192: read file in libaums

azhar5678, 2019-07-08:
how to read file from usb

 UsbFile dir = activity.adapter.getCurrentDir();
                    
i want to read file hello.txt from root dir

and hello1.txt in aaa/bbb/  directorty

something like that
 UsbFile dir = activity.adapter.getCurrentDir();
dir = new File("hello.txt);
dir= new File("/aaa/bbb/hello1.txt");

magnusja, 2019-07-08:
This is in the documentation as well es in the example app

closed 2019-07-08 by magnusja

azhar5678, 2019-07-08:
if i use
UsbFile ff;
BufferedReader br = new BufferedReader(new FileReader(ff);

will  it work

magnusja, 2019-07-08:
No, read the documentation!

azhar5678, 2019-07-08:
UsbFile[] filescc = root.listFiles();
for(int g=0;g<filescc.length;g++)
{
    Toast.makeText(activity, "ff="+filescc[g].getName(), 600).show();

}

i hava 10 directory in usb root but the above code showing only 5 directory

magnusja, 2019-07-08:
Hidden directories are ignored by default

azhar5678, 2019-07-08:
files.length shows 87 directory
but 
Toast.makeText(activity, "ff="+filescc[g].getName(), 600).show();

show only 5 directory if i delete files than it show next directory or files but showing total 5

magnusja, 2019-07-08:
`.inf` is considered hidden, `name.inf` is not considered hidden.

azhar5678, 2019-07-08:
i delete all hidden files , delete special charachters from file name
but ths Lost.dir keep comming and
Toast.makeText(activity, "ff="+filescc[g].getName(), 600).show();

show only 5 directory if i delete files than it show next directory or files but showing total 5

azhar5678, 2019-07-08:
do it show only 5 directory
for(int g=0;g<filescc.length;g++)
{
Toast.makeText(activity, "ff="+filescc[g].getName(), 600).show();

}

magnusja, 2019-07-08:
If it works in the sample application you have to compare what you are doing differently.

It is impossible for me to guess what is wrong with the information you gave me. I am not a magician or sth.

Also why are you using Toasts? Put that in a list or dump it to the debug log.

azhar5678, 2019-07-08:
whats wrong with toast

azhar5678, 2019-07-08:
ok each and every files are showing in alert dialog box,
i will talk to u later thanks for rely

magnusja, 2019-07-08:
> whats wrong with toast

Well as you already noticed yourself you are not seeing every toast if you show 10 at the same time. That is not what they were designed for....

azhar5678, 2019-07-08:
unable to write to file ceating 0 bytes
object="ghfhfhfhfjhjh";
  OutputStream os = new UsbFileOutputStream(filescc127[m]);
os.write(object.getBytes());
                                    os.close();

azhar5678, 2019-07-09:
please reply

azhar5678, 2019-07-09:
what is the meaning of "this is a directory" exception

magnusja, 2019-07-09:
That means that you attempted an operation which is only valid on files on a directory (like read,write,length)

azhar5678, 2019-07-09:
Only valid files inside a directory.
That means files which is writing reading
Is a direcory.

magnusja, 2019-07-09:
Apparently there are not only files but also directories which you treat as files which does not make sense.

azhar5678, 2019-07-09:
i have one more question
I want to access sys/usb /subsystem folder in android 9  but its giving
access denied before android 9 .iam able to read files from there.
how to get rid of permission denied

magnusja, 2019-07-10:
I dont know.

Maybe they changed it so that apps cannot access it anymore, but maybe you just need a newly introduced permission. I would ask that in an Android developer forum.

azhar5678, 2019-07-10:
Thanku please ask .

azhar5678, 2019-07-10:
how to pass currentFs and activity object to another activity

azhar5678, 2019-07-10:
after reading file from usb , computer saying u need to format the derive n (usb) before use

what happen and file manger not showing it , phone saying it damage usb

what happen i want to know

magnusja, 2019-07-10:
Something corrupted the USB device. This cannot happen when reading only. You probably wrote something to the USB stick.

This may either be a bug in this library but can also happen if you remove the device too early so that buffered data was not written.

azhar5678, 2019-07-10:
Iam using ur library for the project.

And daily hundred pendrive will be sent
Which open only by ur library.
Iam writing a vedio piracy protection softwate in android.

Is ur library fully dependable

magnusja, 2019-07-10:
I am trying my best maintaining this library and fixing bugs but obviously I cannot guarantee that it works like implementations on Windows or Linux.

azhar5678, 2019-07-10:
is there any other libray that u want to suggest, i mean if app not running thats ok
but no damage should be done to data in pendrive

magnusja, 2019-07-10:
This one is the only one I know of tbh.

But I think that most Android devices nowadays automatically mount USB pen drives.

Yes corrupting the drive should not happen, I am very sorry for that.

azhar5678, 2019-07-10:
any damage that done , should be done before sending 100 pendrive daily, if i will give the project to the client, than i cant do anything.

magnusja, 2019-07-10:
Well I am happy to welcome you as a regular contributor to this library if you need to rely on it.

azhar5678, 2019-07-10:
but what happen just now, i mean i should give catuions to my client that u should not plug the pendrive out
when it in use or  

or atleast  my code currently  will not damage the usb

i was reading an encrypted file and decrypting it to the internal drive, with aes algorithm

that dynamical added around 60 texviews in activity 
thats where the damage happen
can u tell me why?

magnusja, 2019-07-10:
Are you 100% sure that you only read from the pen drive and did not do a SINGLE write operation??

And are you using the UsbFileInputStream to read a file?

azhar5678, 2019-07-10:
no iam writing something to file 7548.txt in ddd directory but unable to write their with 100 of try
usbfileman finding the file inside the directory but not succesful in writing to it.

and iam using 
 BufferedReader br = new BufferedReader(new FileReader(files[i]));
                               to read a file inside sys /bus/usb1/... modalias file

magnusja, 2019-07-10:
> no iam writing something to file 7548.txt in ddd directory but unable to write their with 100 of try
> usbfileman finding the file inside the directory but not succesful in writing to it.
> 
I dont understand
> and iam using
> BufferedReader br = new BufferedReader(new FileReader(files[i]));
> to read a file inside sys /bus/usb1/... modalias file
This is unrelated to libaums and only has something to do with Android itself

azhar5678, 2019-07-10:
currently this code writing to a file

 StringBuilder hhh = new StringBuilder();
                    StringBuilder hhh1 = new StringBuilder();

                    try {
                        UsbFile dir = currentFs.getRootDirectory();

                        UsbFile[] file2 = dir.listFiles();

                        for(int y=0;y<file2.length;y++) {
                            hhh.append(file2[y].getName());
                            hhh.append("\n");
                            if(file2[y].getName().equals("ddd") && file2[y].isDirectory())
                            {
                                AlertDialog.Builder builderee = new AlertDialog.Builder(activity);

                                builderee.setTitle("New File");
                                builderee.setMessage("foundnnnnndd"+hhh);
                            //    builder.show();

                              UsbFile[] file3 = file2[y].listFiles();
                                for(int s=0;s<file3.length;s++)
                                {
                                    hhh1.append(file3[s].getName());
                                    hhh1.append("\n");
                                    if(file3[s].getName().equals("7534.txt") )
                                    {
                                        AlertDialog.Builder builder1 = new AlertDialog.Builder(activity);

                                        builder1.setTitle("New File");
                                        builder1.setMessage("foundnnnnnddtttt"+hhh1);
                                     //   builder1.show();

                                       file3[s].write(0, ByteBuffer.wrap("hello".getBytes()));
                                        file3[s].flush();
                                     //   file3[s].close();
                                       // activity.adapter.refresh();
                                    }
                                   // file3[s].close();
                                }
                            }
                        //    file2[y].close();

                        }
                    } catch (Exception e) {
                        AlertDialog.Builder builder1 = new AlertDialog.Builder(activity);

                        builder1.setTitle("New File");
                        builder1.setMessage(e.getMessage());
                        builder1.show();

                        Log.e(TAG, "error creating file!", e);
                    }

magnusja, 2019-07-10:
Why dont you use `UsbFile.search(path)` to look for the things you interested in.

Flushing should only happend at the end when you write your file (ie. at the end of the loop)

azhar5678, 2019-07-10:
what is the path is it file name or path like/mnt

azhar5678, 2019-07-10:
i also used this much buffer to read file
 byte[] plainBuf = new byte[1024*1024*1024*300];

azhar5678, 2019-07-10:
bro iam writing the project for past 20 days and now its two or three days left and then i need to message the client.

can i rely on ur library

azhar5678, 2019-07-10:
ok iam going to tell my client that their buyer should backup their 32 gb pendrive
because this library is currently in testing phase

and will see what happen

azhar5678, 2019-07-10:
please tell me what type of testing i should do before sending the project to the client

azhar5678, 2019-07-10:
data is not lost its showing  windows , everything is there, but on android iam unable to write to the usb with any file manager

azhar5678, 2019-07-10:
ok now data is showing in usbfileman app, everything is there nothing lost
but why android system ask me to format my drive, and how everything is there after format from android

magnusja, 2019-07-11:
This is hard to say if I hardly know what exactly you did.

I f you formatted and the data is still there, maybe you formatted a different partition of the USB pen drive

azhar5678, 2019-07-12:
how to check if file exist or not
UsbFile file = dir.search("6677.txt");
if(file.exist())
{
}

file.exist()  is not there

magnusja, 2019-07-12:
are you even trying to read the documentation? Or are you just asking every problem you encounter? This is insane. I have to bill you in the future for consulting work.

https://www.javadoc.io/doc/com.github.mjdev/libaums/0.6.0

azhar5678, 2019-07-21:
@
I dont know.

Maybe they changed it so that apps cannot access it anymore, but maybe you just need a newly introduced permission. I would ask that in an Android developer forum.

any information on that

azhar5678, 2019-08-10:
iam trying to play dash vedio in exoplayer through your usbfileman httpserver
i serve file http://localhost:8000/manifest.mpd

but exoplayer dont play the vedio file, i guess other m4s file also need to serve on http or maybe not

any idea

azhar5678, 2019-08-11:
Please reply

magnusja, 2019-08-12:
Yeah you need to server whole folder then not only the manifest file

azhar5678, 2019-08-12:
Suppose i have play.html and hh.jpg file
Inside hello folder. After serving whole folder how to run play.html
Is it like this.
http://localhost:8000/hello/play.html

magnusja, 2019-08-12:
http://localhost:8000/play.html probably

azhar5678, 2019-08-12:
Solved but i have now same old problem
dir.search(hello/play.html)
not finding the play.html
file inside hello directory

azhar5678, 2019-08-12:
Please reply

magnusja, 2019-08-12:
Not sure what you mean sorry

azhar5678, 2019-08-12:
Instead of putting files in same directory
To server. Gather files from different location and put them on them on the
server.

magnusja, 2019-08-12:
You can just server the root `/` and then just access them using folder names.

azhar5678, 2019-08-20:
a single file i need to serve fom internal storage while others are from usb directory

in nanohttp
 @Override
    public Response serve(IHTTPSession session) {

DocumentFile dir = null;
        DocumentFile fileToServe1;
        try {
            if(session.getUri().contains("manifest"))
            {
               //  file = new File(Environment.getExternalStorageDirectory("","manifest.mpd"));
                 fileToServe1=dir.findFile("manifest.mpd") ;
                if(range == null) {
                    return serveCompleteFile((UsbFile) fileToServe1);
                } else {
                    return serveRangeOfFile((UsbFile) fileToServe1, range);
                }
            }
            else {
                fileToServe = usbFileProvider.determineFileToServe(uri);
                if(range == null) {
                    return serveCompleteFile(fileToServe);
                } else {
                    return serveRangeOfFile(fileToServe, range);
                }
            }

DocumentFile dir = null;
        DocumentFile fileToServe1;
        try {
            if(session.getUri().contains("manifest"))
            {
               //  file = new File(Environment.getExternalStorageDirectory("","manifest.mpd"));
                 fileToServe1=dir.findFile("manifest.mpd") ;
                if(range == null) {
                    return serveCompleteFile((UsbFile) fileToServe1);
                } else {
                    return serveRangeOfFile((UsbFile) fileToServe1, range);
                }
            }
            else {
                fileToServe = usbFileProvider.determineFileToServe(uri);
                if(range == null) {
                    return serveCompleteFile(fileToServe);
                } else {
                    return serveRangeOfFile(fileToServe, range);
                }
            }
}

magnusja, 2019-08-20:
I dont think that you can cast the file on the internal storage to a `UsbFile`. You need to implement a custom method which can serve a `DocumentFile`.

[8 more replies]