magnusja/libaums #193: unable to read file root directyory

azhar5678, 2019-07-09:
try {
                        UsbFile dir1 = currentFs.getRootDirectory();
                        UsbFile[] inputFile = dir1.listFiles();
                        for (int h = 0; h < inputFile.length; h++) {
                            if (inputFile[h].getName().equals("36432.txt")) {
                                InputStream is = new UsbFileInputStream(inputFile[h]);
                                byte[] buffer = new byte[512];
                                is.read(buffer);
                                String line = null;
                                BufferedReader reader = new BufferedReader(new InputStreamReader(is));
                                while(reader.ready()) {
                                    line = reader.readLine();
                                }
                                AlertDialog.Builder builder1 = new AlertDialog.Builder(activity);

                                builder1.setTitle("New File");
                                builder1.setMessage("foundnnnnndd3456"+line);
                                builder1.show();

                            }

                        }

                    }
                    catch(Exception e)
                    {

                        AlertDialog.Builder builder1 = new AlertDialog.Builder(activity);

                        builder1.setTitle("New File");
                        builder1.setMessage("erroor"+e.getMessage());
                        builder1.show();

                    }

magnusja, 2019-07-09:
You are reading into the byte buffer and then try to read a string from the "finished" `InputStream`.

Try printing `String(buffer)`.

I also suggest that you take a primer in Java Input/OutputStream handling as well as a general Java / Programming Intro.

Next step would be to put more effort into your code formatting here and learn how to use Markdown properly.

Please do not open any more issues. This is for bug reports. Not for general questions why your Java Code is not working. Please move to StackOverflow for that.

closed 2019-07-09 by magnusja

azhar5678, 2019-07-09:
InputStream is = new UsbFileInputStream(inputFile[h]);
byte[] buffer = new byte[512];
is.read(buffer);
System.out.println(buffer.tostring());

its printing null

magnusja, 2019-07-09:
As I said, please do some beginner tutorials as you are lacking the most basic Java and Android understanding.

`is.read(buffer)` reads exactly 1 byte you probably want to read more.
`buffer.toString()` gives you a String describing your array object not the actual content of the array. Try `String(buffer)` which I already suggest in my last post but for whatever reason you completely ignored ....
On Android you usually use `Lod.d(...)` to print.

azhar5678, 2019-07-09:
ok i will do whatever u say put more efforts search google before asking but one thing
inputFile[h]
is it passing the file refrence to UsbFileInputStream(inputFile[h]) ; correctly

magnusja, 2019-07-09:
This looks correct yes

azhar5678, 2019-08-04:
iam using libaums with exoplayer and move whole ur app folder in their demo app , i came this
error: incomparable types: com.libaums.driver.scsi.commands.CommandBlockWrapper.Direction and com.github.mjdev.libaums.driver.scsi.commands.CommandBlockWrapper.Direction

magnusja, 2019-08-04:
Best is to install this library as a dependency in gradle and then use it.

azhar5678, 2019-08-04:
i use
 maven { url 'https://dl.bintray.com/magnusja/maven' }
            maven {
                url 'https://maven.google.com/'
                name 'Google'
            }
in build.gradle app folder and added
 implementation 'com.github.mjdev:libaums:0.7.0'

    implementation 'com.github.mjdev:libaums-httpserver:0.5.3'
    implementation 'com.github.mjdev:libaums-storageprovider:0.5.1'
as dependencies , but its not compiling
package javafs does not exist

magnusja, 2019-08-04:
well if you want to use java-fs you also have to declare it as a dependency.

azhar5678, 2019-08-04:
what is the name of javafs dependency , iam unable to find on ur page,
i declared provider in android manifest.xml

magnusja, 2019-08-04:
Oh I am sorry, my bad. java-fs is still experimental and therefore no release available.

So you have to copy the whole java-fs folder to your app and then include it like this: https://github.com/magnusja/libaums/blob/develop/app/build.gradle#L34

magnusja, 2019-08-04:
And you probably also have a file like this where you need to add the module parallel to your app https://github.com/magnusja/libaums/blob/develop/settings.gradle

azhar5678, 2019-10-02:
error device busy
how to open only one instance or close every other instance(if  open) of usbfileman when plug a usb.

azhar5678, 2019-10-25:
how to make usbfileman an instant apk

https://developer.android.com/topic/google-play-instant/getting-started/first-instant-app