magnusja/libaums #194: java.lang.IndexOutOfBoundsException

reemNCQ, 2019-07-11:
Hello Everyone ... I'm facing a problem with reading from USB Mass storage devices and I hope that I can someone to give me a hint ...
In fact the application can successfully connect to the device but once I'm trying to open and read no partition is detected and I get the following error : 
[image]
So after initializing through `device.init()` no partition is found
PS: the test has been done with usb disk storage that actually work with other file browsing apps ...
I hope I got a fast response and many thanks in advance..

magnusja, 2019-07-11:
This library currently only supports MBR and FAT32 so make sure that you are using that!

reemNCQ, 2019-07-11:
Already verified I double checked the storage format.

letisoft, 2019-07-13:
Hello, 

May be it is related to the same issue I reported on the previous thread?

reemNCQ, 2019-07-15:
The device should be FAT32 and MBR at the same ... Thank you everyone I'm going close the issue :)

closed 2019-07-15 by reemNCQ