magnusja/libaums #195: ConcurrentModification

gxshao, 2019-07-16:
### Problem
ConcurrentModification Exception throws

### Expected behavior
parallel create new file and hashmap throws exception down below

### Actual behavior

#### Stacktrace of Excpetion

```java
    java.util.ConcurrentModificationException
        at java.util.HashMap$HashIterator.nextNode(HashMap.java:1441)
        at java.util.HashMap$KeyIterator.next(HashMap.java:1465)
        at com.github.mjdev.libaums.fs.fat32.ShortNameGenerator.containShortName(ShortNameGenerator.java:209)
        at com.github.mjdev.libaums.fs.fat32.ShortNameGenerator.generateShortName(ShortNameGenerator.java:187)
        at com.github.mjdev.libaums.fs.fat32.FatDirectory.createFile(FatDirectory.java:350)
        at com.github.mjdev.libaums.fs.fat32.FatDirectory.createFile(FatDirectory.java:42)
```

### Code where problem occurs

closed 2019-07-19 by gxshao

happyAshley, 2020-08-02:
may i know how you solve this problem? i I had the same problem

happyAshley, 2020-08-03:
@gxshao may i know how you solve this problem? i I had the same problem