magnusja/libaums #203: 插上U盘后，直接init错误

HeartHappy, 2019-08-06:
E/MainActivity: error setting up device
    com.github.mjdev.libaums.driver.scsi.UnitNotReady: Device is not ready (Unsuccessful ScsiTestUnitReady Csw status)
        at com.github.mjdev.libaums.driver.scsi.ScsiBlockDevice.init(ScsiBlockDevice.kt:100)
        at com.github.mjdev.libaums.UsbMassStorageDevice.setupDevice(UsbMassStorageDevice.kt:156)
        at com.github.mjdev.libaums.UsbMassStorageDevice.init(UsbMassStorageDevice.kt:114)
        at com.github.mjdev.libaums.usbfileman.MainActivity.setupDevice(MainActivity.java:761)
        at com.github.mjdev.libaums.usbfileman.MainActivity.discoverDevice(MainActivity.java:745)
        at com.github.mjdev.libaums.usbfileman.MainActivity.onCreate(MainActivity.java:697)
        at android.app.Activity.performCreate(Activity.java:7009)
        at android.app.Activity.performCreate(Activity.java:7000)
        at android.app.Instrumentation.callActivityOnCreate(Instrumentation.java:1214)
        at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2731)
        at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:2856)
        at android.app.ActivityThread.-wrap11(Unknown Source:0)
        at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1589)
        at android.os.Handler.dispatchMessage(Handler.java:106)
        at android.os.Looper.loop(Looper.java:164)
        at android.app.ActivityThread.main(ActivityThread.java:6494)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:438)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:807)
        at de.robv.android.xposed.XposedBridge.main(XposedBridge.java:108)

magnusja, 2019-08-06:
Does this work with version `0.6.0` ?

kingiis, 2019-10-23:
作者都提醒你升级最新版本哦

kgsharathkumar, 2019-12-02:
I'm also facing same error. When I connect the usb for second time after giving all the permissions it will not show me any pop up. But not able to proceed because of the above error.

Kindly help me out?

magnusja, 2019-12-02:
Unplugging the usb device does not help? This should reset the device usually.

kgsharathkumar, 2019-12-03:
It didn't work. It is showing the below error exactly.

Device is not ready (Unsuccessful ScsiTestUnitReady Csw status).

This is happening every second  time on words when I insert the usb after giving all the permissions.

magnusja, 2019-12-09:
Maybe related to  #209 still looking into fixing this though ..

kiss2sky, 2020-01-02:
device.init();//初始化
Device is not ready (Unsuccessful ScsiTestUnitReady Csw status).