magnusja/libaums #206: Questions on possible changes to be performed

Depau, 2019-08-18:
Hi, I'm opening this issue because, while libaums fulfills most of my needs for [EtchDroid](https://github.com/EtchDroid/EtchDroid), there are currently a few issues which are causing me to keep my incompatible [fork](https://github.com/EtchDroid/libdepaums), and I'd reeeeally like to avoid that.

So these are a few changes I'd like to merge here, let me know what you think about them and what kind of changes you would like.

---

First of all, and we already talked about this, is that I'd need to access the block device directly.
In my fork I made `blockDevice` public:

https://github.com/EtchDroid/libdepaums/blob/master/libdepaums/src/main/java/eu/depau/etchdroid/libdepaums/UsbMassStorageDevice.kt#L106

I remember you rejected my PR for this because I didn't have any tests written. Before you ask: no, I'm not planning to write unit tests for the SCSI protocol :joy:

It should probably be better to write instrumented tests for that instead of unit tests. And by the way I'm planning to do that! I'm working on a Qemu setup with Android-x86 that emulates a USB 3.0 stick, it works-ish but I need to get some issues fixed.

Other than that it's just a matter of running qemu with:

```
  -device nec-usb-xhci,id=xhci \
  -drive if=none,id=usbstick,file=usb.img,format=raw \
  -device usb-storage,bus=xhci.0,drive=usbstick
```

---

(Pretty easy to merge if above also gets merged)

I wrote `InputStream` and `OutputStream` implementations on top of `BlockDeviceDrivers`: 

https://github.com/EtchDroid/libdepaums/tree/master/libdepaums/src/main/java/eu/depau/etchdroid/libdepaums/streams

With tests:

https://github.com/EtchDroid/libdepaums/tree/master/libdepaums/src/test/java/eu/depau/etchdroid/libdepaums/streams

The use case is simply to be able to copy data in a `dd` kinf of fashion, which is what EtchDroid does. Having a wrapper that implements a standard Java interface makes it easy for me to reuse code and provide the same kind of functionality on top of other implementations.

---

(Breaks API)

I don't necessarily need to access the filesystem inside of the USB drive, the fact that by default it reads the partition table kind of gets in my way, so I had to incompatibly split that away:

https://github.com/EtchDroid/libdepaums/blob/dafec1f38fd2028bd983cc570d4dbf7efbbd3d85/libdepaums/src/main/java/eu/depau/etchdroid/libdepaums/UsbMassStorageDevice.kt#L124-L194

While this is a deal-breaker for me, I could probably be able to keep a soft-fork if you'd prefer not to break the API. Or if you have a better solution that doesn't break the API I'll be happy to adopt it.

Let me know :)

Depau, 2019-08-18:
I don't know why I didn't think of it earlier, but I realized I could actually write a wrapper around this library.

https://github.com/EtchDroid/libaums_wrapper

I would still like to propose an API change (incremental), as I've had to use reflection to retrieve some fields:

https://github.com/EtchDroid/libaums_wrapper/blob/993810e9aa2374a80351dbeaa233b4eb5c06921b/libaums_wrapper/src/main/java/eu/depau/etchdroid/libaums_wrapper/kotlinexts/blockdevicedriver.kt#L8-L30

I also heavily modified `UsbMassStorageDevice`, but it wasn't a big issue as I can provide my own alternative without breaking the API.

https://github.com/EtchDroid/libaums_wrapper/blob/master/libaums_wrapper/src/main/java/eu/depau/etchdroid/libaums_wrapper/EtchDroidUsbMassStorageDevice.kt

I'd still prefer to have everything here, though, so do let me know if you want any changes for any pull requests to be accepted :)

magnusja, 2019-08-18:
Hey there,

yes I remember. The problem I see that the raw block device access just makes it more complicated for the regular user.

I remember the first one. I think this was public but just got lost in the Kotlin port, sorry about that. Please open a PR for that. 
Instrumentation tests sound awesome, please keep me posted for that one. How exactly does the QEMU thing work? Did you write the other side of the USB device for QEMU offers QEMU virtual devices for that out of the box?

Feel free to add a size property as well. For the block device based streams I would suggest that you keep it in your wrapper for now.

For the `UsbMassStorageDevice` I would like to avoid making an extra call to `initPartitions` necessary. Are you sure that this is that bad for you? Worst case this method throws an exception which you can probably catch right? Other option would be to add a flag `initPartition` to `init`
which is true by default.

Depau, 2019-08-19:
> Hey there,
> 
> yes I remember. The problem I see is that the raw block device access just makes it more complicated for the regular user.

Alright. As long as the classes stay public it's good for me, a separate implementation of `UsbMassStorageDevice` shouldn't be too hard to keep.

> I remember the first one. I think this was public but just got lost in the Kotlin port, sorry about that. Please open a PR for that.

I will :)

> Instrumentation tests sound awesome, please keep me posted for that one. How exactly does the QEMU thing work? Did you write the other side of the USB device for QEMU offers QEMU virtual devices for that out of the box?

QEMU offers a shitload of emulated devices, including a few different implementations of USB storage devices. I haven't been able to use them with the standard Android emulator (which uses QEMU under the hood), but maybe I did something wrong.

It does work with upstream QEMU + an [Android-x86](https://android-x86.org) build (the USB emulation, not libaums - I haven't tested it yet).

Basically you fire up the emulator with options to add a USB bus, connect a SCSI bulk-only USB mass storage device to it and attach an image file. Doing a quick check with `lsusb -v` it's almost identical to what I get on the host machine connecting a cheap USB drive known to work with EtchDroid.

I'm not totally satisfied with upstream Android-x86 builds though as they include the Google Play services and some bloat. Also Android Pie builds have not been uploaded. So I fired up my server and it's currently making plain Oreo and Pie builds for x86_64 PC.

Once that's done I'll try to establish a reproducible test environment and upload them somewhere to easily download them for tests.

> Feel free to add a size property as well. For the block device based streams I would suggest that you keep it in your wrapper for now.

Alright :)

> For the `UsbMassStorageDevice` I would like to avoid making an extra call to `initPartitions` necessary. Are you sure that this is that bad for you? Worst case this method throws an exception which [...]

I haven't checked with the latest changes, the problem was that if I initialized the device and it had a different partition table type or no partition table at all, it would throw an exception.

I guess I could catch it, but why have this extra work done when I'm certainly not going to need it?

I understand that your use case is a bit different and that most users actually want to access the filesystem. I can completely skip this part with my separate `UsbMassStorageDevice`, so no problem if you'd like to keep it as it is.

magnusja, 2019-08-19:
Ok I understand. So with QEMU we/I could test the library as a whole without writing any more code to emulate a USB device, that is really cool :)

Regarding the `UsbMassStorageDevice`, we could also have `initBlockDevice` and `initPartitions`, both public and still keep `init` which just calls `initBlockDevice` and `initPartitions`. Regular user still uses `ìnit` but you use `initBlockDevice` instead.

magnusja, 2019-09-09:
Hey @Depau 

do you want to still implement the changes proposed? Otherwise I would close this now.

Depau, 2019-09-09:
Hey! I should be free in a week. I think everything in this issue has either been addressed or it has it's own issue, am I missing something?

- [ ] QEMU changes #209 
- [x] [My wrapper](https://github.com/EtchDroid/libaums_wrapper) with additional functionality
- [x] `BlockDeviceDriver.getSize()` #207 
- [x] Minor improvements #208

magnusja, 2019-09-10:
Well I suggested this 
> Regarding the UsbMassStorageDevice, we could also have initBlockDevice and initPartitions, both public and still keep init which just calls initBlockDevice and initPartitions. Regular user still uses ìnit but you use initBlockDevice instead.

But if you prefer to have that in your wrapper this is fine with me, then I will just close this, thanks!

closed 2019-09-10 by magnusja

Depau, 2019-09-12:
I thought you wouldn't like to change the API. If that's okay with you I can submit a PR, otherwise I can just keep it in the wrapper.

Up to you :)

magnusja, 2019-09-12:
This would change the API but it would be perfectly backwards compatible, so that would be fine for me :)

Depau, 2019-10-07:
Alright, I will send you a pull request soon:tm: ;)