magnusja/libaums #209: "Error setting up the device" on QEMU

Depau, 2019-08-23:
I would have tried to troubleshoot this issue on my own but I won't have a lot of time for it in the next days, I'll leave it here in case you do have some time.

The issue doesn't look like a regression as it doesn't work with any build of EtchDroid (both with my old fork of libaums and with upstream + wrapper) but also with your sample app.

As I mentioned in #206 I wasn't able to emulate a USB drive using the standard Android emulator, so I just used stock QEMU with Android-x86.

---

Steps to get the emulator running:
<details>
<summary>Set up using QEMU builds</summary>

- Grab a [Bliss OS QEMU rpm build](https://sourceforge.net/projects/blissos-x86/files/Official/bleeding_edge/QEMU%20builds%20-%20Pie/) (I ended up using Bliss instead of [Android-x86](https://www.android-x86.org) as the guys behind it are maintaining a lot more actively than upstream, though they do mainline their contributions eventually; I've had a bunch of issues with upstream Ax86 that Bliss has fixed already)
- Extract the RPM using [rpmextract](https://www.archlinux.org/packages/extra/any/rpmextract/)
- `cd` into it (`[rpm]/bliss-something`)
- Create virtual disk images for userdata and the emulated USB drive (make sure you use a filesystem that handles sparse files):
```
qemu-img create -f raw data.img 10G
mkfs.ext4 -L userdata data.img

qemu-img create -f raw usb.img 4G
```
- I also `unsquashfs`d the system image in order to remove some bloat

In order to run the emulator with an emulated USB drive, I patched the script that comes with the Ax86 RPM builds: https://gist.github.com/Depau/6c0ccc7377949c68bb30608bef87b9cd

The script expects the disk images to be in the current working directory, so `cd` into the android image directory and run it.
</details>

<details>
<summary>Set up using ISO images</summary>

- Download an Android-x86 or Bliss-x86 ISO
- You need to run the emulator once with the iso to install it, then run it again with the USB drive attached to run tests

```
qemu-img create -f raw hdd.img 10G
qemu-img create -f raw usb.img 4G

# BIOS boot
qemu-system-x86_64 -enable-kvm -boot d -hda hdd.img -m 1024 -display sdl,gl=on -serial mon:stdio -cdrom build.iso
# UEFI boot in case BIOS doesn't boot
# You need to install `ovmf`; paths may vary on your distro; append:
-bios /usr/share/ovmf/x64/OVMF_CODE.fd

# USB 3.0 SCSI bulk only drive; append:
-drive if=none,id=usbstick,file=usb.img,format=raw \
-device nec-usb-xhci,id=xhci \
-device usb-storage,bus=xhci.0,drive=usbstick

# export adb tcp; append
-netdev user,id=mynet,hostfwd=tcp::5555-:5555 -device virtio-net-pci,netdev=mynet
```
When you're done installing, remove `-cdrom`
</details>

This produces an emulated device that looks very close to my real USB drive:
<details>
<summary>lsusb -v (qemu)</summary>

```
$ sudo lsusb -v -d 46f4:

Bus 002 Device 002: ID 46f4:0001  
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               3.00
  bDeviceClass            0 
  bDeviceSubClass         0 
[... 317 lines cut ...]
        at com.github.mjdev.libaums.driver.scsi.ScsiBlockDevice.transferCommand(ScsiBlockDevice.kt:159)
        at com.github.mjdev.libaums.driver.scsi.ScsiBlockDevice.init(ScsiBlockDevice.kt:108)
        at com.github.mjdev.libaums.UsbMassStorageDevice.setupDevice(UsbMassStorageDevice.kt:145)
        at com.github.mjdev.libaums.UsbMassStorageDevice.init(UsbMassStorageDevice.kt:100)
        at com.github.mjdev.libaums.storageprovider.UsbDocumentProvider.addRoot(UsbDocumentProvider.java:416)
        at com.github.mjdev.libaums.storageprovider.UsbDocumentProvider.discoverDevice(UsbDocumentProvider.java:388)
        at com.github.mjdev.libaums.storageprovider.UsbDocumentProvider.access$000(UsbDocumentProvider.java:53)
        at com.github.mjdev.libaums.storageprovider.UsbDocumentProvider$1.onReceive(UsbDocumentProvider.java:287)
        at android.app.LoadedApk$ReceiverDispatcher$Args.lambda$getRunnable$0(LoadedApk.java:1391)
        at android.app.-$$Lambda$LoadedApk$ReceiverDispatcher$Args$_BumDX2UKsnxLVrE6UJsJZkotuA.run(Unknown Source:2)
        at android.os.Handler.handleCallback(Handler.java:873)
        at android.os.Handler.dispatchMessage(Handler.java:99)
        at android.os.Looper.loop(Looper.java:193)
        at android.app.ActivityThread.main(ActivityThread.java:6724)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:495)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:859)
```
</details>

---

From what I've seen, the issue occurs when libaums tries to get the lun size; `bulkInTransfer` times out returning -1, no errno.

Note that I've also experienced the issue on EtchDroid with known working USB drives/Android versions, for example when re-using the same USB drive after the app was stopped for whatever reason without reattaching the USB drive and killing the app.

magnusja, 2019-08-26:
The LUN size is queried using a `controlTransfer`. Do you mean that or the very first `bulkTransfer` when initializing the device?

I have seen this error as well but I did not find any solution except replugging the device (https://github.com/magnusja/libaums/issues/95#issuecomment-318274183)

Depau, 2019-08-27:
[image]
[image]

Depau, 2019-08-27:
I blindly tried to run it again using the debugger and this is what I get, for what it matters:

[image]

I haven't checked what exactly that command is supposed to be doing, but since this is an emulator I guess I can try stuff :man_shrugging: haha

Depau, 2019-08-27:
This might be related (page 155): https://www.seagate.com/files/staticfiles/support/docs/manual/Interface%20manuals/100293068j.pdf

[image]

I'll see if I can implement and test the 16 byte command.

magnusja, 2019-08-27:
Not sure if that is the problem here. Every device should be able to support the READ CAPACITY(10) command. Also I would assume that it responds with an error and not time out.

What does the version field of the SCSI Inquiry Response say?

Depau, 2019-08-27:
```
D/ScsiBlockDevice: inquiry response: ScsiInquiryResponse [peripheralQualifier=0,
peripheralDeviceType=0, removableMedia=false, spcVersion=5, responseDataFormat=2]
```

magnusja, 2019-08-27:
Ok I see. That sounds more like the error I mentioned earlier. Some people suggested that sleeping for a couple of seconds helped in between sending commands to the device.

magnusja, 2019-08-27:
What might also be worth trying if you can get total commander or USB Media explorer or similar running. Then we know that it has to work somehow...

Depau, 2019-08-27:
Could it be worth implementing something similar to Linux sd driver's spinup sequence?

https://github.com/torvalds/linux/blob/master/drivers/scsi/sd.c#L2066

It does seem to include some quirks for weird USB flash drives which actually do not require to be spun up.

It looks that in this particular case it simply waits one second between TEST UNIT READY and returning (and then it goes on with READ CAPACITY (16) which falls back to READ CAPACITY (10)).

Reimplementing the whole thing in Kotlin might be a big effort considering that it considers a lot of different cases.

For now I'll see what happens if I `Thread.sleep(1000)` after `ScsiTestUnitReady`.

Depau, 2019-08-27:
Nope, it still fails with the 1 sec delay and READ CAPACITY 16

Depau, 2019-08-28:
I attached a real USB drive to QEMU and analyzed stuff with Wireshark. The USB drive worked for a while, then the app crashed because of filesystem issues.

Which means that what happens next is `IOException: Could not write to device, result == -1 errno 0 null`

So, this is what I see on Wireshark when this occurs:

[app is started]
[image]
[communication times out]
[app crashes]

In this case, the app times out at `transferCommand` for `ScsiInquiry`:

```
D/UsbMassStorageDevice: setup device
I/UsbMassStorageDevice: MAX LUN 0
E/MainActivity: error setting up device
    java.io.IOException: Could not write to device, result == -1 errno 0 null
        at com.github.mjdev.libaums.usb.JellyBeanMr2Communication.bulkOutTransfer(JellyBeanMr2Communication.kt:34)
        at com.github.mjdev.libaums.driver.scsi.ScsiBlockDevice.transferCommand(ScsiBlockDevice.kt:144)
        at com.github.mjdev.libaums.driver.scsi.ScsiBlockDevice.init(ScsiBlockDevice.kt:75)
        at com.github.mjdev.libaums.UsbMassStorageDevice.setupDevice(UsbMassStorageDevice.kt:145)
        at com.github.mjdev.libaums.UsbMassStorageDevice.init(UsbMassStorageDevice.kt:100)
        at com.github.mjdev.libaums.usbfileman.MainActivity.setupDevice(MainActivity.java:761)
        at com.github.mjdev.libaums.usbfileman.MainActivity.selectDevice(MainActivity.java:717)
        at com.github.mjdev.libaums.usbfileman.MainActivity.access$600(MainActivity.java:99)
        at com.github.mjdev.libaums.usbfileman.MainActivity$4.onItemClick(MainActivity.java:681)
[... unrelated]
```

And this is the regular flow before the app crashes (except for the READ CAPACITY (10) command which I have replaced with READ CAPACITY (16)):
[image]

Depau, 2019-08-28:
Alright, in order to capture the specific case of the emulated USB drive on Wireshark, I did a qemu-inception: I booted a GNU/Linux VM with the emulated USB drive attached, then booted the Android VM inside of it and connected the emulated drive as a host USB device.

[image]

Wireshark pcap: [qemu-usb-capture.pcapng.gz](https://github.com/magnusja/libaums/files/3549550/qemu-usb-capture.pcapng.gz)

If you check the timestamps you can clearly see when the app was started.

I'll look into all this info, in the meantime I'll just drop this here in case it might be useful.

Depau, 2019-08-28:
I think I found the issue. Apparently, the device is STALLing the transfer
[image]
https://www.kernel.org/doc/html/latest/driver-api/usb/URB.html#how-to-submit-an-urb

[This library](https://github.com/KunYi/libusb_scsi_command/blob/master/sb.c#L192) apparently works around this by simply clearing the fault and retrying up to 5 times.

[image]

I'm not sure how to do it on Android, but I will look into it in the next few days.

magnusja, 2019-08-28:
Hey there, I had a look at the file. But I am not sure what exactly is happening.
Interesting are the URB errors
1. URB status: Operation now in progress (-EINPROGRESS) (-115)
2. URB status: No such file or directory (-ENOENT) (-2)

For BULK transfer first and then also for control transfer. What that really means I do not know.

Did you have time to try Total Commander?

magnusja, 2019-08-28:
For some reason Wireshark does not parse the bulk in transfers from device to host to USBMS protocol frames.

magnusja, 2019-08-28:
Sorry did not see your latest comment. 

Clearing the halt seems to be fairly low level and might not be supported in Android. Maybe releasing the USB interface/device altogether can help here. At least it sounds different from https://github.com/magnusja/libaums/issues/95#issuecomment-318274183 so there is a chance that this might be possible.

Thanks for the support I really appreciate it!

Depau, 2019-08-28:
> Hey there, I had a look at the file. But I am not sure what exactly is happening.
> Interesting are the URB errors
> 
>     1. URB status: Operation now in progress (-EINPROGRESS) (-115)

That doesn't seem to be an error. See this capture from a valid communication (with the kernel):

[after-vm-detach.pcapng.gz](https://github.com/magnusja/libaums/files/3550087/after-vm-detach.pcapng.gz)

> 
>     2. URB status: No such file or directory (-ENOENT) (-2)

https://www.kernel.org/doc/html/latest/driver-api/usb/error-codes.html

I also noticed that the interface is set to 0xFFFF in our requests, which also seems to be the reason WireShark is not parsing the payloads as they look just fine to me.

> 
> 
> For BULK transfer first and then also for control transfer. What that really means I do not know.
> 
> Did you have time to try Total Commander?

No, I haven't purchased it.

magnusja, 2019-08-28:
I think Total Commander with the USB plugin is free (or at least some test version)

magnusja, 2019-08-28:
https://play.google.com/store/apps/details?id=de.hechler.tcplugins.usbstick
https://play.google.com/store/apps/details?id=com.ghisler.android.TotalCommander&hl=en

Depau, 2019-08-28:
> Sorry did not see your latest comment.
> 
> Clearing the halt seems to be fairly low level and might not be supported in Android. Maybe releasing the USB interface/device altogether can help here. At least it sounds different from [#95 (comment)](https://github.com/magnusja/libaums/issues/95#issuecomment-318274183) so there is a chance that this might be possible.
> 
> Thanks for the support I really appreciate it!

Reading this it seems to be just a matter of sending a controlTransfer command: https://developer.electricimp.com/resources/usberrors

However, checking libusb's source code it seems it can be done with an ioctl:
https://github.com/libusb/libusb/blob/958ec5255ba5707197486dfe4827d153f48e91ba/libusb/os/linux_usbfs.c#L1604

Which we can do, given that UsbCommunication exposes the underlying file descriptor (which can also be passed to libusb eventually)

magnusja, 2019-08-28:
> > ```
> > 2. URB status: No such file or directory (-ENOENT) (-2)
> > ```
> 
> https://www.kernel.org/doc/html/latest/driver-api/usb/error-codes.html
> 

I see, so this actually seems to indicate that the endpoint is stalled currently and not available for communication.

magnusja, 2019-08-28:
> Reading this it seems to be just a matter of sending a controlTransfer command: https://developer.electricimp.com/resources/usberrors

This might be convenient because sending a controlTransfer is possible within the Android API.

> However, checking libusb's source code it seems it can be done with an ioctl:
> https://github.com/libusb/libusb/blob/958ec5255ba5707197486dfe4827d153f48e91ba/libusb/os/linux_usbfs.c#L1604
>
> Which we [...]

Which also means we might be able to just use libusb :D

Depau, 2019-08-28:
I can confirm it works with Total Commander.

... a little reverse engineering never killed nobody :man_shrugging: 

> Which also means we might be able to just use libusb :D

That's what I was thinking. I was also thinking that with enough infrastructure and wrappers in place we might just be able to link Linux's `sd` + `scsi` driver to libusb and use that instead, and enjoy the stability of heavily production tested code. But I haven't investigated how hard it would be, considering that the Linux USB stack is a completely separate driver and that shims for it need to be written.

Also this would mean that GPLv2 code would have to be embedded in this Apache-licensed library, which would require a relicensing to GPLv2 or GPLv3 unless I missed something.

magnusja, 2019-08-28:
Ok awesome, then we just need to figure out what exactly they do differently.

I feel like this might a fairly big change especially designing it in a way that we can easily merge upstream back to libaums every now and then. That is why I would suggest go for the control transfer first and then maybe try `ioctl`

magnusja, 2019-08-28:
libusb even has a readme for Android https://github.com/libusb/libusb/tree/master/android

Depau, 2019-08-28:
> I feel like this might a fairly big change especially designing it in a way that we can easily merge upstream back to libaums every now and then. That is why I would suggest go for the control transfer first and then maybe try `ioctl`

That's the hardest thing. I quickly checked how all the components talk to each other and it's quite messy, it's likely not a matter of copying files and wrapping them.

> libusb even has a readme for Android https://github.com/libusb/libusb/tree/master/android

Yep, but they don't provide real JNI wrappers and we would have to write them on our own. We'd need to see how bad Android's API gets in our way, because if we need just a bunch of ioctls we might just as well duct-tape them onto the API.

Depau, 2019-08-28:
I opened the USB plugin for Total Commander on Jadx and it looks like it's completely unobfuscated.

Also there are strings that refer to libusb, but I haven't seen anything JNI related so it might be just bad naming.

Anyway, that's it for me for today, I should really be doing other things right now. I should free in two weeks and I will gladly look deeper into it :)

magnusja, 2019-08-28:
Interesting, if that is really the case we might even be able to ask them for the source code, given they linked libusb statically (LGPL).

magnusja, 2019-09-19:
Hey there, it seems that your note server is down :/

Depau, 2019-09-20:
Hi, thanks for catching that, I'm taking it back up.

magnusja, 2019-12-11:
btw, I talked to the developer of Nexus Media Importer a couple of years back:

1. UsbDeviceConnection.bulkTransfer is sometimes returning -1, so no bytes have been transferred. After that only re-plugging device helps. I can reproduce this pretty well when I try to communicate with multiple mass storages devices using an USB hub.

His reply

Yes, I'm painfully aware of #1.  According to the Mass Storage spec, you should be able

>  to a Bulk-Only Mass Storage Reset and recover.  Unfortunately, it doesn't work or at least I could never get it work.  What I learned later on is that there is the Mass Storage spec and what Windows does.  A great number of (cheap) devices only work reliably if you do what Windows does.  FYI:  Windows does a port reset to recover from this.  I couldn't figure out how to do that without root.

Maybe this helps at some point :D

Depau, 2019-12-19:
Thanks for pointing that out. I think I read about the Windows behavior somewhere. BTW libusb does have a function to perform port reset.

As I said in the other thread I won't be working on this until the next year, but I did start tinkering on it last time I got the chance, I'll see if I can find some diffs in the meantime.

Depau, 2020-01-12:
Here's what I use to run Android on QEMU: https://gist.github.com/Depau/77b06b18320f511db8fdae58e8bceead

Note that I'm using Android v8.1-r2, not the latest one, I'll test it 9 too.

magnusja, 2020-01-13:
```
D/ScsiBlockDevice: Reset bulk-only mass storage
W/ScsiBlockDevice: sending bulk only mass storage request
D/ScsiBlockDevice: Trying to clear halt on both endpoints
W/native_libusbcom: jint Java_me_jahnen_libaums_libusbcommunication_LibusbCommunication_nativeClearHalt(JNIEnv *, jobject, jlong, jint):98 libusb clear halt
D/LibusbCommunication: libusb clearFeatureHalt returned -5
W/native_libusbcom: jint Java_me_jahnen_libaums_libusbcommunication_LibusbCommunication_nativeClearHalt(JNIEnv *, jobject, jlong, jint):98 libusb clear halt
D/LibusbCommunication: libusb clearFeatureHalt returned -5
D/ScsiBlockDevice: inquiry response: ScsiInquiryResponse [peripheralQualifier=0, peripheralDeviceType=0, removableMedia=false, spcVersion=5, responseDataFormat=2]
E/native_libusbcom: jint Java_me_jahnen_libaums_libusbcommunication_LibusbCommunication_nativeBulkTransfer(JNIEnv *, jobject, jlong, jint, jbyteArray, jint, jint, jint):73 libusb_bulk_transfer returned -7
E/ScsiBlockDevice: Error transferring command; errno 0 null
D/ScsiBlockDevice: Reset bulk-only mass storage
W/ScsiBlockDevice: sending bulk only mass storage request
D/ScsiBlockDevice: Trying to clear halt on both endpoints
W/native_libusbcom: jint Java_me_jahnen_libaums_libusbcommunication_LibusbCommunication_nativeClearHalt(JNIEnv *, jobject, jlong, jint):98 libusb clear halt
D/LibusbCommunication: libusb clearFeatureHalt returned -5
W/native_libusbcom: jint Java_me_jahnen_libaums_libusbcommunication_LibusbCommunication_nativeClearHalt(JNIEnv *, jobject, jlong, jint):98 libusb clear halt
```
an inquiry response at some point is all I can get :/

magnusja, 2020-01-13:
When using genymotion and a real usb pen drive I get this

```
D/ScsiBlockDevice: inquiry response: ScsiInquiryResponse [peripheralQualifier=0, peripheralDeviceType=0, removableMedia=false, spcVersion=4, responseDataFormat=2]
```
Maybe there is also some protocol version libaums does not support which is the problem

Depau, 2020-01-13:
> Maybe there is also some protocol version libaums does not support which is the problem

There is indeed another protocol and that is UAS which should be what is used by hard drive caddies, IDK how it's different from regular mass storage devices protocol. QEMU can emulate that too.

Anyway, is that different from what you get if you connect it directly to an Android device?

magnusja, 2020-01-13:
I meant that the spc/sbc version (scsi block commads) is different, but I did some research and libaums is only using the bare minimum which should be supported by every device. The only thing I noticed is that with spc>=5 there is a `REPORT LUNS` command, maybe the control transfer to find out the luns is a problem then. Although I doubt that ..

Depau, 2020-01-13:
For what it's worth, [here](https://www.systutorials.com/docs/linux/man/8-sg_luns/#lbAF) it says:

> The SCSI REPORT LUNS command is important for Logical Unit (LU) discovery. After a target device is discovered (usually via some transport specific mechanism), a REPORT LUNS command should either be sent to LUN 0 (which is Peripheral device addressing method with bus_id=0 and target/lun=0) or to the REPORT LUNS well known LUN (i.e. 0xc101000000000000). SAM-5 requires that one of these responds with an inventory of LUNS that are contained in this target device.

I haven't read the specifications not I can test it at the moment so I don't know if this may actually be useful, I'll leave it here just in case.

Depau, 2020-07-18:
I've got some free time to work on libaums :tada: (but just for a few weeks)

I'll try to work on QEMU testing again, hopefully it will also resolve the issues on real devices as we predicted.

commit d31c5378 (2020-07-20): Only request `transferLength` bytes from device

Solves issue with QEMU <= 5.0.0, it might also (hopefully) solve some issues with real devices

Closes #209
closed 2020-07-21 by magnusja