magnusja/libaums #211: CopyTask function copies empty file from usb to internal storage.

SynergySaugat, 2019-08-28:
This is the function from where file is copied 
UsbMassStorageDevice[] devices = UsbMassStorageDevice.getMassStorageDevices(getActivityContext() /* Context or Activity */);
        for (UsbMassStorageDevice device : devices) {

            // before interacting with a device you need to call init()!
            //considering only one device is inserted in every case
            try {
                device.init();
                // Only uses the first partition on the device
                FileSystem currentFs = device.getPartitions().get(0).getFileSystem();
                UsbFile folder = currentFs.getRootDirectory().search(sBackupFolder).search(folderName);
                if (folder == null) {
                    Utilities.Toaster(getActivityContext(), "Backup folder not found");
                } else {
                    Utilities.Toaster(getActivityContext(), "Backup folder found. Checking for files");
                    folder.listFiles();
                    int fileCount = 0;
                    List<String> list = new ArrayList<>();
                    if (folder.list() != null) {
                        for (int i = 0; i < folder.listFiles().length; i++) {
                            UsbFile backupFile = folder.listFiles()[i];
                            if (backupFile.getName().equals("DB1V" + DaoMaster.SCHEMA_VERSION + "Backup.db") || backupFile.getName().equals("DB2V" + 3 + "Backup.db")) {
                                fileCount++;
                            }
                        }
                        if (fileCount < 2)
                            Utilities.Toaster(getActivityContext(), "Backup files missing");
                        else {
                            //todo copy db to desired location
                            Utilities.Toaster(getActivityContext(), "Backup files found");
                            for (int i = 0; i < folder.listFiles().length; i++) {
                                UsbFile backupFile = folder.listFiles()[i];
                                if (backupFile.getName().equals("DB1V" + DaoMaster.SCHEMA_VERSION + "Backup.db")) {
                                    //copy greenDB to app folder
                                    CopyTaskParam sourceDestination = new CopyTaskParam();
//                                    File dbFile = new File(getActivityContext().getDatabasePath("green-db").getPath());
                                    File dbFile = new File(Environment.getExternalStorageDirectory()+"/green.db");
                                    sourceDestination.from = backupFile;
                                    sourceDestination.to = dbFile;

                                    new CopyTask(getActivityContext()).execute(sourceDestination);

                                }
                              /*  if (backupFile.getName().equals("DB2V" + 3 + "Backup.db")) {
                                    //copy sugarDB to app folder
                                    CopyTaskParam sourceDestination = new CopyTaskParam();
                                    File dbFile = new File(getActivityContext().getDatabasePath("syntech_tabletbanking.db").getPath());
                                    sourceDestination.from = backupFile;
                                    sourceDestination.to = dbFile;
                                    new CopyTask(getActivityContext()).execute(sourceDestination);
                                }*/
                            }
                        }

                    } else {
                        Utilities.Toaster(getActivityContext(), "Backup files missing");
                    }
                    setupRecyclerView(list);
                }
                device.close();
            } catch (IOException | NullPointerException e) {
                e.printStackTrace();
                new Handler().postDelayed(new Runnable() {
                    @Override
                    public void run() {
                        Toast.makeText(getActivityContext(), "Restore failed " + e.toString(), Toast.LENGTH_SHORT).show();
                        unregisterReceiver(mUsbReceiver);
                        dialog.dismiss();
                    }
                }, 3000);
            }
        }

This is the file I created from your reference.

public class CopyTask extends AsyncTask<CopyTaskParam, Integer, Void> {

    private ProgressDialog dialog;
    private CopyTaskParam param;
    private Context context;

    public CopyTask(Context context) {
        this.context = context;
        dialog = new ProgressDialog(context);
        dialog.setTitle("Copying file");
        dialog.setMessage("Copying a file to the internal storage, this can take some time!");
        dialog.setIndeterminate(true);
        dialog.setCancelable(false);
    }

    @Override
    protected void onPreExecute() {
        dialog.show();
    }

    @Override
    protected Void doInBackground(CopyTaskParam... params) {
        long time = System.currentTimeMillis();
        param = params[0];
        try {
            OutputStream out = new BufferedOutputStream(new FileOutputStream(param.to));

            Log.d("TAG", "Copy file with length: " + param.from.getLength());
            out.write(convertIntoBytes(param.from.getAbsolutePath()));
            out.close();
        } catch (IOException e) {
            Log.e("TAG", "error copying!", e);
        }
        Log.d("TAG", "copy time: " + (System.currentTimeMillis() - time));
        return null;
    }

    @Override
    protected void onPostExecute(Void result) {
        dialog.dismiss();
        Utilities.Toaster(context, "Database Restored");
    }

    private byte[] convertIntoBytes(String db) {
        try {
            File dbFile = new File(db);
            FileInputStream fis = new FileInputStream(dbFile);
            // Transfer bytes from the input file to the output file
            byte[] bytesArray = new byte[(int) dbFile.length()];
            fis.read(bytesArray);
            fis.close();
            return bytesArray;
        } catch (IOException e) {
            e.printStackTrace();
            return null;
        }
    }
}

magnusja, 2019-08-28:
What means means empty?

If you look at the documentation (https://docs.oracle.com/javase/7/docs/api/java/io/FileInputStream.html#read(byte[])) you realize that you cannot read a whole file like you want to. The documentation says that it reads up to byte array length but most likely much less. That is why you want to use a for loop or something like apache commons `copyStream`

SynergySaugat, 2019-08-29:
@magnusja The file is written but its size is 0 byte. No data is written only the file exists..

if i use loop instead of converting the file into bytes then will that work..?

older method : 
 out.write(convertIntoBytes(param.from.getAbsolutePath()));

and newer (your code)
{
   OutputStream out = new BufferedOutputStream(new FileOutputStream(param.to));
         InputStream inputStream = new UsbFileInputStream(param.from);

            byte[] bytes = new byte[param.chunkSize];
            int count;

            while ((count = inputStream.read(bytes)) != -1) {
                out.write(bytes, 0, count);
            }
//            out.write(convertIntoBytes(param.from.getAbsolutePath()));
            out.close();
}

SynergySaugat, 2019-08-29:
Using the above method still copies 0 byte file.. I cannot understand what is wrong here. I copied methods from your example app..

magnusja, 2019-08-29:
Can you log the count variable?

SynergySaugat, 2019-08-29:
Ok will see that and let u know in a while :D

SynergySaugat, 2019-08-29:
E/TAG: error copying!
    java.io.IOException: Could not write to device, result == -1 errno 16 Device or resource busy
        at com.github.mjdev.libaums.usb.JellyBeanMr2Communication.bulkOutTransfer(JellyBeanMr2Communication.java:45)
        at com.github.mjdev.libaums.driver.scsi.ScsiBlockDevice.transferCommand(ScsiBlockDevice.java:143)
        at com.github.mjdev.libaums.driver.scsi.ScsiBlockDevice.read(ScsiBlockDevice.java:209)
        at com.github.mjdev.libaums.driver.ByteBlockDevice.read(ByteBlockDevice.java:67)
        at com.github.mjdev.libaums.fs.fat32.FAT.getChain(FAT.java:134)
        at com.github.mjdev.libaums.fs.fat32.ClusterChain.<init>(ClusterChain.java:64)
        at com.github.mjdev.libaums.fs.fat32.FatFile.initChain(FatFile.java:90)
        at com.github.mjdev.libaums.fs.fat32.FatFile.read(FatFile.java:153)
        at com.github.mjdev.libaums.fs.UsbFileInputStream.read(UsbFileInputStream.java:84)
        at com.syntech.tabletbanking.dbbackupmanagement.CopyTask.doInBackground(CopyTask.java:58)
        at com.syntech.tabletbanking.dbbackupmanagement.CopyTask.doInBackground(CopyTask.java:25)
        at android.os.AsyncTask$2.call(AsyncTask.java:288)
        at java.util.concurrent.FutureTask.run(FutureTask.java:237)
        at android.os.AsyncTask$SerialExecutor$1.run(AsyncTask.java:231)
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1112)
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:587)
        at java.lang.Thread.run(Thread.java:841)

magnusja, 2019-09-02:
This happens immediately after writing? This might be a known error, I do not have any solution unfortunately.
Can you try a different usb pen drive?

SynergySaugat, 2019-09-02:
currently i have solved the issue from a different approach... I converted the UsbFile into bytearray ani wrote it into internal storage so it works now.. Thankyou for your feedback :D
I read your thesis also.. Had a great help from there too.

magnusja, 2019-09-02:
Ok interesting. Then I will close this for now, reopen if you encounter this again.

closed 2019-09-02 by magnusja