magnusja/libaums #212: unavailable for exFAT

linzh93, 2019-08-29:
hi ,  I get your demo and build it, but when the udisk is exFAT, it can not be checked and read，but i saw your project has this partitionTypes, what shall we do to solve or use it , thank you

magnusja, 2019-09-02:
If it is MBR and exFAT it should work if you actvate `java-fs`. I you are doing that if you use my demo app?

Adrian-Rosoga, 2019-09-07:
Hi Magnus.

I have encountered the same issue with Usbfileman with FAT or exFAT USB sticks.
See below the stacktrace.

What exactly your suggestion "activate java-fs" means? The exception is thrown from the javafs module - is this related to java-fs you mention? I have just gave Usbfileman a go today, didn't dig much into the code. Working with filesystems on USB devices seems involving, I was expecting a walk in the park but it's not the case ;).

Many thanks,

Adrian

```
E/AndroidRuntime: FATAL EXCEPTION: main
    Process: com.github.mjdev.usbfileman, PID: 562
    java.lang.NoSuchMethodError: No virtual method position(I)Ljava/nio/ByteBuffer; in class Ljava/nio/ByteBuffer; or its super classes (declaration of 'java.nio.ByteBuffer' appears in /system/framework/core-oj.jar)
        at org.jnode.fs.exfat.ExFatSuperBlock.read(ExFatSuperBlock.java:72)
        at org.jnode.fs.exfat.ExFatFileSystem.<init>(ExFatFileSystem.java:46)
        at org.jnode.fs.exfat.ExFatFileSystemType.create(ExFatFileSystemType.java:37)
        at org.jnode.fs.exfat.ExFatFileSystemType.create(ExFatFileSystemType.java:32)
        at com.github.magnusja.libaums.javafs.JavaFsFileSystemCreator.read(JavaFsFileSystemCreator.java:73)
        at com.github.mjdev.libaums.fs.FileSystemFactory.createFileSystem(FileSystemFactory.kt:57)
        at com.github.mjdev.libaums.partition.Partition$Companion.createPartition(Partition.kt:87)
        at com.github.mjdev.libaums.UsbMassStorageDevice.initPartitions(UsbMassStorageDevice.kt:173)
        at com.github.mjdev.libaums.UsbMassStorageDevice.setupDevice(UsbMassStorageDevice.kt:158)
        at com.github.mjdev.libaums.UsbMassStorageDevice.init(UsbMassStorageDevice.kt:100)
        at com.github.mjdev.libaums.storageprovider.UsbDocumentProvider.addRoot(UsbDocumentProvider.java:416)
...
```

magnusja, 2019-09-09:
Can you try compiling with target Java version 8 instead of 9?

See
https://github.com/apache/felix/pull/114
https://github.com/sannies/mp4parser/issues/323

azrael8576, 2019-09-10:
Hi 
How can I actvate java-fs in this project?

Adrian-Rosoga, 2019-09-10:
Hi Magnus,

I will get up to speed again with Android development after a break of some time. I am giving you below details about my setup. Can Android Developer even target Java 9?

Target Compatibility shows as options: 1.8 (Java 8), 1.7 (Java 7), 1.6 (Java 6) and Kotlin 1.3. Same for Source Compatibility.

I tried with Target Java 7 - same issue. No stack trace as the debugger was misbehaving.

My questions and some comments:
- Was exFAT tested and blessed? What was the setup?
- Github says "Currently it supports the SCSI command set and the FAT32 file system.", No mention of exFAT or FAT. Documentation behind?
- If answer is yes, can Usbfileman apk be made available somehow so that I can try with exFAT on my phone? If it works I will dig further to find out if there's a build problem or whatever.
- Should maybe try with Kotlin build with exFAT?
- Usbfileman is not verbose when it crashes. Maybe I can contribute here to add some info. If added on Google Play it may send back crash info to you. That could help.
- Stick with the master branch? Or have you pushed stuff to have on the develop one?

Thanks for your work, it was an eye opener, I was expecting a walk in the park but looks like user-space fs support is not trivial ;).

My setup:
Android Studio 3.5
Build #AI-191.8026.42.35.5791312, built on August 9, 2019
JRE: 1.8.0_202-release-1483-b03 amd64
JVM: OpenJDK 64-Bit Server VM by JetBrains s.r.o
Windows 10 10.0
- Using SDK and JRE that come with Android Studio
- Phone is Samsung Galaxy S7
- USB stick formatted on Windows 10, tried with Nikon S6 XQD card as well, same issue

magnusja, 2019-09-10:
@azrael8576 Just have a look in the example app code, it us being used there.

@Adrian-Rosoga Good point. Have not tried it in a long time. exFAT is highly experimental probably has performance issues and is pretty buggy. Use at your own risk. This might be a problem with my release of java-fs because I compiled with JDK 11 ... Will try tomorrow

azrael8576, 2019-09-11:
@magnusja 
Sorry, I didn't describe it in detail.
I successfully built Usbfileman app.
FAT32 file system is work.
But can't work in exFat.
I see JavaFsFileSystemCreator Class has this ExFatFileSystemType from javafs module

magnusja, 2019-09-11:
What does the exception say?

azrael8576, 2019-09-12:
java.io.IOException: Could not read from device, result == -1 errno 0 null

HBiSoft, 2019-10-04:
@magnusja 

You can cast `ByteBuffer` to `Buffer` then you can compile the project with JDK9 and have it running on JDK8.

dashifuguye, 2019-10-29:
> java.io.IOException: Could not read from device, result == -1 errno 0 null

I get the same error with you, have you fix it when read exFAT?

shenqinwei, 2019-11-07:
upgraded to the version 0.7.1 and exfat still doesn't work

magnusja, 2019-12-03:
see https://github.com/magnusja/libaums/issues/235#issuecomment-561121268 for a fix

closed 2019-12-03 by magnusja

Wakudoki, 2020-04-16:
Excuse me, is there a plan to release a version which supports exFAT recent?

magnusja, 2020-04-16:
java-fs you have to compile yourself, there is no plan to release this anytime soon as it is highly experimental

trihoang3011, 2020-05-04:
@magnusja 

I have tested with java-fs but it seems that implementation details with exfat are still not completed. 

[https://github.com/magnusja/java-fs/blob/master/src/main/java/org/jnode/fs/exfat/NodeFile.java](https://github.com/magnusja/java-fs/blob/master/src/main/java/org/jnode/fs/exfat/NodeFile.java
)
With the NodeFile: Write file or flush file are still missing.

[https://github.com/magnusja/java-fs/blob/master/src/main/java/org/jnode/fs/exfat/NodeDirectory.java](https://github.com/magnusja/java-fs/blob/master/src/main/java/org/jnode/fs/exfat/NodeDirectory.java)
Similar to NodeDirectory which lacks the add file, add directory, remove file or directory. 

Do you have any support for these implementations?

magnusja, 2020-05-04:
Hey there,

yes you are right a lot of the file systems are read only. I have currently no plans to improve this situation. But I am happy to take contributions.

MestevInge, 2020-09-16:
Hello,

Do you have any solution for writting with extFat system ?
How do work Paragon ? We can mount/unmount partition and R/W easily with the File classe

Depau, 2020-09-16:
I'd also like to add that exfat is technically proprietary by Microsoft and there are patents on it. The patents have been granted for inclusion in the Linux kernel, however that doesn't imply that they're also granted for other projects.

(Notice: IANAL) Since the Linux kernel code is GPL2-licensed, you could technically take it out, make it work outside of Linux and license it as a GPL2 project as well. This must be true because the GPL2 explicitly grants that and Microsoft agreed to grant the patents for a GPL2-licensed project.

However, I'm pretty sure you **cannot** link an Apache 2 project to a GPL2 project, only LGPL2 with the restrictions that apply to libusbcommunication (see https://github.com/magnusja/libaums/tree/develop/libusbcommunication#licensing).

The only way that such a thing would be feasible (and I'm not even sure) would be to place a standalone binary in the shared storage, with all the security considerations (guys, please don't even try).

Another safer way would be to license the entire app as GPL2/3, build the exfat implementation separately, then plug the implementation into libaums. That way the entire app can be rebuilt and replaced by the users.

I'm not sure an exfat implementation would be legal in any other way, unless Microsoft blesses us or royalties are paid.