magnusja/libaums #215: Creating multiple files on SD card fails if all files are created before the first one is written

Derpalus, 2019-09-11:
My SD card (Fat32) has a folder structure like this:
```
Root
|   File1.txt
|
+---Folder1a
|   \---Folder1b
|           File1b1.txt
|           File1b2.txt
|           File1b3.txt
|
\---Folder2a
        File2a.txt
```

However, there are issues with files 1b1 and 1b2 not getting written.

It seems like if you create all the files at the start of the function, and then write to them later every file except the last one written to each directory fails to write.

The code is essentially

1. Open device
1. Create Folder1a
1. Create Folder1b
1. Create File1b1
1. Create File1b2
1. Create File1b3
1. Create Folder2a
1. Create File2a
1. Create File1
1. Write File1b1 (failure)
1. Write File1b2 (failure)
1. Write File1b3 (success)
1. Write File2a (success)
1. Write File1 (success)
1. Close device

The same happens if the directories and files already exist and there is just an attempt to overwrite them.

What you HAVE to do is to create or access the file just before you intend to write to it. I'm speculating that a `UsbFile` essentially stores a "pointer" to a location in memory and if 3 files are created before any are written, they all point to the same location. Each file gets written in turn, but the next file just overwrites whatever was written by the previous. But I might be wrong.

If this can't be handled, at least there should be an instruction about it or something.

magnusja, 2019-09-11:
Hey there,

thanks for reporting. This is definitely a bug. I tried to cover this with a test case: https://github.com/magnusja/libaums/pull/216/files

But tests succeed, so there must be something that I do differently. Please have a look and let me know.

magnusja, 2019-09-13:
Hey there,

so I now what the issue is but I do not know how to solve this.

The problem is that with your `getFile` method you are creating new files which point to a different `parent` instance. This parent instance is used to write the file length after closing the OutputStream. Because of the different instances of the parent which only do know about the file length of one particular file, only the last file length gets written to disk.

Derpalus, 2019-09-13:
Well, I can live with limitations like that, it's just important to indicate this somewhere in the documentation where it won't be missed. Other than that, perhaps the parent instance should be a singleton or something like that? There is only one file system right, no reason to have more than one instance?

magnusja, 2019-09-16:
Well no one noticed this so far, that is why it is not in the documentation. 
From a user perspective I think it is confusing that this is not working, so I will try to think of a fix. The singleton might go into the right direction, but it is more complex than having a singleton instance.

magnusja, 2019-09-18:
Hey @Derpalus 

thanks again for reporting this issue and also #219 

I released v0.7.1 where both of these should be fixed. Let me know if there is anything else I can help with.

closed 2019-09-18 by magnusja