magnusja/libaums #219: UsbMassStorageDevice partitions not resetting

Derpalus, 2019-09-18:
See the following code:

```Java
UsbMassStorageDevice device = getDevice();
device.init();
// Do stuff
device.close();
device.init(); // The device now has twice as many partitions
// Do stuff
device.close();
```

You add the partition table on each init without removing the old partition table. I don't know if it's not allowed to re-init a device that has already been initialized? In that case that should be mentioned in the instructions.

I do it this way because I don't want to have the device claimed when I'm not using it, but still want to keep it in memory.

magnusja, 2019-09-18:
Hey there, thanks for reporting.
What version are you using?

Derpalus, 2019-09-18:
0.7.0

magnusja, 2019-09-18:
Ok. I will release a big fix release soonish. This is definitely fixed on develop.

closed 2019-09-18 by magnusja