magnusja/libaums #222: SLR CAMERA

zq798778610, 2019-10-15:
### Problem
use Canon SLR CAMERA  connect phone ，show  no device  
 UsbMassStorageDevice[] devices=UsbMassStorageDevice.getMassStorageDevices(this);
        Log.i("zqq", "devices: " + devices.length);
 devices.length = 0 
support SLR CAMERA？
### Expected behavior

### Actual behavior

#### Stacktrace of Excpetion

```
(if exists)
```

### Code where problem occurs

magnusja, 2019-10-15:
Hey There,

this is not supported because it is a completely different protocol. See https://en.wikipedia.org/wiki/Media_Transfer_Protocol and https://en.wikipedia.org/wiki/Picture_Transfer_Protocol.

Happy about contributions regarding that though ...

zq798778610, 2019-10-17:
Thankyou

xiaocainiaooo, 2021-09-27:
Ptp support has other lib to support. android Framework had MTP code ,if you want to use. you can get this way @zq798778610