magnusja/libaums #223: Read a micro:bit

kingiis, 2019-10-25:
using genymotion & virtaulbox  
read micro:bit  as usb 
i got the basic info
i got the permission  to acces usb 
but when i try to read  ,its stop at a ctach

  public ArrayList<UsbFile> readDevice(UsbMassStorageDevice device) {
        ArrayList<UsbFile> usbFiles = new ArrayList<>();
        try {
            //初始化
            device.init();
            //获取partition
            Partition partition = device.getPartitions().get(0);
            FileSystem currentFs = partition.getFileSystem();
            //获取根目录
            UsbFile root = currentFs.getRootDirectory();
            currentFolder = root;
            //将文件列表添加到ArrayList中
            Collections.addAll(usbFiles, root.listFiles());
        } catch (Exception e) {
            e.printStackTrace();
            Log.i("tag","异常"+e.toString());
        }
        return usbFiles;
    }
com.github.mjdev.libaums.partition.PartitionTableFactory$UnsupportedPartitionTableException
    libraryVersion = '0.7.1'

magnusja, 2019-10-25:
Make sure that the drive is formatted with MBR and FAT32

magnusja, 2019-10-25:
FAT32 only is also fine. Maybe connect that thing to a PC or mac and check how exactly it is formatted.

kingiis, 2019-10-26:
Thanks, I finally found that the virtual USB provided by the chip is fat, but not FAT32.

The function is OK when debugging the normal USB

closed 2019-10-26 by kingiis