magnusja/libaums #224: How to load images file with glide

leestars, 2019-11-12:


magnusja, 2019-11-12:
For questions about glide please refer to this repository: https://github.com/bumptech/glide

closed 2019-11-12 by magnusja

leestars, 2019-11-12:
Sorry, I mean how to load UsbFile with glide

magnusja, 2019-11-12:
Ok I see. SO looking at the documentation (https://bumptech.github.io/glide/javadocs/490/index.html) it seems that this library does not support loading images via an `InputStream`. What you could do instead is using the libaums http module and then use the URI version of loading. Or you have a look at this tutorial where you can create your custom loader (ie. read file from USB using libaums): https://bumptech.github.io/glide/tut/custom-modelloader.html

leestars, 2019-11-15:
Thanks for your help, I wrote a custom custom ModelLoader that can load image files, but can't load video files. Can you help me?
```
import androidx.annotation.NonNull;

import com.bumptech.glide.Priority;
import com.bumptech.glide.load.DataSource;
import com.bumptech.glide.load.data.DataFetcher;
import com.github.mjdev.libaums.fs.UsbFile;
import com.github.mjdev.libaums.fs.UsbFileInputStream;

import java.io.IOException;
import java.io.InputStream;

public class UsbDataFetcher implements DataFetcher<InputStream> {
    private final UsbFile usbFile;

    UsbDataFetcher(UsbFile usbFile) {
        this.usbFile = usbFile;
    }

    private UsbFileInputStream is;

    @Override
    public void loadData(@NonNull Priority priority, @NonNull DataCallback<? super InputStream> callback) {
        is = new UsbFileInputStream(usbFile);
        callback.onDataReady(is);
    }

    @Override
    public void cleanup() {
        if (is != null) {
            try {
                is.close();
            } catch (IOException e) {
                e.printStackTrace();
            }
        }
    }

    @Override
    public void cancel() {
        if (is != null) {
            try {
                is.close();
            } catch (IOException e) {
                e.printStackTrace();
            }
        }
    }

    @NonNull
    @Override
    public Class<InputStream> getDataClass() {
        return InputStream.class;
    }

    @NonNull
    @Override
    public DataSource getDataSource() {
        return DataSource.LOCAL;
    }
}
```

leestars, 2019-11-15:
I changed the “loadData” method
```
@Override
    public void loadData(@NonNull Priority priority, @NonNull DataCallback<? super ByteBuffer> callback) {
        is = new UsbFileInputStream(usbFile);

        ByteArrayOutputStream outStream = new ByteArrayOutputStream();
        byte[] buffer = new byte[32768];
        int bytesRead = 0;
        try {
            while ((bytesRead = is.read(buffer)) != -1) {
                outStream.write(buffer, 0, bytesRead);
            }
            byte[] bytes = outStream.toByteArray();
            callback.onDataReady(ByteBuffer.wrap(bytes));
        } catch (IOException e) {
            e.printStackTrace();
            callback.onLoadFailed(e);
        }
    }
```
Loading the image file is still no problem, but the video file still fails to load.

```
class com.bumptech.glide.load.engine.GlideException: Failed to load resource
      Cause (1 of 1): class com.bumptech.glide.load.engine.GlideException: Failed LoadPath{HeapByteBuffer->Object->Drawable}, LOCAL
        Cause (1 of 3): class com.bumptech.glide.load.engine.GlideException: Failed DecodePath{HeapByteBuffer->GifDrawable->Drawable}
        Cause (2 of 3): class com.bumptech.glide.load.engine.GlideException: Failed DecodePath{HeapByteBuffer->Bitmap->Drawable}
        Cause (3 of 3): class com.bumptech.glide.load.engine.GlideException: Failed DecodePath{HeapByteBuffer->BitmapDrawable->Drawable}
```
I don't know how to get the thumbnail from UsbFileInputStream

magnusja, 2019-11-15:
This looks overly complicated. You do not need the `ByteArrayOutputStream`.

Are you sure glide supports video thumbnails? Looks like it only supports gifs.

magnusja, 2019-11-15:
Have a look here: https://futurestud.io/tutorials/glide-displaying-gifs-and-videos

## Display of Local Video Thumbnails

> One more time: it's important to note that this only works for local videos. Videos, which are not stored on the device (for example Internet URLs), will not work! Additionally, this will only display the first frame of that video.

> If you want to play videos or view videos from an Internet URL, look into the VideoView class.

leestars, 2019-11-15:
How do I convert a UsbFileInputStream to a ByteBuffer?
I know how to use glide to load a local video file, but now UsbFileInputStream is not a local video file, so I don't know how to get the first frame from UsbFileInputStream？

magnusja, 2019-11-15:
> How do I convert a UsbFileInputStream to a ByteBuffer?

There is no straight forward conversion. The first one is a stream as the name says the second one just some array in memory. For videos you probably do not want to load the whole file. So that is why you want to read some data from the stream and analyse it and then decide if you need more.

> I know how to use glide to load a local video file, but now UsbFileInputStream is not a local video file, so I don't know how to get the first frame from UsbFileInputStream？

Well I have never done this, but I guess you can google for that. In essence you need to find out the video format and then interpret the file. There are certainly libraries for that. Maybe this is helpful: https://medium.com/@mujtahidah/get-thumbnail-video-from-url-android-f71533168228

leestars, 2019-11-15:
I know this api (MediaMetadataRetriever), but this api still needs local video files, anyway, thank you for your help.

magnusja, 2019-11-15:
I think people have used this library in combination with libaums: https://github.com/wseemann/FFmpegMediaMetadataRetriever

(which supports URLs and might be usable with the http module)