magnusja/libaums #225: Appending to file

p-santos, 2019-11-14:
Sorry for noob question/issue but just got desperate, looked for days with no luck.
How do you append text to existing file? I tried using offset but get exceptions...

    // Testing writing log...
            String logMsgStr   = "Hello log 7\n";
            String logFileName = "testLog.txt";

            UsbFile usbLogFileO = null;

            try {
                usbLogFileO = rootUSB.search(logFileName);
            } catch (IOException e) {
                e.printStackTrace();
            }
            if (usbLogFileO != null) {
                Log.d("USB", "Found our log file!");
            }
            else {
                Log.d("USB", "Log file not found, creating it...");
                try {
                    usbLogFileO = rootUSB.createFile(logFileName);
                } catch (IOException e) {
                    Log.e("USB Error:", e.toString());
                }
            }

            //OutputStream os = new UsbFileOutputStream(usbLogFileO);
            // buffered....
            OutputStream os = UsbFileStreamFactory.createBufferedOutputStream(usbLogFileO, currentFs);

            Log.d("USB", "Writing file...");
            try {
                //os.write(logMsgStr.getBytes());
                // Append?
                os.write(logMsgStr.getBytes(), (int)usbLogFileO.getLength() + 1, logMsgStr.length());
                os.close();
            } catch (IOException e) {
                Log.e("USB Error writting:", e.toString());
            }
            Log.d("USB", "Write/close done.");

 Caused by: java.lang.ArrayIndexOutOfBoundsException: src.length=12 srcPos=12 dst.length=8192 dstPos=0 length=12
        at java.lang.System.arraycopy(System.java:521)
        at java.io.BufferedOutputStream.write(BufferedOutputStream.java:128)
        at com.example.VetteAlarmP1.MainActivity.onCreate(MainActivity.java:413)

Build 7.1 on Android Things, 8.0, API 27...

TIA,
Paul

magnusja, 2019-11-15:
Good catch, this is currently not implemented. I will implement ASAP and then release a new version, sorry for the inconvenience.

magnusja, 2019-11-15:
PR is merged, please try develop branch #226

closed 2019-11-15 by magnusja

p-santos, 2019-11-15:
Awesome thanks! Any chance you could give me some pointers on how to use the 'develop' branch in Android Studio/Gradle? I've been trying many variations of implementation and local dir/includes but no luck :(

TIA,
Paul

p-santos, 2019-11-16:
I found a way to reference 'develop' branch in AS/gradle config but then I get this error...

ERROR: Using multiple versions of the Android Gradle plugin in the same build is not allowed.
	'C:\Users\Me\AndroidStudioProjects\VetteAlarm-P1\app' is using version 3.5.1
	'C:\Users\Me\AndroidStudioProjects\VetteAlarm-P1\.gradle\vcs-1\libaums_5p8v364bbyy5nz7c4llq6t14e\libaums\app' is using version 3.5.0

Wondering if you know if better way?
In any case, I'll keep digging and sorry for my ignorance.

Paul

magnusja, 2019-11-16:
Just copy the contents of this (https://github.com/magnusja/libaums/tree/develop/libaums/src/main) to your src folder.

p-santos, 2019-11-16:
Still no luck but, no worries. I have plenty other work to do so will probably just wait for your next release/tag.

Thanks again,
Paul

magnusja, 2019-11-21:
released `v0.7.2`

p-santos, 2019-11-21:
Yep, saw that. Cool. Will test ASAP and will comment/feed back.

Thanks,
Paul

p-santos, 2019-11-22:
The append flag is now there but....I'm getting out of memory errors now even without the flag. Has anyone tried this build  yet?

2019-11-21 19:18:17.455 2943-2943/com.example.thingsgpio_1 I/zygote: Starting a blocking GC Alloc
2019-11-21 19:18:18.011 2943-2943/com.example.thingsgpio_1 I/zygote: Alloc concurrent copying GC freed 9(16KB) AllocSpace objects, 0(0B) LOS objects, 3% free, 54MB/56MB, paused 125us total 555.732ms
2019-11-21 19:18:18.012 2943-2943/com.example.thingsgpio_1 W/zygote: Throwing OutOfMemoryError "Failed to allocate a 83070912 byte allocation with 2097152 free bytes and 73MB until OOM, max allowed footprint 58974240, growth limit 134217728"
2019-11-21 19:18:18.057 2943-2943/com.example.thingsgpio_1 D/AndroidRuntime: Shutting down VM
2019-11-21 19:18:18.240 2943-2943/com.example.thingsgpio_1 E/AndroidRuntime: FATAL EXCEPTION: main
    Process: com.example.thingsgpio_1, PID: 2943
    java.lang.OutOfMemoryError: Failed to allocate a 83070912 byte allocation with 2097152 free bytes and 73MB until OOM, max allowed footprint 58973712, growth limit 134217728
        at java.util.Arrays.copyOf(Arrays.java:3139)
        at java.util.Arrays.copyOf(Arrays.java:3109)
        at java.util.ArrayList.grow(ArrayList.java:275)
        at java.util.ArrayList.ensureExplicitCapacity(ArrayList.java:249)
        at java.util.ArrayList.ensureCapacityInternal(ArrayList.java:241)
        at java.util.ArrayList.add(ArrayList.java:467)
        at com.github.mjdev.libaums.fs.fat32.FAT.getChain$libaums_release(FAT.kt:123)
        at com.github.mjdev.libaums.fs.fat32.ClusterChain.<init>(ClusterChain.kt:106)
        at com.github.mjdev.libaums.fs.fat32.FatFile.initChain(FatFile.kt:75)
        at com.github.mjdev.libaums.fs.fat32.FatFile.write(FatFile.kt:109)
        at com.github.mjdev.libaums.fs.UsbFileOutputStream.write(UsbFileOutputStream.kt:75)
        at java.io.BufferedOutputStream.flushBuffer(BufferedOutputStream.java:82)
        at java.io.BufferedOutputStream.flush(BufferedOutputStream.java:140)
        at java.io.FilterOutputStream.close(FilterOutputStream.java:158)
        at com.example.VetteAlarmP1.MainActivity.setupUSB(MainActivity.java:1191)
        at com.example.VetteAlarmP1.MainActivity.onCreate(MainActivity.java:312)
        at android.app.Activity.performCreate(Activity.java:7010)
        at android.app.Activity.performCreate(Activity.java:7001)
        at android.app.Instrumentation.callActivityOnCreate(Instrumentation.java:1214)
        at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2731)
        at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:2856)
        at android.app.ActivityThread.-wrap11(Unknown Source:0)
        at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1589)
        at android.os.Handler.dispatchMessage(Handler.java:106)
        at android.os.Looper.loop(Looper.java:164)
        at android.app.ActivityThread.main(ActivityThread.java:6494)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:438)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:807)
    	Suppressed: java.lang.OutOfMemoryError: Failed to allocate a 83070912 byte allocation with 2097152 free bytes and 73MB until OOM, max allowed footprint 58974240, growth limit 134217728
        at java.util.Arrays.copyOf(Arrays.java:3139)
        at java.util.Arrays.copyOf(Arrays.java:3109)
        at java.util.ArrayList.grow(ArrayList.java:275)
        at java.util.ArrayList.ensureExplicitCapacity(ArrayList.java:249)
        at java.util.ArrayList.ensureCapacityInternal(ArrayList.java:241)
        at java.util.ArrayList.add(ArrayList.java:467)
        at com.github.mjdev.libaums.fs.fat32.FAT.getChain$libaums_release(FAT.kt:123)
        at com.github.mjdev.libaums.fs.fat32.ClusterChain.<init>(ClusterChain.kt:106)
        at com.github.mjdev.libaums.fs.fat32.FatFile.initChain(FatFile.kt:75)
        at com.github.mjdev.libaums.fs.fat32.FatFile.setLength(FatFile.kt:58)
        at com.github.mjdev.libaums.fs.UsbFileOutputStream.close(UsbFileOutputStream.kt:51)
        at java.io.FilterOutputStream.close(FilterOutputStream.java:159)
        		... 15 more
2019-11-21 19:18:18.443 424-1331/system_process W/ActivityManager:   Force finishing activity com.example.thingsgpio_1/com.example.VetteAlarmP1.MainActivity

Unless I'm doing something wrong but, it's the same test code I had before....

Thanks,
Paul

magnusja, 2019-11-22:
Hey @p-santos 

that is interesting, You are sure that `0.7.1` works in the exact same setup? I did not change anything regarding that really. Only the append flag and https://github.com/magnusja/libaums/pull/230 (which hopefully should not affect reading the cluster chain)

p-santos, 2019-11-22:
I should be able to revert back to 0.7.1 and run same test right? I'll try it tonight.

p-santos, 2019-11-22:
Ok, it was doing same thing with 7.1 so...I took the USB stick and did a complete format FAT32 from my laptop. Wanted to rule that out before rechecking code, although the test code is very simple and brief.
Reason I thought of the reformat was cause the first time I tested with the append flag, when I checked the file from my laptop, it displayed an error trying to open the file. At the time, I figured the append flag was not working correctly or I wasn't using it correctly.

Anywhoo, after the reformat it was working ok on 7.1 so I then switched to 7.2 again and it worked fine. I then set the append flag and it also worked fine.

I did couple more appends to the file then checked the USB stick on my laptop and was able to open the file ok; no errors.

So...Not sure if the file system got corrupted somehow when I first used the append flag? Dunno.
The code is simple, does a write, closes the file, and closes the device.

I will continue testing later tonight...

Thanks,
Paul

p-santos, 2019-11-23:
Been banging on it all night; still working good. So...

The initial corruption is a mystery. Any Ideas about that?

Thanks for all your help.

Paul

magnusja, 2019-11-25:
Could be a nasty bug in the cluster chain which might be hard to reproduce. But if its working now, I guess open a new issue if you ever encounter that again :)

p-santos, 2019-11-25:
Sounds good. Thanks.

Paul