magnusja/libaums #228: how long will it take , to google(android) to give the path normally for usb ,like windows

azhar5678, 2019-11-18:
how long will it take , to google(android) to give the path normally for usb ,like windows

magnusja, 2019-11-18:
First of all, I am not affiliated with Google in any way nor am I part of Android development, so I do not know. 
Second, this works already for a lot of devices e.g. Samsung, Huawei and Sony.

closed 2019-11-18 by magnusja

azhar5678, 2019-11-23:
i developed the project of piracy protection software, for 100 pendrive daily sending client, and send one pendrive succsessfuly as a demo to the byuer,

but as a final doubt in client mind, they dont want runtimpermission message to be shown
and is it possible to run dex file any how from usb 

if u help me in this and i get the project
then i can pay u on monthly basis (100  to 150) dollor per month

magnusja, 2019-11-25:
If you want to read from USB you obviously need to have permission to do so. Android takes a different approach compared to Windows where everything is allowed by default and resulted in a significant amount of malware and viruses I guess. That is why there is no way around this permission in Android.

As far as I know it is not possible to run apps without installing them except in google play (https://developer.android.com/topic/google-play-instant)