magnusja/libaums #232: massStorageDevices[currentDevice].getPartitions() NO partitions

MixApplications, 2019-11-29:
first of all i would like to thank you for this great lib.
1st : i have issue
massStorageDevices[currentDevice].getPartitions() return 0 paritions when using NTFS usb file system and it work fine with FAT32.

2nd : i will be very pleasure if you add option for format USB device and changing its file system without root 

Thanks & Best Regards.

magnusja, 2019-11-29:
Hey there,

1. NTFS is not currently supported. Focus is on FAT32. If you include the module `java-fs` (unreleased you have to copy the code) then you can try an experimental version of NTFS (slow and probably unstable).
2. Can you give me a code snippet which reproduces this problem so I can try this easily?
3. Formatting is a whole different story, and I currently do not have plans to implement that. But do not hesitate to open a PR for this

MixApplications, 2019-11-29:
for (UsbFile usbFile : root.listFiles()) {
                                    try {
                                        if (!usbFile.getName().equalsIgnoreCase("System Volume Information"))
                                            usbFile.delete();
                                        freeMemory();
                                    } catch (Exception ignored) {
                                        Log.d(TAG, "start: can not delete file : " + usbFile.getName());
                                    } 
                                }

when using this code i got: OutOfMemoryError: Failed to allocate a 4308492 byte allocation with 467872 free bytes and 456KB until OOM . or some thing like that .
then app crash 
it happend when usb have alot of files more than 100 file

----------------------------------------------------------------------
an other new issue : 
UsbFile file = cur_dir.createFile(path);
OutputStream os = UsbFileStreamFactory.createBufferedOutputStream(file, currentFs);
ExtractOperationResult result = item.extractSlow(new ISequentialOutStream() {
                                        @Override
                                        public int write(byte[] bytes) throws SevenZipException {
                                            if (bytes == null || bytes.length == 0) {
                                                Log.d(TAG, "write: Data Null");
                                            } else {
                                                try {
                                                    os.write(bytes);
                                                    os.close();
                                                } catch (IOException e) {
                                                    e.printStackTrace();
                                                }
                                            }
                                            return bytes != null ? bytes.length : 0;
                                        }

this code working fine with first files but after some of files it makes null files .
this files name  looking like  bytes[].tostring()
and its size = 0 byte
and missing alot of files too.

note: extractor working fine with : Environment.getExternalStorageDirectory() no miss file no wrong file

sorry for my bad english

MixApplications, 2019-11-30:
@magnusja any help?

magnusja, 2019-12-02:
This looks a lot like this one: https://github.com/magnusja/libaums/issues/112

which version are you using?

magnusja, 2019-12-09:
@MixApplications can you please test this with `0.7.3`

closed 2019-12-09 by magnusja

MixApplications, 2019-12-09:
good job man , working very well

‫في الاثنين، 9 ديسمبر 2019 في 5:40 م تمت كتابة ما يلي بواسطة ‪Magnus‬‏ <‪
notifications@github.com‬‏>:‬