magnusja/libaums #233: corrupted and missing files

MixApplications, 2019-12-01:
missing files and corrupted files while copy from phone to usb 

 files name  looking like  bytes[].tostring() files name like : (  ïU¼â─R.h0   , ╝ëm╕k╦dï.]╝≈   , â∞ u└╟.  ,  p w wΘ.¬ )
[image]

file size not only 0 butes there is files its size larger than usb pen drive size

"volume label in non root dir!"  logcat

magnusja, 2019-12-02:
Hey there,

do you see these files also when you plug it back into your Android phone and use libaums? Do you see only additional junk files or do you also miss files which should be there? And again, what version are you using?

MixApplications, 2019-12-03:
Yes i see this files in Android too i using 0.7.2 version and this files can't delete it too i must format it

magnusja, 2019-12-03:
When you say ANdroid do you mean the Android system itself or the usbfileman app in this repo?

What do you extract? I `dd` the iso to a usb drive and plug it into my phone and then what?

magnusja, 2019-12-03:
[image]
I think I am finally able to reproduce this 🙌

MixApplications, 2019-12-04:
nice finally you understand me :)
sorry for bad English

kuza2010, 2019-12-05:
Hi, guys!✌ 
The firstly, I want to say, that it really amazing library!
But, unfortunately, I am facing this issue too. 
I try to copy a folder with mixed files from internal storage to USB.

After copy folder on usb looks like:
[image]

Please, let me know how I can help you.

magnusja, 2019-12-09:
Hey @kuza2010 @MixApplications 

can you please test the new version `0.7.3`

Thanks

kuza2010, 2019-12-09:
Hi @magnusja, I found your latest changes.
Now I **can't see** some ugly files! But, I getting a new error. I try to copy many files and in copy progress, I get the next error:
`java.io.IOException:` Could not write to device, result == -1 errno 0 null` 
Every time with different stack trace. 

For example, when I try OutputStream.write(byte):
[image]

Any hints?

Thanks

magnusja, 2019-12-09:
This one is a nasty one. Maybe try another pen drive. Are you using the storage provider module?

kuza2010, 2019-12-09:
No, I am not using the storage provide module.
Okay, I will try it with another usb. I'll let you know what I can do.

Thanks a lot

magnusja, 2019-12-09:
Okay let me know how it goes. I will close this for now, feel free to ask some questions over gitter.

closed 2019-12-09 by magnusja