magnusja/libaums #235: java.lang.NoSuchMethodError: No virtual method position(I)Ljava/nio/ByteBuffer; in class Ljava/nio/ByteBuffer; or its super classes (declaration of 'java.nio.ByteBuffer' appears in /system/framework/core-oj.jar)

GHdeng, 2019-12-03:
java.lang.RuntimeException: An error occurred while executing doInBackground()
        at android.os.AsyncTask$3.done(AsyncTask.java:325)
        at java.util.concurrent.FutureTask.finishCompletion(FutureTask.java:354)
        at java.util.concurrent.FutureTask.setException(FutureTask.java:223)
        at java.util.concurrent.FutureTask.run(FutureTask.java:242)
        at android.os.AsyncTask$SerialExecutor$1.run(AsyncTask.java:243)
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1133)
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:607)
        at java.lang.Thread.run(Thread.java:761)
     Caused by: java.lang.NoSuchMethodError: No virtual method position(I)Ljava/nio/ByteBuffer; in class Ljava/nio/ByteBuffer; or its super classes (declaration of 'java.nio.ByteBuffer' appears in /system/framework/core-oj.jar)
        at org.jnode.util.ByteBufferUtils$ByteArray.refreshByteBuffer(ByteBufferUtils.java:75)
        at org.jnode.fs.ntfs.NTFSFile.read(NTFSFile.java:112)
        at com.github.magnusja.libaums.javafs.wrapper.fs.UsbFileWrapper.read(UsbFileWrapper.java:171)
        at com.github.mjdev.libaums.fs.UsbFileInputStream.read(UsbFileInputStream.kt:79)
        at com.github.mjdev.libaums.usbfileman.MainActivity$CopyTask.doInBackground(MainActivity.java:317)
        at com.github.mjdev.libaums.usbfileman.MainActivity$CopyTask.doInBackground(MainActivity.java:285)
        at android.os.AsyncTask$2.call(AsyncTask.java:305)
        at java.util.concurrent.FutureTask.run(FutureTask.java:237)
        at android.os.AsyncTask$SerialExecutor$1.run(AsyncTask.java:243) 
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1133) 
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:607) 
        at java.lang.Thread.run(Thread.java:761)

magnusja, 2019-12-03:
See https://github.com/magnusja/libaums/issues/212

magnusja, 2019-12-03:
Please try `compile 'com.github.magnusja:java-fs:0.1.5'`

magnusja, 2019-12-03:
I rebuilt `0.1.5` with JDK version 1.8.x

closed 2019-12-03 by magnusja