magnusja/libaums #238: FAT 12

thiblades, 2019-12-11:
i am currently trying to use the libaums library to read an .ibf file (generated by an Fat 12 device)

I have read the listed issues on Github and I found that it could possible to adapt this library to Fat 16/12.

#### Stacktrace of Excpetion

```2019-12-09 11:47:05.423 27679-27679/com.github.mjdev.usbfileman E/AndroidRuntime: FATAL EXCEPTION: main
    Process: com.github.mjdev.usbfileman, PID: 27679
    java.lang.NoSuchMethodError: No virtual method clear()Ljava/nio/ByteBuffer; in class Ljava/nio/ByteBuffer; or its super classes (declaration of 'java.nio.ByteBuffer' appears in /system/framework/core-oj.jar)
        at org.jnode.fs.jfat.FatMarshal.getByteBuffer(FatMarshal.java:57)
        at org.jnode.fs.jfat.FatRootDirectory.getFatDirEntry(FatRootDirectory.java:58)
        at org.jnode.fs.jfat.FatEntriesFactory.hasNext(FatEntriesFactory.java:59)
        at org.jnode.fs.jfat.FatDirectory.scanDirectory(FatDirectory.java:262)
        at org.jnode.fs.jfat.FatRootDirectory.<init>(FatRootDirectory.java:40)
        at org.jnode.fs.jfat.FatFileSystem.createRootEntry(FatFileSystem.java:85)
        at org.jnode.fs.jfat.FatFileSystem.createRootEntry(FatFileSystem.java:35)
        at org.jnode.fs.spi.AbstractFileSystem.getRootEntry(AbstractFileSystem.java:111)
        at com.github.magnusja.libaums.javafs.wrapper.fs.FileSystemWrapper.getRootDirectory(FileSystemWrapper.java:30)
        at com.github.mjdev.libaums.usbfileman.MainActivity.setupDevice(MainActivity.java:662)
        at com.github.mjdev.libaums.usbfileman.MainActivity.discoverDevice(MainActivity.java:635)
        at com.github.mjdev.libaums.usbfileman.MainActivity.onCreate(MainActivity.java:585)
        at android.app.Activity.performCreate(Activity.java:7327)
        at android.app.Activity.performCreate(Activity.java:7318)
        at android.app.Instrumentation.callActivityOnCreate(Instrumentation.java:1271)
        at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:3096)
        at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:3259)
        at android.app.servertransaction.LaunchActivityItem.execute(LaunchActivityItem.java:78)
        at android.app.servertransaction.TransactionExecutor.executeCallbacks(TransactionExecutor.java:108)
        at android.app.servertransaction.TransactionExecutor.execute(TransactionExecutor.java:68)
        at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1950)
        at android.os.Handler.dispatchMessage(Handler.java:106)
        at android.os.Looper.loop(Looper.java:214)
        at android.app.ActivityThread.main(ActivityThread.java:7073)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:493)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:965)```

### Code where problem occurs

magnusja, 2019-12-11:
Hmm I thought I fixed that. Which version of java-fs are you using?

thiblades, 2019-12-12:
Ok I really don't know how but it's working now on the sample application!

magnusja, 2019-12-12:
This is why https://github.com/magnusja/libaums/commit/3151142505d479fe21d5487e78c8760b32e82e6e :D

thiblades, 2019-12-12:
Ok thank you!! just what i was missing in my gradle

thiblades, 2019-12-12:
An other trouble is that in my application i am unable to do 

```compile 'com.github.magnusja:java-fs:0.1.5```

I get this error message : 
``` ERROR: Failed to resolve: com.github.magnusja:java-fs:0.1.5```

magnusja, 2019-12-12:
```
allprojects {
    repositories {
        jcenter()
        maven { url 'https://dl.bintray.com/magnusja/maven' }
        maven {
            url 'https://maven.google.com/'
            name 'Google'
        }
    }
}
```

closed 2019-12-12 by magnusja