magnusja/libaums #245: Getting .c extension error

Basavaraj1991, 2020-01-23:
CMake Error at CMakeLists.txt:11 (add_library):
Cannot find source file:
null/libusb/core.c
Tried extensions .c .C .c++ .cc .cpp .cxx .m .M .mm .h .hh .h++ .hm .hpp
.hxx .in .txx

Getting above error when i try to run the code

Depau, 2020-01-23:
Hi,
you need to dowload a `libusb-1.0.23` tarball somewhere in your system, then add

```
libusb.dir=/path/to/libusb-1.0.23
```

to `local.properties`.

Depau, 2020-01-23:
Please beware of the [licensing issues](https://github.com/magnusja/libaums/tree/develop/libusbcommunication) you might incur if you want to include the libusb-based UsbCommunication in a non-open source application

Basavaraj1991, 2020-01-23:
Hey I've requirement I've copy from android internal storage to USB(pendrive) I'm not getting how to do can you please help me.

magnusja, 2020-01-23:
Any reason that you are compiling libaums yourself? If you want to use it I would just declare it as a dependency.

This is described in the readme as well as supported in the example application, please consult both of these resources.

closed 2020-01-23 by magnusja